emsesp/EMS-ESP32 #2132: heating_active and tapwater_active constant 'off'

mikkel75, 2024-10-23:
### DESCRIPTION

After updating to 3.6.5 (from <3.6) I now see the special functions/values heating_active and tapwater_active as constant 'off' even when I know the boiler is in fact active.
The active values worked fine in old version.

### REQUESTED INFORMATION

```json
{
"System Info": {
"version": "3.6.5",
"uptime": "000+16:12:51.319",
"uptime (seconds)": 58371,
"platform": "ESP32",
"arduino": "Tasmota Arduino v2.0.14",
"sdk": "4.4.6.240105",
"free mem": 146,
"max alloc": 79,
"used app": 1591,
"free app": 393,
"partition": "app1",
"reset reason": "Software reset CPU / Software reset CPU"
},
"Network Info": {
"network": "WiFi",
"hostname": "ems-esp",
"RSSI": -69,
"TxPower setting": 78,
"static ip config": false,
"enable IPv6": true,
"low bandwidth": false,
"disable sleep": false,
"enable MDNS": true,
"enable CORS": false,
"AP provision mode": "disconnected",
"AP security": "wpa2",
"AP ssid": "ems-esp"
},
"NTP Info": {
"NTP status": "connected",
"enabled": true,
"server": "time.google.com",
"tz label": "Europe/Copenhagen"
},
"OTA Info": {
"enabled": true,
"port": 8266
},
"MQTT Info": {
"MQTT status": "connected",
"MQTT publishes": 7039,
"MQTT queued": 0,
"MQTT publish fails": 0,
"MQTT connects": 1,
"enabled": true,
"client id": "ems-esp",
"keep alive": 60,
"clean session": false,
"entity format": 1,
"base": "ems-esp",
"discovery prefix": "homeassistant",
"discovery type": 0,
"nested format": 1,
"ha enabled": true,
"mqtt qos": 0,
"mqtt retain": false,
"publish time heartbeat": 60,
"publish time boiler": 30,
"publish time thermostat": 30,
"publish time solar": 30,
"publish time mixer": 30,
"publish time other": 30,
"publish time sensor": 30,
"publish single": false,
"publish2command": false,
"send response": false
},
"Syslog Info": {
"enabled": false
},
"Sensor Info": {},
"API Info": {
"API calls": 1,
"API fails": 1
},
"Bus Info": {
"bus status": "connected",
"bus protocol": "HT3",
"bus telegrams received (rx)": 161847,
"bus reads (tx)": 38060,
"bus writes (tx)": 0,
"bus incomplete telegrams": 1,
"bus reads failed": 0,
"bus writes failed": 0,
"bus rx line quality": 100,
"bus tx line quality": 100
},
"Settings": {
"board profile": "CUSTOM",
"locale": "en",
"tx mode": 1,
"ems bus id": 11,
"shower timer": false,
"shower alert": false,
"phy type": 0,
"rx gpio": 17,
"tx gpio": 16,
"dallas gpio": 0,
"pbutton gpio": 0,
"led gpio": 0,
"hide led": false,
"notoken api": false,
"readonly mode": false,
"fahrenheit": false,
"dallas parasite": false,
"bool format": 1,
"bool dashboard": 1,
"enum format": 1,
"analog enabled": false,
"telnet enabled": false,
"max web log buffer": 25,
"web log buffer": 7
},
"Devices": [
{
"type": "boiler",
"name": "Enviline/Compress 6000AW/Hybrid 3000-7000iAW/SupraEco/Geo 5xx/WLW196i/WSW196i",
"device id": "0x08",
"product id": 172,
"version": "01.20",
"entities": 147,
"handlers received": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
"handlers fetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC",
"handlers pending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF",
"handlers ignored": "0x049F 0x04A0 0x04A3 0xF7 0x0240 0x0291 0x029B 0xE7 0x02E0 0x02EA 0x061E 0x04AC 0x2E 0x3B 0x0497"
},
{
"type": "thermostat",
"name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II/HPC410",
"device id": "0x10",
"product id": 158,
"version": "73.03",
"entities": 60,
"handlers received": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
"handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E",
"handlers pending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E",
"handlers ignored": "0x0508 0x0509 0x050A 0x0507 0x0239 0xBF"
},
{
"type": "gateway",
"name": "MX300",
"device id": "0x48",
"product id": 252,
"version": "08.02",
"entities": 0,
"handlers ignored": "0x2040 0xF9 0x0495 0x0494 0x3B 0xE5 0xE9 0x0CBD"
}
]
}

```

### TO REPRODUCE

Constant problem, have tried repower and reboot.

proddy, 2024-10-23:
Many changes since 3.6.4 - could you upgrade again to 3.7.0 and see if that helps

mikkel75, 2024-10-23:
I have now updated to latest beta 3.7 - it's still same problem with the '*_active' values.
The Compressor activity and power seems OK - but the 'active' value is constant 'off':

[image]

proddy, 2024-10-23:
Both boiler and tapwater are calculated values in EMS-ESP, doing some complex algorithms. Can you 
- tell us which boiler you have
- verify that you see the topics tapwater_active and heating_active (use HA or MQTTExplorer)
- that you're removed all old MQTT discovery topics from the MQTT broker/server (using MQTTExplorer) as you may be using the old naming
- the output of `read 8 e4`
- the output of `read 8 e3`
- the output of `read 8 18`

hopefully we can find the issue and fix in the upcoming release. Next time please don't wait a year before upgrading! :-)

mikkel75, 2024-10-23:
Hi,

- Boiler type etc is part of the initial system info, right? Need more than that?
- Yes I see them - but as always 'off'
[image]
- I have not removed anything on discovery, but I see them as 'off' on the EMS-ESP GUI as well sooo

Where do I do those reads? Some CLI?

mikkel75, 2024-10-23:
Results:

2024-10-23 15:24:51.776 I 9: [command] Called command system/read (send read request) with value 8 e4
2024-10-23 15:24:51.972 N 10: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 25 01 27 2F 00 02 00 02 B2 00 00 01 58 00 00 00 00 01 77 01 27
2024-10-23 15:24:51.990 N 11: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 27)
2024-10-23 15:25:55.715 I 12: [command] Called command system/read (send read request) with value 8 e3
2024-10-23 15:25:56.258 N 13: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 26 00 64 55
2024-10-23 15:26:13.081 I 14: [command] Called command system/read (send read request) with value 8 18
2024-10-23 15:26:13.516 N 15: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFast(0x18), data: <empty>

proddy, 2024-10-23:
can't see what is wrong here. Any ideas @MichaelDvP ?

MichaelDvP, 2024-10-24:
Reading e3/e4 is fine, but useless without knowing the state of heating/tapwater. 
Please post the telegrams e3 and e4 for state off, state heating and state tapwater to see the difference. 
The detection was not for years, what version has working?

mikkel75, 2024-10-24:
Hi,

The issue is that the heating and tapwater is always 'off' - also when I did the readings.
I was using 3.5.x before updating the other day...

mikkel75, 2024-10-24:
I've changed my scripts to use the boiler_hpactivity instead on rely on interpreted activity variables - that seems to solve it for me.
Maybe others see the active-variables working and it's just my setup, I don't know - but they are constant off for me.

Is it correct that the boiler_hpactivity is coming directly from the boiler? Why is that not used to generate the active-variables?

[image]

MichaelDvP, 2024-10-25:
> I've changed my scripts to use the boiler_hpactivity instead on rely on interpreted activity variables - that seems to solve it for me.

For you this may solve it, but if you like to help and send us the telegrams and we can fix it for all users.

Moving4407, 2024-11-04:
> Hi,
> 
> The issue is that the heating and tapwater is always 'off' - also when I did the readings. I was using 3.5.x before updating the other day...

I had the same problem on a Moduline 3000 after changing from weather controlled to room controlled. I simply switched dhw and hc to active and they have worked fine for days now. Still wonder why this was triggered.

red-black-bonsai, 2024-11-06:
Seems like I'm also seeing this problem atm. Also, "dhw current tap water flow" is always 0 for me. I'll try to give the requested debugging info:

The boiler in question is a Logatherm wlw166i-14 e heatpump that is identified by EMS-ESP as a "Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i (Boiler/HP)"

I switched the heating on/off with command boiler/heatingactivated, ran the tapwater, and executed the given commands 4 times (for all 4 possible combinations of tapwater and heating on/off):

## heating off (boiler/heatingactivated off) / tapwater off

    ems-esp:$ read 8 e4
    000+00:24:25.192 N 3: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 00 01 ED 00 00 00 00 02 B2 00 00 01 D5 00 00 00 00 01 FB 01 EE

    ems-esp:$ read 8 e3
    000+00:24:35.159 N 4: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 ED 00 64 55

    ems-esp:$ read 8 18
    000+00:24:45.009 N 5: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFast(0x18), data: <empty>

## heating on (boiler/heatingactivated on) / tapwater off

    ems-esp:$ read 8 e4
    000+00:28:41.216 N 7: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 00 01 DD FF 00 00 00 02 B2 00 00 01 D1 00 00 00 00 01 C4 01 DD
    000+00:28:41.237 N 8: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 27)

    ems-esp:$ read 8 e3
    000+00:28:56.083 N 9: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 DD 00 64 55

    ems-esp:$ read 8 18
    000+00:29:08.017 N 10: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFast(0x18), data: <empty>

## tapwater running / heating on (boiler/heatingactivated on)

    ems-esp:$ read 8 e4
    000+00:34:35.027 N 11: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 00 01 D0 FF 00 00 00 02 B2 00 00 01 CE 00 00 00 00 01 BD 01 CF
    000+00:34:35.052 N 12: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 27)

    ems-esp:$ read 8 e3
    000+00:34:44.065 N 13: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 CF 00 64 55

    ems-esp:$ read 8 18
    000+00:34:52.311 N 14: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFast(0x18), data: <empty>

## tapwater running / heating off (boiler/heatingactivated off)

    ems-esp:$ read 8 e4
    000+00:40:50.601 N 16: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 00 01 C2 00 00 00 00 02 B2 00 00 01 C7 00 00 00 00 01 B2 01 C1
    000+00:40:50.622 N 17: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 27)

    ems-esp:$ read 8 e3
    000+00:40:58.731 N 18: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorSlowPlus2(0xE3), data: 04 00 00 00 00 00 00 00 00 00 00 01 C2 00 64 55

    ems-esp:$ read 8 18
    000+00:41:08.227 N 19: [emsesp] boiler(0x08) -W-> me(0x0B), UBAMonitorFast(0x18), data: <empty>

Does this help?

MichaelDvP, 2024-11-06:
Thank you, that is great. 
> Also, "dhw current tap water flow" is always 0 for me

Flow sensors are only installed on flow systems, but i think heatpumps work always with dhw buffer. The heat power is too less for heating the water on the flow. So running tapwater does not work, it's just sourced from the buffer and heating dhw is only while charging the buffer. 

But i can see that we have the state in offset 11 of 0xe4 always zero, but 0xe3 offset 2 shows heating. I guess dhw prepare is then 0xe3 offset 5 as for the HIU unit.

Please check if it works with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest.

red-black-bonsai, 2024-11-08:
Can confirm that with your dev build `heating active` follows changes of `heating activated` with a ~30 sec delay. :+1: 

Did you want me to check something about the tapwater? I don't see a value of "dhw prepare" on my devices. Is `tapwater active` meant to change now if the dhw buffer is refilled or should I just ignore it?

commit 6ab3ce8b (2024-11-14): boilerstate from 0xE3 for heatpumps, fix #2132
closed 2024-11-14 by proddy