emsesp/EMS-ESP32 #2140: Emulated RC100H causes many incomplete Rx

Sbried, 2024-10-26:
### DESCRIPTION

Normally I don´t see warnings in the log, the system runs stable for long periods. 
Now I´m experimenting with the emulated thermostat RC100H.
I activated it with sending RC100H to `ems-esp/thermostat/hc1/control`. Through the API I send the actual temperature and humidity data to `ems-esp/thermostat/hc1/remotetemp` and `ems-esp/thermostat/hc1/remotehum`. All is recognized well and the HMC310 is showing the correct data.
The issue now is that I receive a lot of incomplete Rx warnings in the log. All sent by the emulated RC100H 0x38 to the thermostat HMC310 0x10.
After a random time between some hours and a day the ems-esp reboots.

I´ve read a lot in the issues/discussions and tried with different Tx protocols EMS and EMS+. Both result in the same. The power supply should be good, I don´t have issues without the emulated RC100H. The ems-esp is connected to the service jack of the heat pump.
There is no correlation between updating the temperature/humidity through the API and the warnings. As I understood the code the temperature/humidity data is sent every 15 sec to the thermostat 0x10. Some of them may end up in a warning being incomplete.

Some lines from the log:
2024-10-26 08:07:45.976 WARNING 10: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 FD
2024-10-26 08:25:58.028 WARNING 11: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8B 0A
2024-10-26 08:35:09.035 WARNING 12: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 08:36:14.049 WARNING 13: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 08:43:46.058 WARNING 14: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F3
2024-10-26 09:06:05.124 WARNING 15: [telegram] Incomplete Rx: 38 10 FF 00 03 2B FE
2024-10-26 09:10:09.132 WARNING 16: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8B 0A
2024-10-26 09:11:13.110 WARNING 17: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 38 00 8B 0F E7
2024-10-26 09:12:33.135 WARNING 18: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8B 0A
2024-10-26 09:32:16.547 WARNING 19: [telegram] Incomplete Rx: 38 10 FF 00 03 38 00 8C 0D
2024-10-26 09:35:56.547 WARNING 20: [telegram] Incomplete Rx: 38 10 FF 00 03 E9 F2
2024-10-26 09:37:51.204 WARNING 21: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8C 0D
2024-10-26 09:45:11.659 WARNING 22: [telegram] Incomplete Rx: 38 10 FF 00 03 E9 F2
2024-10-26 09:48:26.565 WARNING 23: [telegram] Incomplete Rx: 38 10 FF 00 03 2F FE 00 0D
2024-10-26 09:49:31.230 WARNING 24: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8C 0D
2024-10-26 09:51:07.232 WARNING 25: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8C 0D
2024-10-26 09:55:14.233 WARNING 26: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 09:58:28.250 WARNING 27: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 10:04:33.257 WARNING 28: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 10:09:42.283 WARNING 29: [telegram] Incomplete Rx: 38 10 02 FE 28 FF 98
2024-10-26 10:20:36.297 WARNING 30: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 38 00 8E 8D
2024-10-26 10:33:49.714 WARNING 31: [telegram] Incomplete Rx: 38 10 02 00 C8 28 04 00 98
2024-10-26 10:35:58.421 WARNING 32: [telegram] Incomplete Rx: 38 10 FF 00 A0 00 E9 F2
2024-10-26 10:44:00.022 WARNING 33: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 38 00 8C 9E
2024-10-26 10:54:42.839 WARNING 34: [telegram] Incomplete Rx: 38 10 02 00 C8 28 04 00 98
2024-10-26 10:54:56.381 WARNING 35: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F0
2024-10-26 11:08:37.164 WARNING 36: [telegram] Incomplete Rx: 38 10 FF 00 03 0E 38 00 8E 0F
2024-10-26 11:15:39.444 WARNING 37: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F0
2024-10-26 11:22:22.452 WARNING 38: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F0
2024-10-26 11:29:49.474 WARNING 39: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 11:40:28.005 WARNING 40: [telegram] Incomplete Rx: 38 10 FF 00 03 7B FE 00 FF E4
2024-10-26 11:47:59.521 WARNING 41: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F0
2024-10-26 11:48:17.013 WARNING 42: [telegram] Incomplete Rx: 38 10 FF 00 A0 0E 38 00 8E 0F
2024-10-26 11:48:49.303 WARNING 43: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 38 00 8E 89
2024-10-26 11:52:54.531 WARNING 44: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 8E 0F
2024-10-26 12:08:38.571 WARNING 45: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F0
2024-10-26 12:20:02.052 WARNING 46: [telegram] Incomplete Rx: 38 10 FF 00 A0 0E 38 00 8C 0D
2024-10-26 12:30:59.624 WARNING 47: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 12:31:16.624 WARNING 48: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 12:33:26.758 WARNING 49: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 38 00 8C 89
2024-10-26 12:36:56.644 WARNING 50: [telegram] Incomplete Rx: 38 10 FF 00 03 2B 00 F2
2024-10-26 12:55:25.700 WARNING 51: [telegram] Incomplete Rx: 38 10 02 00 28 04 00 FF 98
2024-10-26 12:59:15.701 WARNING 52: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 89 34
2024-10-26 13:00:53.703 WARNING 53: [telegram] Incomplete Rx: 38 10 FF 00 03 7B 0E 89 34

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {
    "system": {
[... 164 lines cut ...]
            "entities": 0,
            "handlersIgnored": "0x2040 0x0495 0x0494 0x3B 0xE5 0xE9 0xF9"
        }
    ]
}
```

### TO REPRODUCE

Maybe also others can reproduce it by setting up an emulated thermostat.
I can reproduce it by removing the emulated RC100H (no more warnings in the log) and setting the RC100H again (log warning start again)

### EXPECTED BEHAVIOUR

Emulated thermostat does not cause incomplete Rx

### SCREENSHOTS

na

### ADDITIONAL CONTEXT

na

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-10-27:
I can't explain the restarts. Next time it happens see if you capture the log as it'll show the 'reason'.

@MichaelDvP any idea on those short Rx telegrams?

MichaelDvP, 2024-10-27:
For the reboots: was it #2122, i can see that now you're using dev48 and uptime is > 5 days.

The incompletes are difficult. As you've found it sends the message every 15 sec, the wrong readback is one in 5-10 min, so it's only 3% of the telegrams sent, that have issue with readback. Also the master thermostat detect the remote constantly, so maybe only the readback in emsesp have problems. For sendings from ems-esp-device-address we do not show the readback, maybe there are also issues. The messages do not affect operation.
Sending and receiving the same time is difficult because the sending (drain current from bus) can effect a local drop in bus voltage, resulting in a read-error only on the sending device. 
Try with tx-mode EMS+, that has more delay after sending, helping to stabilize the bus-voltage.

Sbried, 2024-10-28:
Actually the uptime was >5 hrs from the system info above but anyhow - it is running till now, > 2 days. The dev48 seemed to fix the reboots.
I had the EMS+ tx-mode before and also had a lot of incomplete messages. Maybe not so many, I didn´t make statistics about it. 
If it is hardware related, maybe not showing the readback is also a solution. Then the emulated remote thermostat should have a special marker to detect it´s emulated or a real device. Maybe a special SW version for the emulated one.

MichaelDvP, 2024-10-29:
I have to correct me, we show the echos of emsesp sending (it was an early version that suppress the echos).

Can you make a full log of the telegram to see in which context the errors happen. I suspect that this is not the normal 15 sec sending but an answer to a query, maybe with offset/length different from answer (the emulation always send the complete telegram).

MichaelDvP, 2024-10-29:
Yes, they do not give a warning or tx-error count, but in full log they are shown with loglevel trace (T).

MichaelDvP, 2024-10-30:
Could you post a full log (log all) to see if the incompletes are only for thermostat or also for other emsesp sendings.

Sbried, 2024-10-30:
Attached is an 8 minutes log all.
[logall.txt](https://github.com/user-attachments/files/17578999/logall.txt)
It contains 10 incomplete messages. One is from the emulated RC100H (0x38), the other 9 are from EMS-ESP (0x0B). But the incompletes from EMS-ESP itself do not show up in the warnings, they go only in the trace.
This is what I also would like for the emulated RC100H (and other emulated things) too.

MichaelDvP, 2024-10-31:
> This is what I also would like for the emulated RC100H (and other emulated things) too.

This should work with my build https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest Please check, In my system i don't have incompletes for own sendings.

How is your emsesp connected to bus? Try if it get's better with a shorter/thicker bus-cable.

mattreim, 2024-11-12:
Hello everyone, I have now finished my tests with dev.1 and dev.3. Over 1 million telegrams were transmitted in just under 6 days.

3.7.1-dev.1:
[image]

3.7.1-dev.3:
[image]

I think a reduction from 98 to 10 incompletes errors is very good.

```
CPU: ESP32-D0WD-V3
Platform: ESP32-4MB
EMS Tx mode: EMS
Emulated remote: RC100H
```

closed 2024-12-19 by proddy