emsesp/EMS-ESP32 #2142: Problem with daylight time in Australia

VlastiBroucek, 2024-10-27:
### DESCRIPTION

Problem with daylight time starting/ending in Australia

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {
    "System Info": {
        "version": "3.6.5",
        "uptime": "000+00:06:57.467",
        "uptime (seconds)": 417,
        "platform": "ESP32",
        "arduino": "ESP32 Arduino v2.0.14",
        "sdk": "v4.4.6-dirty",
        "free mem": 156,
        "max alloc": 107,
        "used app": 2240,
        "free app": 320,
        "partition": "boot",
        "reset reason": "Power on reset / APP CPU reset by PRO CPU"
    },
    "Network Info": {
        "network": "Ethernet",
        "hostname": "ems-esp",
        "TxPower setting": 0,
        "static ip config": false,
        "enable IPv6": true,
        "low bandwidth": false,
        "disable sleep": false,
        "enable MDNS": true,
        "enable CORS": false,
        "AP provision mode": "disconnected",
        "AP security": "wpa2",
        "AP ssid": "ems-esp"
    },
    "NTP Info": {
        "NTP status": "disconnected",
        "enabled": false,
        "server": "time.cloudflare.com",
        "tz label": "Australia/Hobart"
    },
    "OTA Info": {
        "enabled": false,
        "port": 8266
    },
    "MQTT Info": {
        "MQTT status": "connected",
        "MQTT publishes": 388,
        "MQTT queued": 0,
        "MQTT publish fails": 0,
        "MQTT connects": 1,
        "enabled": true,
        "client id": "esp32-d767b4fc",
        "keep alive": 60,
        "clean session": false,
        "entity format": 1,
        "base": "ems-esp",
        "discovery prefix": "homeassistant",
        "discovery type": 0,
        "nested format": 1,
        "ha enabled": true,
        "mqtt qos": 0,
        "mqtt retain": false,
        "publish time heartbeat": 60,
        "publish time boiler": 10,
        "publish time thermostat": 10,
        "publish time solar": 10,
        "publish time mixer": 10,
        "publish time other": 60,
        "publish time sensor": 10,
        "publish single": false,
        "publish2command": false,
        "send response": false
    },
    "Syslog Info": {
        "enabled": false
    },
    "Sensor Info": {
        "temperature sensors": 3,
        "temperature sensor reads": 249,
        "temperature sensor fails": 0,
        "analog sensors": 0,
        "analog sensor reads": 0,
        "analog sensor fails": 0
    },
    "API Info": {
        "API calls": 0,
        "API fails": 0
    },
    "Bus Info": {
        "bus status": "connected",
        "bus protocol": "HT3",
        "bus telegrams received (rx)": 856,
        "bus reads (tx)": 279,
        "bus writes (tx)": 100,
        "bus incomplete telegrams": 0,
        "bus reads failed": 0,
        "bus writes failed": 0,
        "bus rx line quality": 100,
        "bus tx line quality": 100
    },
    "Settings": {
        "board profile": "E32V2",
        "locale": "en",
        "tx mode": 3,
        "ems bus id": 11,
        "shower timer": false,
        "shower alert": false,
        "hide led": false,
        "notoken api": false,
        "readonly mode": false,
        "fahrenheit": false,
        "dallas parasite": false,
        "bool format": 1,
        "bool dashboard": 1,
        "enum format": 1,
        "analog enabled": true,
        "telnet enabled": true,
        "max web log buffer": 100,
        "web log buffer": 100
    },
    "Devices": [
        {
            "type": "boiler",
            "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
            "device id": "0x08",
            "product id": 95,
            "version": "12.16",
            "entities": 69,
            "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
            "handlers fetched": "0x14 0x16 0x33",
            "handlers pending": "0xBF 0xC2 0x1A 0x35 0x2A",
            "handlers ignored": "0x23"
        },
        {
            "type": "thermostat",
            "name": "RC200/CW100",
            "device id": "0x18",
            "product id": 157,
            "version": "43.02",
            "entities": 42,
            "handlers received": "0x06 0xA2 0x031D",
            "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A",
            "handlers pending": "0xA3 0x12 0x13 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x04",
            "handlers ignored": "0xBF"
        },
        {
            "type": "controller",
            "name": "HT3",
            "device id": "0x09",
            "product id": 95,
            "version": "12.16",
            "entities": 0
        }
    ]
}

```

### TO REPRODUCE

This might be rather complex to explain. I have first noticed this issue on 6 October when most Australian states moved to daylight saving time.
1. the thermostat time did not change, even though I had NTP set up and believe correctly working
2. I tried all possible advice from various previous issues to no avail. When I finally managed change the time, I got flooded with:
```
thermostat time correction from ntp
```
in my log files.

So, I looked at thermostat and it has DLS setting that is supposed to change thermostat time to from daylight saving times automatically and it was set to OFF. After turning it to ON, I somehow managed to get time in thermostat and EMS-ESP same and the messages about correction finally stopped.

However, at the same moment, I have noticed that according to the manual, my CR50 thermostat seems to be hard wired for Europe, since the manual states that DLS option does:
```
Activate or deactivate automatic changeover between summer and winter time.
If ON is set, the time of day is automatically changed from 02:00 to 03:00 (on the last Sunday in March and from 03:00 to 02:00 on
the last Sunday in October).
```
Lo and behold, thermostat changed time one hour back today (last Sunday of October) and I cannot get it back to all synchronised with NTP turned on. If I have it ON, I get again flooded with the messages mentioned above. So, it is currently off.

I think, but did not leave it running long enough to confirm 100%, that NTP is somehow trying to move time to one hour back in the thermostat because after a while it was behind the time of the EMS-ESP. So, it seems there is somewhere incorrectly set daylight saving flag?

Could it be read only "hc1 summer mode (hc1/summermode)" which shows as "winter"? But it has been showing as winter all the time...

### EXPECTED BEHAVIOUR

To be honest, I am not sure what I expect. Ideally, that NTP changes correctly thermostat time in line with EMS-ESP device. Probably with DLS parameter on thermostat in OFF???

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

Happy to provide whatever you need, just please let me know what you need.

**(Please remember to close the issue when it has been addressed)**

VlastiBroucek, 2024-10-27:
I looked at the thermostat:
```
Thermostat: RC200/CW100 (DeviceID:0x18, ProductID:157, Version:43.02) (42)
  date/time (datetime): 27.10.2024 13:18
  damped outdoor temperature (dampedoutdoortemp): -10.0 °C
  dhw mode (wwmode): comfort
  dhw circulation pump mode (wwcircmode): auto
  dhw charge duration (wwchargeduration): 60
  dhw charge (wwcharge): off
  dhw circuit 1 extra (wwextra1): 0 °C
  dhw disinfecting (wwdisinfecting): off
  dhw disinfection day (wwdisinfectday): tu
  dhw disinfection time (wwdisinfecttime): 120
  hc1 selected room temperature (seltemp): 21.0 °C
  hc1 current room temperature (currtemp): 22.7 °C
  hc1 mqtt discovery current room temperature (haclimate): roomTemp
  hc1 mode (mode): auto
  hc1 mode type (modetype): comfort
  hc1 eco temperature (ecotemp): 18.5 °C
  hc1 manual temperature (manualtemp): 26.0 °C
  hc1 comfort temperature (comforttemp): 21.0 °C
  hc1 summer temperature (summertemp): 17 °C
  hc1 design temperature (designtemp): 45 °C
  hc1 offset temperature (offsettemp): 0 °C
  hc1 min flow temperature (minflowtemp): 25 °C
  hc1 max flow temperature (maxflowtemp): 54 °C
  hc1 room influence (roominfluence): 3 °C
  hc1 room influence factor (roominflfactor): 4.0
  hc1 current room influence (curroominfl): 0.0 °C
  hc1 nofrost mode (nofrostmode): room
  hc1 nofrost temperature (nofrosttemp): 5 °C
  hc1 target flow temperature (targetflowtemp): 0 °C
  hc1 heating type (heatingtype): floor
  hc1 set summer mode (summersetmode): auto
  hc1 summer mode (summermode): winter
  hc1 control mode (controlmode): room
  hc1 program (program): prog 1
  hc1 temporary set temperature automode (tempautotemp): -1.0 °C
  hc1 fast heatup (fastheatup): 0 %
  hc1 switch-on optimization (switchonoptimization): off
  hc1 reduce mode (reducemode): reduce
  hc1 no reduce below temperature (noreducetemp): -31 °C
  hc1 off/reduce switch temperature (reducetemp): 5 °C
  hc1 cooling (coolingon): off
  hc1 control device (control): RC310
```

and tried to change summer mode from AUTO to SUMMER, using `"hc1 set summer mode (summersetmode):` . However, this did not change it in `hc1 summer mode (summermode): `

MichaelDvP, 2024-10-27:
I don't think it makes a difference for this issue, but using the latest dev is good to find actual issues in code, there are so many changes since 3.6.5.
The summermode is not a clock setting, it switches the heating (not dhw) off for outside temperatures above `summertemp` setting.

I guess the NTP for emsesp is working well and time is set. Only the sync to thermostat is not working. 
I suspect that for this thermostat the time is not writeable via bus (we see this for some thermostats like RC20.
Or writing the clock with dst (daylightsavingtime) bit "wrong" causes the the thermostat clock is not set. But i don't think so, because between 6.10 and 27.10 both timezones have DST set.
Could you enable NTP again and make a log (log all) where we can see the sync try from emsesp, sending/read back  telegram 0x06 and log the message `thermostat time correction from ntp`.
If both clocks are set to same time the automatic sync should not trigger, in this case please start logging and trigger a write by using the web thermostat->datetime and type `NTP`

VlastiBroucek, 2024-10-27:
Hi there,

done as asked and the log file is below. I started with clock on thermostat matching the clock in ESP-HOME and enabled NTP. Then after some time

``000+00:03:29.024 I 552: [command] Calling command 'thermostat/datetime' (date/time) with value NTP``

tried to set clock manually using NTP. Hope this helps. I have done this on PROD (3.6.5) only to notice moments later that 3.7.0 has been released. I can do same exercise after upgrading to 3.7.0, but will hold on until I hear from you.

```
ems-esp:$ read 0x18 0x06
000+00:00:54.819 N 1: [emsesp] thermostat(0x18) -W-> Me(0x0B), RCTime(0x06), data: 18 0A 09 1C 24 20 00 00 18 00 00
ems-esp:$ log all
Log level: ALL
000+00:01:58.616 I 2: [emsesp] Starting NTP service
000+00:02:00.293 T 3: [emsesp] Me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 0x1B
000+00:02:00.319 T 4: [emsesp] boiler(0x08) -W-> Me(0x0B), UBATotalUptime(0x14), data: 15 7B A0
000+00:02:00.440 T 5: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 02 2D 00 00 00 22 40 C0 01 7E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:02:00.648 T 6: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 3C 01 7E 80 00 81 00 00 01 00 00 9B 26 00 9F BC 00
000+00:02:00.918 T 7: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 0x1B
000+00:02:00.960 T 8: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 36 54 00 00 F6 03 01 03 64 0A 04
000+00:02:01.102 T 9: [emsesp] Me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 0x1B
000+00:02:01.111 T 10: [emsesp] boiler(0x08) -W-> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C 00 00 00 00 02 46 00 FF FF
000+00:02:01.231 T 11: [emsesp] Me(0x0B) -R-> boiler(0x08), UBASettingsWW(0x26), length: 0x1B
000+00:02:01.253 T 12: [emsesp] boiler(0x08) -W-> Me(0x0B), UBASettingsWW(0x26), data: <empty>
000+00:02:01.405 T 13: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Monitor(0x02A5), length: 0x19
000+00:02:01.523 T 14: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 E0 21 2A 00 00 2A 25 02 AB 03 03 01 02 AB 01 15 00 00 11 01 03 08 C1 00
000+00:02:01.640 T 15: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Monitor(0x02A5), length: 0x19 (offset 25)
000+00:02:01.708 T 16: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Monitor(0x02A5), data: 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00 (offset 25)
000+00:02:01.811 T 17: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Set(0x02B9), length: 0x19
000+00:02:01.907 T 18: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Set(0x02B9), data: FF 2E 2A 26 25 03 00 FF FF 05 34 01 E1 20 01 0F 05
000+00:02:02.046 T 19: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Summer(0x02AF), length: 0x19
000+00:02:02.104 T 20: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Summer(0x02AF), data: 03 28 00 00 4B 2D 11 01 19 1E 00
000+00:02:02.265 T 21: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Curves(0x029B), length: 0x19
000+00:02:02.312 T 22: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Curves(0x029B), data: 04 03 00 00 00 01 05 36 30
000+00:02:02.457 T 23: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Summer2(0x0471), length: 0x19
000+00:02:02.484 T 24: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Summer2(0x0471), data: <empty>
000+00:02:02.592 T 25: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300Set2(0x02CC), length: 0x19
000+00:02:02.636 T 26: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300Set2(0x02CC), data: <empty>
000+00:02:02.797 T 27: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPSet(0x0467), length: 0x19
000+00:02:02.820 T 28: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPSet(0x0467), data: <empty>
000+00:02:02.957 T 29: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPMode(0x0291), length: 0x19
000+00:02:02.990 T 30: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPMode(0x0291), data: 01 00 00 00
000+00:02:03.108 T 31: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPMode(0x0292), length: 0x19
000+00:02:03.131 T 32: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPMode(0x0292), data: <empty>
000+00:02:03.264 T 33: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPMode(0x0293), length: 0x19
000+00:02:03.291 T 34: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPMode(0x0293), data: <empty>
000+00:02:03.457 T 35: [emsesp] Me(0x0B) -R-> thermostat(0x18), HPMode(0x0294), length: 0x19
000+00:02:03.482 T 36: [emsesp] thermostat(0x18) -W-> Me(0x0B), HPMode(0x0294), data: <empty>
000+00:02:03.608 T 37: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300WWmode(0x02F5), length: 0x19
000+00:02:03.669 T 38: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300WWmode(0x02F5), data: 01 00 02 02 00 00 08 01 00 08 04 00
000+00:02:03.742 T 39: [emsesp] thermostat(0x18) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+00:02:03.840 T 40: [emsesp] thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 00 E0 21 2A 00 00 2A 25 02 AB 03 03 01 02 AB 01 15 00 00 11 01 03
000+00:02:04.080 T 41: [emsesp] thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 08 C1 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00 (offset 22)
000+00:02:04.326 T 42: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300WWtemp(0x031B), length: 0x19
000+00:02:04.353 T 43: [emsesp] thermostat(0x18) -W-> Me(0x0B), RC300WWtemp(0x031B), data: <empty>
000+00:02:04.436 T 44: [emsesp] thermostat(0x18) -B-> All(0x00), RC300WWmode2(0x031D), data: 00 00 05 07
000+00:02:04.706 T 45: [emsesp] Me(0x0B) -R-> thermostat(0x18), RC300OutdoorTemp(0x023A), length: 0x19
[... cut ...]
000+00:05:15.661 T 1247: [emsesp] thermostat(0x18) -B-> All(0x00), RC300Monitor(0x02A5), data: 02 A9 (offset 13)
000+00:05:17.334 T 1248: [emsesp] thermostat(0x18) -W-> boiler(0x08), ?(0x23), data: 00 00 00
000+00:05:20.340 T 1249: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 00 02 28 00 00 00 22 40 C0 01 77 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:05:20.549 T 1250: [emsesp] boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 3C 01 76 80 00 81 00 00 01 00 00 9B 26 00 9F BC 00
```

VlastiBroucek, 2024-10-27:
I have now updated to 3.7.0 and the behaviour is the same.

VlastiBroucek, 2024-10-29:
I did little bit more testing today, trying to find out where the DLS setting in thermostat is stored. I am thinking it is in 0x06 at offset 9 - 00 means tit is OFF, FF means it is ON. But still not sure how it affects NTP sync... And why on Earth is Bosch selling in Australia something that has European settings only...
[image]

Anyway, there is the log, I am guessing that the change is reflected at ``001+02:33:06.460 T 37:``. Later I tried to activate NTP again and got flooded by errors.
```
ems-esp:$ log all
Log level: ALL
001+02:32:39.870 T 1: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 88 00 00 00 22 40 C0 01 E0 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+02:32:40.170 T 2: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 E1 80 00 A1 00 00 01 00 00 9B 43 00 9F D1 00
001+02:32:49.862 T 3: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 87 00 00 00 22 40 C0 01 E0 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+02:32:50.170 T 4: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 E0 80 00 A1 00 00 01 00 00 9B 43 00 9F D1 00
001+02:32:52.963 T 5: [emsesp] thermostat(0x18) -B-> all(0x00), RCTime(0x06), data: 18 0A 0D 1D 27 39 01 00 18 00 00
001+02:32:57.938 T 6: [emsesp] thermostat(0x18) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
001+02:32:58.035 T 7: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 00 E6 21 2A 00 00 2A 25 01 B9 03 03 01 01 B9 02 07 00 00 11 01 03
001+02:32:58.276 T 8: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 08 FF 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00 (offset 22)
001+02:32:58.633 T 9: [emsesp] thermostat(0x18) -B-> all(0x00), RC300WWmode2(0x031D), data: 00 00 05 07
001+02:32:58.833 T 10: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 01 B8 02 08 (offset 13)
001+02:32:59.162 T 11: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 01 B8 (offset 8)
001+02:32:59.818 T 12: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 87 00 00 00 22 40 C0 01 E0 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+02:33:00.170 T 13: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 DF 80 00 A1 00 00 01 00 00 9B 43 00 9F D1 00
001+02:33:01.858 T 14: [emsesp] me(0x0B) -R-> boiler(0x08), UBATotalUptime(0x14), length: 27
001+02:33:01.884 T 15: [emsesp] boiler(0x08) -W-> me(0x0B), UBATotalUptime(0x14), data: 15 82 33
001+02:33:01.884 T 16: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameters(0x16), length: 27
001+02:33:01.926 T 17: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameters(0x16), data: FF 36 54 00 00 F6 03 01 03 64 0A 04
001+02:33:01.926 T 18: [emsesp] me(0x0B) -R-> boiler(0x08), UBAParameterWW(0x33), length: 27
001+02:33:01.989 T 19: [emsesp] boiler(0x08) -W-> me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C 00 00 00 00 02 46 00 FF FF
001+02:33:01.989 T 20: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300Monitor(0x02A5), length: 25
001+02:33:02.106 T 21: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300Monitor(0x02A5), data: 00 E6 21 2A 00 00 2A 25 01 B8 03 03 01 01 B8 02 08 00 00 11 01 03 08 FF 00
001+02:33:02.236 T 22: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300Monitor(0x02A5), length: 25 (offset 25)
001+02:33:02.307 T 23: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300Monitor(0x02A5), data: 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00 (offset 25)
001+02:33:02.307 T 24: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300Set(0x02B9), length: 25
001+02:33:02.419 T 25: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300Set(0x02B9), data: FF 2E 2A 26 25 03 00 FF FF 05 34 01 E1 20 01 0F 05
001+02:33:02.419 T 26: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300Summer(0x02AF), length: 25
001+02:33:02.496 T 27: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300Summer(0x02AF), data: 03 28 00 00 4B 2D 11 01 19 1E 00
001+02:33:02.496 T 28: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300Curves(0x029B), length: 25
001+02:33:02.567 T 29: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300Curves(0x029B), data: 04 03 00 00 00 01 05 36 30
001+02:33:02.567 T 30: [emsesp] me(0x0B) -R-> thermostat(0x18), HPMode(0x0291), length: 25
001+02:33:02.615 T 31: [emsesp] thermostat(0x18) -W-> me(0x0B), HPMode(0x0291), data: 01 00 00 00
001+02:33:02.615 T 32: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300WWmode(0x02F5), length: 25
001+02:33:02.672 T 33: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300WWmode(0x02F5), data: 01 00 02 02 00 00 08 01 00 08 04 00
001+02:33:02.672 T 34: [emsesp] me(0x0B) -R-> thermostat(0x18), RC300OutdoorTemp(0x023A), length: 25
001+02:33:02.725 T 35: [emsesp] thermostat(0x18) -W-> me(0x0B), RC300OutdoorTemp(0x023A), data: FF 9C
001+02:33:04.153 T 36: [emsesp] thermostat(0x18) -B-> all(0x00), RCError(0xA2), data: 00 00 00 00 00
001+02:33:06.460 T 37: [emsesp] thermostat(0x18) -B-> all(0x00), RCTime(0x06), data: FF (offset 9)
001+02:33:09.798 T 38: [emsesp] boiler(0x08) -B-> all(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00
001+02:33:10.162 T 39: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 87 00 00 00 22 40 C0 01 DF 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+02:33:10.278 T 40: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 DF 80 00 A1 00 00 01 00 00 9B 43 00 9F D1 00
[... cut ...]
001+02:39:16.176 I 112: [emsesp] Starting NTP service
001+02:39:27.706 I 113: [command] Called command thermostat/datetime (date/time) with value NTP
001+02:39:27.973 I 114: [thermostat] thermostat time correction from ntp
001+02:39:27.973 I 115: [thermostat] thermostat time correction from ntp
001+02:39:28.565 I 116: [thermostat] thermostat time correction from ntp
001+02:39:28.755 I 117: [thermostat] thermostat time correction from ntp

...

001+02:39:45.482 W 183: [system] NTP disconnected
```

VlastiBroucek, 2024-10-29:
Hi @MichaelDvP ,

I think the problem is that there is missing bit 0 in 0x06 telegram at offset 7. That should currently be set to 1 (since there is DST here), but it is not. 

[image]

I tried to create custom entity and set it, but nothing happens. So, it appears to me that whole BYTE is probably read only and only thermostat's hardware itself can set it. That's why I had to activate the DLS option in Termostat when DST started here at the beginning of October and the sync was working until the bit was removed by the thermostat.

I would like to try this by setting the thermostat manually back to second week of October and see if the bit will be set. However, I don't want to mess data in my HA, so I think I would need to shut down MQTT temporarily. Is it safe to just unclick it in  EMS-ESP settings? Won't I lose any settings etc? 

[image]

MichaelDvP, 2024-10-29:
Yes, i also thinnk the problem is the dst setting. Since 6.10. NTP gives dst set and the clock command sets bit 0 in offset 7. But the thermostat clock was set to "no dst" for australia and does not accept the setting command. Now you have set "with dst" but thermostat expects  dst bit not set for 27.10 and later. 
Does thermostat accept a manul setting command from emsesp? In Web devices click the datetime and enter:
`dd.mm.yyyy-hh:mm:ss-day(0-6)-dst(0/1)` with day 0 (mo) and dst 0

I will limit the automatic ntp correction to 3 retries to avoid flooding the bus/log and will try to check offset 9 before setting the clock.

VlastiBroucek, 2024-10-29:
Hi Michael,
yes, the thermostat accepts the manual command in the complex format (as well as using NTP). I tried that earlier today by moving clock back one hour, staring NTP and issuing the command. However, both the manual command and the NTP command do not set the bit at offset 7.

The behaviour is the same no matter whether offset 9 is set to 00 or FF.

MichaelDvP, 2024-10-29:
> The current behaviour is the same no matter whether offset 9 is set to 00 or FF.

Yes, sure, because europe has actual dst=0. I think you should set offset 9 to 0 and send the actual local time with dst=0. If local time is ok, the correction from NTP will not trigger.

VlastiBroucek, 2024-10-29:
I've offset 9 at 00. I've set local clock manually using dst=0. And made sure I sent the command so precisely that ems-esp clock and thermostat clock are nearly perfectly same. If I start NTP, I start getting the messages after something like 30 seconds and they never stop.

I'll play with this bit more tomorrow (I'm already in bed now) and I'll let you know.

MichaelDvP, 2024-10-29:
I've compiled your branch and flashed it on a spare esp32s3 (env: s3_16M_p). No errors, running well. Also if i compile my branch.

What hardware do you use? Connection wifi or eth? Any changes to board profile or pins? Flashed the right firmware?

MichaelDvP, 2024-10-29:
Try with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

VlastiBroucek, 2024-10-29:
> Try with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Installed and tried - still the same issue:

```
vlasti@VlastiHP:~$ telnet ems-esp.modem
Trying 192.168.0.35...
Connected to ems-esp.modem.
Escape character is '^]'.

┌───────────────────────────────────────┐
│  EMS-ESP version 3.7.1-dev.1          │
│                                       │
│  help to show available commands      │
│  su to access admin commands          │
│                                       │
│  https://github.com/emsesp/EMS-ESP32  │
│                                       │
└───────────────────────────────────────┘

000+00:00:38.023 I 0: [shell] User session opened on console pty0
000+00:00:46.134 I 1: [emsesp] Starting NTP service
ems-esp:$ read 0x18 0x06
000+00:01:37.151 N 2: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 0A 1E 1A 11 02 00 18 00 00
000+00:01:37.151 I 3: [thermostat] thermostat time correction from ntp
000+00:01:37.516 I 4: [thermostat] thermostat time correction from ntp
000+00:01:37.860 I 5: [thermostat] thermostat time correction from ntp
000+00:01:38.235 I 6: [thermostat] thermostat time correction from ntp
000+00:01:38.235 I 7: [thermostat] thermostat time correction from ntp
000+00:01:38.818 I 8: [thermostat] thermostat time correction from ntp
000+00:01:39.041 I 9: [thermostat] thermostat time correction from ntp
000+00:01:39.273 I 10: [thermostat] thermostat time correction from ntp
000+00:01:39.480 I 11: [thermostat] thermostat time correction from ntp
000+00:01:39.480 I 12: [thermostat] thermostat time correction from ntp
000+00:01:40.215 I 13: [thermostat] thermostat time correction from ntp
000+00:01:40.215 I 14: [thermostat] thermostat time correction from ntp
000+00:01:40.215 I 15: [thermostat] thermostat time correction from ntp
000+00:01:40.917 I 16: [thermostat] thermostat time correction from ntp
000+00:01:41.141 I 17: [thermostat] thermostat time correction from ntp
000+00:01:41.141 I 18: [thermostat] thermostat time correction from ntp
000+00:01:41.731 I 19: [thermostat] thermostat time correction from ntp
000+00:01:41.731 I 20: [thermostat] thermostat time correction from ntp
000+00:01:42.171 I 21: [thermostat] thermostat time correction from ntp
000+00:01:42.171 I 22: [thermostat] thermostat time correction from ntp
000+00:01:42.537 I 23: [thermostat] thermostat time correction from ntp
000+00:01:42.977 I 24: [thermostat] thermostat time correction from ntp
000+00:01:42.977 I 25: [thermostat] thermostat time correction from ntp
000+00:01:43.416 I 26: [thermostat] thermostat time correction from ntp
000+00:01:43.416 I 27: [thermostat] thermostat time correction from ntp
000+00:01:43.416 I 28: [thermostat] thermostat time correction from ntp
000+00:01:44.443 I 29: [thermostat] thermostat time correction from ntp
000+00:01:44.443 I 30: [thermostat] thermostat time correction from ntp
```

MichaelDvP, 2024-10-30:
Please add a full log to see the telegrams for the time read/set. Also a reference what is actual ntp time. in the system.
Sure the difference trigger every time, but the setting and logging should only happen 3 times, the counter is only resetted if difference does not trigger. Is the clock setback after some time? 
I have to read how mktime/difftime handles the time if only dst flag is manipulated.

VlastiBroucek, 2024-10-30:
Hi Michael,

Both times are roughly same, NTP is switched off

[image]

[image]

now I have switched NTP on and after while off. Hope this is long enough log. 

```
ems-esp:$ log all
Log level: ALL
000+08:25:33.194 T 1: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 77 00 00 00 22 40 C0 01 C6 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+08:25:33.386 T 2: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 C6 80 00 A1 00 00 01 00 00 9B 63 00 9F E8 00
000+08:25:43.194 T 3: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 76 00 00 00 22 40 C0 01 C6 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+08:25:43.381 T 4: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 C5 80 00 A1 00 00 01 00 00 9B 63 00 9F E8 00
000+08:25:44.357 T 5: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 03 3E (offset 15)
000+08:25:44.548 T 6: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 00 82 (offset 13)
000+08:25:44.890 T 7: [emsesp] thermostat(0x18) -B-> all(0x00), RC300Monitor(0x02A5), data: 00 82 (offset 8)
000+08:25:48.256 T 8: [emsesp] thermostat(0x18) -W-> boiler(0x08), ?(0x23), data: 00 00 00
000+08:25:48.592 I 9: [emsesp] Starting NTP service
000+08:25:53.154 T 10: [emsesp] boiler(0x08) -B-> all(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00
000+08:25:53.385 T 11: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 76 00 00 00 22 40 C0 01 C5 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+08:25:53.594 T 12: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 C5 80 00 A1 00 00 01 00 00 9B 63 00 9F E8 00
000+08:25:53.824 T 13: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 00 00 B9 62 02 87 0C 00 00 00 01 EB A8 00 19 7A 80 00
000+08:26:03.160 T 14: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 76 00 00 00 22 40 C0 01 C5 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+08:26:03.369 T 15: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 C5 80 00 A1 00 00 01 00 00 9B 63 00 9F E8 00
000+08:26:04.271 T 16: [emsesp] thermostat(0x18) -B-> all(0x00), RCTime(0x06), data: 18 0A 12 1E 32 15 02 00 18 00 00
000+08:26:04.271 I 17: [thermostat] thermostat time correction from ntp
000+08:26:04.496 T 18: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 33 02 03
000+08:26:04.675 T 19: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:04.719 T 20: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 15 02 00
000+08:26:04.719 I 21: [thermostat] thermostat time correction from ntp
000+08:26:04.719 T 22: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 33 02 03
000+08:26:05.014 T 23: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:05.055 T 24: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 15 02 00
000+08:26:05.055 I 25: [thermostat] thermostat time correction from ntp
000+08:26:05.055 T 26: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 34 02 03
000+08:26:05.357 T 27: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:05.398 T 28: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 16 02 00
000+08:26:05.398 I 29: [thermostat] thermostat time correction from ntp
000+08:26:05.398 T 30: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 34 02 03
000+08:26:05.644 T 31: [emsesp] thermostat(0x18) -B-> all(0x00), RCTime(0x06), data: 18 0A 12 1E 32 34 02 00 18 00 00
000+08:26:05.644 I 32: [thermostat] thermostat time correction from ntp
000+08:26:05.870 T 33: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 34 02 03
000+08:26:06.045 T 34: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:06.085 T 35: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 34 02 00
000+08:26:06.085 I 36: [thermostat] thermostat time correction from ntp
000+08:26:06.085 T 37: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:06.085 T 38: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 34 02 00
000+08:26:06.085 I 39: [thermostat] thermostat time correction from ntp
000+08:26:06.085 T 40: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 35 02 03
000+08:26:06.085 T 41: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 35 02 03
000+08:26:06.085 T 42: [emsesp] thermostat(0x18) -B-> all(0x00), RCTime(0x06), data: 18 0A 12 1E 32 35 02 00 18 00 00
000+08:26:06.085 I 43: [thermostat] thermostat time correction from ntp
000+08:26:06.651 T 44: [emsesp] me(0x0B) -W-> thermostat(0x18), RCTime(0x06), data: 18 0A 12 1E 32 35 02 03
000+08:26:06.794 T 45: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:06.835 T 46: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 35 02 00
000+08:26:06.835 I 47: [thermostat] thermostat time correction from ntp
000+08:26:06.835 T 48: [emsesp] me(0x0B) -R-> thermostat(0x18), RCTime(0x06), length: 8
000+08:26:06.835 T 49: [emsesp] thermostat(0x18) -W-> me(0x0B), RCTime(0x06), data: 18 0A 12 1E 32 35 02 00
000+08:26:06.835 I 50: [thermostat] thermostat time correction from ntp
[... cut ...]
000+08:26:33.180 T 425: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 00 02 74 00 00 00 22 40 C0 01 C2 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+08:26:33.421 T 426: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 3C 01 C2 80 00 A1 00 00 01 00 00 9B 63 00 9F E8 00
ems-esp:$ log info
Log level: INFO
ems-esp:$
```

Since ``static uint8_t setTimeRetry = 0;`` is local to the ``void Thermostat::process_RCTime(std::shared_ptr<const Telegram> telegram)`` doesn't it get zeroed every time when telegram 0x06 is processed?

MichaelDvP, 2024-10-30:
Sure that you have flashed the software from my GH https://github.com/MichaelDvP/EMS-ESP32/releases and not the one from ems-esp32? Clock reports offset 9 as 0, then my software should send the correction withot dst set, but here it sends with dst.

VlastiBroucek, 2024-10-30:
Just a little thought without knowing  how exactly you do all the time syncing. But how about syncing ems-esp with NTP and thermostat with local ems-esp instead of with NTP?

VlastiBroucek, 2024-10-30:
> Sure that you have flashed the software from my GH https://github.com/MichaelDvP/EMS-ESP32/releases and not the one from ems-esp32? Clock reports offset 9 as 0, then my software should send the correction withot dst set, but here it sends with dst.

### Bugger, now you have nailed it!!!

My BAD!!!!! I indeed downloaded 3.7.1-dev.1 from ems-esp32 not from your repo... Misse3d that little difference. I am very sorry!!!

Seems that it is now working!!! Started NTP, got few of those  ``thermostat time correction from ntp`` and now nothing and it seems to be all nice and happy!!!

### Thanks!!!

proddy, 2024-10-30:
Merged into 3.7.1-dev. Thanks Michael.

VlastiBroucek, 2024-10-31:
Thanks for sorting this out!

closed 2024-10-31 by VlastiBroucek