emsesp/EMS-ESP32 #2151: Web update from 3.6.5 to 3.7.0 on E32 V2 defaults to S32 board profile

bbqkees, 2024-10-28:
I got several emails already from users with an E32 V2 who updated from 3.6.5 to 3.7.0 release.
After upload and the reboot, the Ethernet port does not work anymore, so they think they bricked the device.
I checked it and can reproduce it: when you try to update from 3.6.5 to 3.7.0 via the web interface, it points to the [ESP32.bin](https://github.com/emsesp/EMS-ESP32/releases/download/v3.7.0/EMS-ESP-3_7_0-ESP32.bin) instead of the ESP32-16MB+.bin. At reboot the board profile is incorrectly set to S32.

This will disabled Ethernet and also f.i. the reset button does not work anymore.

proddy, 2024-10-28:
damn, I thought we tested this all. I'll take a look. Maybe its a simple fix by including different named files in the 3.7.0 GH assets

proddy, 2024-10-28:
as a quick fix I made the EMS-ESP-3_7_0-ESP32.bin the same as the 16MB+ (with PSRAM) so that will help E32 users automatically upgrade.

But it will crash any older S32 4MB users. Need to look into what is causing that.

(the change was part of https://github.com/emsesp/EMS-ESP32/pull/1941 )

proddy, 2024-10-28:
I can't find an easy way to fix this. I'll remove the binaries and people will need to manually upgrade using the table on the BBQKees wiki.

proddy, 2024-10-28:
won't fix.

bbqkees, 2024-10-28:
Maybe for future updates have EMS-ESP check a special page like updates.emsesp.org. There you could run a check or a table and load the correct bin file for each device. So a relay-page.

proddy, 2024-10-28:
I'll think of another way to detect an upgrade from 3.6.5 to 3.7.0, which bypasses the hardware detection. Maybe that will fix it. I spent 3 hours trying to get around the problem, it isn't trivial. The problem is that E32V2 came out after v3.6.5 and before we had the newer variations (like PSRAM) with different partition tables. The Firmware download check in 3.6.5 is quite simple and doesn't take into account chip-sets for example. 

The logic in 3.7.0 is better. No need for a relay-page anymore.

proddy, 2024-10-29:
redirect users to download.emsesp.org (when it's up) for the time being

closed 2024-10-29 by proddy