emsesp/EMS-ESP32 #2152: Update to Development Build v3.7.1-dev.1: Undefined behavior

NongJoWo, 2024-10-28:
I have just updated to the 3.7.1 dev.1 version and at first glance everything looked fine. Now I've noticed that the upload/download and System Memory window always says "loading". A Roobot isn't possible either. It says "not found". A roobot with power off didn't help either. What can I do? Many thanks!

[image]

[image]

[image]

NongJoWo, 2024-10-28:
That was the file I have used: EMS-ESP-3_7_1-dev_1-ESP32_S3.bin

proddy, 2024-10-28:
which version did you upgrade from?

NongJoWo, 2024-10-28:
3.7.0-dev.17

proddy, 2024-10-28:
ctrl-R to refresh browser cache?

NongJoWo, 2024-10-28:
Yes, that was the solution. It can be so easy ;-)) Thank you for the quick support!

NongJoWo, 2024-10-28:
But I am not sure if that is normal in log file

[image]

proddy, 2024-10-29:
some errors are normal, its due to noise on the EMS bus line. What's important is the quality - check the Rx/Tx quality in the Data Traffic page. Anything less than 100% is a concern

NongJoWo, 2024-10-29:
Rx/Tx quality in the Data Traffic page all by 100%.
What I am struggling now with are some sensos:

Name: ems-esp Boiler Sollwert Vorlauftemperatur
sensor.boiler_setflowtemp: Not available 
(Is used in my dasboard to display target flow temperature of heating)

FYI:
sensor.boiler_setburnpow: Not available
sensor.boiler_wwsetpumppower: Not available

NongJoWo, 2024-10-29:
Never touch a running system :-((

The Version 3.7.0-dev.17 was running without any issues on my system. Does it make sense to donwgrade? If yes where can I find it?

NongJoWo, 2024-10-29:
All ESP MQTT devices in HA deleted an ESP rebooted. Now all entities are there without unavailables. 
But I am not sure which entity I can now use for "target flow temperature of heating"

I could only find 2 entities with "flowtemp"
- number.boiler_selflowtemp
- sensor.thermostat_hc1_targetflowtemp

Anyone an idea?

proddy, 2024-10-29:
can you send over the System Info so we can see the system, and help you. Don't know which thermostat you have, or whether its a boiler, heatsource, mixer, heatpump etc...

NongJoWo, 2024-10-29:
Buderus GB172-24K with RC310 thermostat installed inside the boiler.

sensor.thermostat_hc1_targetflowtemp
Seems to provide useful data.

NongJoWo, 2024-10-29:
The name was “sensor.boiler_setflowtemp”.
I have just compared it with “sensor.thermostat_hc1_targetflowtemp” and both had the same values in the past.

MichaelDvP, 2024-10-30:
We have removes some dublicate values, setflowtemp was the same value as selflowtemp, but received from a different telegram. (setflowtemp: command to the boiler, selflowtemp: value stored in the boiler) Same for setburnpower and some more.
And yes, thermostat targetflowtemp is the same value sent from thermostat to the boiler.

NongJoWo, 2024-10-30:
Many thanks for confirmation. Everything looks ok now…

closed 2024-10-31 by proddy