emsesp/EMS-ESP32 #2153: Junkers ZSBE w. FW120  doenst show datetime in HA any more

g6094199, 2024-10-29:
### DESCRIPTION

datetime was shown some time but has vanished.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0-dev.48","uptime":"004+02:49:00.516","uptimeSec":355740,"platform":"ESP32","cpuType":"ESP32-D0WDQ6","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":166,"maxAlloc":107,"freeCaps":116,"usedApp":1730,"freeApp":254,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"Junkers-Heatronic-WDB","RSSI":-70,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"1.de.pool.ntp.org","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":113336,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":5,"enabled":true,"clientID":"junkers-heatronic-wdb","keepAlive":60,"cleanSession":false,"entityFormat":4,"base":"junkers-heatronic-wdb","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":285212,"busReads":59110,"busWrites":6,"busIncompleteTelegrams":2,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"CUSTOM","locale":"de","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"phyType":0,"rxGPIO":23,"txGPIO":22,"dallasGPIO":18,"pbuttonGPIO":0,"ledGPIO":2,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":23,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"18.11","entities":68,"handlersReceived":"0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x16 0x33","handlersPending":"0xBF 0xC2 0x1A 0x35 0x26 0x2E 0x3B","handlersIgnored":"0x23 0x22 0x03"},{"type":"thermostat","name":"FW120","deviceID":"0x10","productID":192,"brand":"","version":"53.02","entities":24,"handlersReceived":"0x06","handlersFetched":"0x016F 0x0165 0x01D3","handlersPending":"0xA3 0xA2 0x12 0x13 0x0170 0x0171 0x0172 0x0166 0x0167 0x0168 0xBB 0x23","handlersIgnored":"0xBE 0x0128 0xC3 0x011E 0x01C9"},{"type":"solar","name":"ISM1","deviceID":"0x30","productID":101,"brand":"","version":"23.04","entities":8,"handlersFetched":"0x0103 0x0101","handlersPending":"0x0104","handlersIgnored":"0xBE"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"18.11","entities":0,"handlersIgnored":"0x29"}]}
```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS

[image]

[image]

### ADDITIONAL CONTEXT

proddy, 2024-10-29:
please provide more information. When did you last see it? What happened in between?

g6094199, 2024-10-29:
> please provide more information. When did you last see it? 
i dont know. this is a info one doesnt recognize very often. this ems-esp runs roughly for a year now
>What happened in between?
many many HA updates, many ems.esp updates.

what info can i get u?

g6094199, 2024-10-30:
> `read 10 6`

needed some thinking to get that there is a telnet console..... ;-)

`2024-10-30 11:41:04.649 I 70792: [telnet] New connection from [192.168.1.240]:51276 accepted
2024-10-30 11:41:04.649 I 70793: [shell] Allocated console pty0 for connection from [192.168.1.240]:51276
2024-10-30 11:41:04.649 I 70794: [shell] User session opened on console pty0
2024-10-30 11:41:13.281 T 70795: [emsesp] me(0x0B) -R-> thermostat(0x10), RCTime(0x06), length: 27
2024-10-30 11:41:13.331 N 70796: [emsesp] thermostat(0x10) -W-> me(0x0B), RCTime(0x06), data: 18 0A 0B 1E 2A 2E 03 00
2024-10-30 11:41:13.476 T 70797: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 24 02 61 64 00 01 03 00 C0 80 00 80 00 02 01 FF FF 1B 00 00 00 00 00 00 00
2024-10-30 11:41:13.717 T 70798: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 28 01 DC 01 DC A1 00 04 03 00 02 75 BF 00 00 05 00
2024-10-30 11:41:19.733 T 70799: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBAFlags(0x35), data: 11 00 28 (offset 1)
2024-10-30 11:41:19.761 T 70800: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 24
2024-10-30 11:41:19.794 T 70801: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
2024-10-30 11:41:19.831 T 70802: [emsesp] thermostat(0x10) -W-> boiler(0x08), JunkersSetMixer(0x23), data: 24 64 00
2024-10-30 11:41:19.860 T 70803: [emsesp] thermostat(0x10) -W-> boiler(0x08), UBASetPoints(0x1A), data: 03 (offset 4)
2024-10-30 11:41:20.709 T 70804: [emsesp] solar(0x30) -B-> all(0x00), ISM1StatusMessage(0x0103), data: 07 0C 04 EF 00 F5 00 C7 00 00 07 10 34
2024-10-30 11:41:23.427 T 70805: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 24 02 60 64 00 01 03 00 C0 80 00 80 00 01 FF FF FF 1B 00 00 00 00 00 00 00
2024-10-30 11:41:23.637 T 70806: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorWW(0x34), data: 28 01 DC 01 DC A1 00 04 03 00 02 75 BF 00 00 05 00
2024-10-30 11:41:24.463 T 70807: [emsesp] boiler(0x08) -B-> all(0x00), UBAMonitorFast(0x18), data: 24 02`

proddy, 2024-10-31:
Sorry, just realised that the date/time is shown in EMS-ESP, but not HA. Is that correct? If so, the best place to look is the Home Assistant logs. And also use MQTTExplorer or something similar to look at the values in the MQTT topics

`homeassistant/sensor/ems-esp/thermostat_datetime/config` and `thermostat_data`

You can also use HA's Developer Tools->States to inspect these.

g6094199, 2024-10-31:
mqtt sees the date. but is it correct?

[image]

proddy, 2024-10-31:
Looks ok. datetime is a sensor and not text. You may have old and new values in home assistant left over from previous installs. I would remove everything in MQTT under 'homeassistant' using MQTT Explorer, restart EMS-ESP and update your dashboards. If you want to see  if the values are correct in HA use Settings->Devices&Services->MQTT

g6094199, 2024-10-31:
hmm.. did it but no enhancement.

HA still sees it as text and NOT as sensor.

proddy, 2024-10-31:
my mistake, it will  be Text if it's an EMS-ESP command, i.e. the date/time entity is writable and can be set. Which is true for some thermostats. If it shows as unavailable in HA, click on it and trace it to the source. The data is there somewhere

g6094199, 2024-11-01:
> my mistake, it will be Text if it's an EMS-ESP command, i.e. the date/time entity is writable and can be set. Which is true for some thermostats. 

yes, e.g. for my Junkers FW120

>If it shows as unavailable in HA, click on it and trace it to the source. The data is there somewhere

`entity: climate.junkers_heatronic_wdb_thermostat_hc1`

not available.

the entity changed to

` entity: sensor.junkers_heatronic_wdb_thermostat_datetime`

but this one also is not available, but the only one thats detected. bug in HA?

[image]

proddy, 2024-11-01:
the datetime is there somewhere, I thnk you're just picking the wrong data. Don't use the climate entity as this is a special one for HA only. In HA, go to Developer tools->States, type in `datetime` and post a screenshot of the output.

this is mine:

[image]

g6094199, 2024-11-01:
nope

[image]