emsesp/EMS-ESP32 #2166: Unexpected Application Error in 3.7.1-dev

tp1de, 2024-11-02:
Loading new dev versions  of 3.7.1 I get the following error while selecting devices / customizations:

[image]

tp1de, 2024-11-02:
When re-starting the device-list works for 1-2 minutes. Afterwards the error occurs.

tp1de, 2024-11-03:
This error co-exists with api errors on .../api/system. API command works for 1-2 minutes afterwards not anymore. 
Same as for web-ui errors on devices.

The error occurs on 3.7.1-dev.1 onwards.
@proddy Please have a look at api response for mixer mm100. This might be the cause of errors.

[API.txt](https://github.com/user-attachments/files/17609920/API.txt)

MichaelDvP, 2024-11-03:
Seems to be the name of the MM100 from telegram 1, can you give us the output of `read 21 1`

tp1de, 2024-11-03:
ems-esp:$ read 21 1
000+00:03:10.932 N 3: [emsesp] mixer(0x21) -W-> me(0x0B), ?(0x01), data: 32 35 33 30 38 35 34 30 30 31 37 36 39 37 37 33 38 31 31 30 31 31 34 00 FF FF FF
000+00:03:11.001 N 4: [emsesp] mixer(0x21) -W-> me(0x0B), ?(0x01), data: 4D 04 1F 1E 1E 00 00 00 00 00 00 00 00 48 00 30 (offset 27)
ems-esp:$

MichaelDvP, 2024-11-03:
Seems offset 27 is not always a null terminated ascii name.

MichaelDvP, 2024-11-03:
Try to set customizations custom name for MM100.

tp1de, 2024-11-03:
> Try to set customizations custom name for MM100.

Yes this works.

proddy, 2024-11-03:
seems so, well spot Michael. I hate that Bosch is so inconsistent. Another user (in #2073) with an MM100 shows the text "MM100" in the telegram 0x01 at offset 27. I haven't been able to decipher the logic of that telegram yet.

MichaelDvP, 2024-11-03:
@tp1de Please update to my testbuild and check if it works without custom name. (I added a check for printable characters). 
Also the thermostat timer-json should work again.

tp1de, 2024-11-03:
yes it works with new testbuild but name is now M/M100 which does not make a lot of sense.
... and I recognized that Dashboard does not show entities anymore (maybe I switched too often between Versions)

tp1de, 2024-11-03:
Update: Dashboard seems to be related to FireFox / Browser Update. Edge / Chrome shows the dashboard correctly

proddy, 2024-11-03:
Just downloaded and installed Firefox 132.0, and Dashboard works

MichaelDvP, 2024-11-04:
I always use firefox and have seen something like that often. It's a cache issue and ctrl-R sometimes only helps short. Clear website data and cache for this device.

MichaelDvP, 2024-11-04:
> yes it works with new testbuild but name is now M/M100 which does not make a lot of sense.

This is what is written in your MM100 telegram 01: only a singel `M`, then EOT marker (ascii 04). Maybe we should ignore the 01-telegram name if it is not a null-terminated string?  On the other hand this gives info about different versions of device, some users have communication issues special with MM100 and emsesp, for others it works.

tp1de, 2024-11-04:
> I always use firefox and have seen something like that often. It's a cache issue and ctrl-R sometimes only helps short. Clear website data and cache for this device.

I was running the debugger (F12) and I have seen that there are http get commands every second. 
Within reply there are JSON-errors as well. I do not know if this has an impact.

tp1de, 2024-11-04:
> This is what is written in your MM100 telegram 01: only a singel `M`, then EOT marker (ascii 04). Maybe we should ignore the 01-telegram name if it is not a null-terminated string? On the other hand [...]

It is up to you. I haven't really understood why you intend to read the name out of the telegram.

Within Bosch-Group products there are many similar products with different names and firmware releases.
I believe tat the productID is sometimes the same for different brands - just the product sticker changes.
Since the firmware is not flashable / upgradeable there are a lot of active versions existing.

proddy, 2024-11-04:
> I was running the debugger (F12) and I have seen that there are http get commands every second. Within reply there are JSON-errors as well. I do not know if this has an impact.

Can you post some screenshots?

tp1de, 2024-11-04:
[image]

tp1de, 2024-11-04:
[image]

proddy, 2024-11-04:
thanks, the page should be refreshing every 3 seconds. Maybe that is too much. I also see the 

content-type in the header is set to `application/json` which is incorrect as we're using binary compressed with msgpack for the response payload. I'll fix that.

MichaelDvP, 2024-11-04:
I've checked the json error, it affects all msgpack sendings since a very long time and mny emsesp versions (devcies->deviceData, Customization). 
Setting in AsyncJson.h: AsyncJsonResponse `_contentType = isMsgPack ? MSGPACK_MIMETYPE : JSON_MIMETYPE;` 
with mimetype `application/msgpack` solves this. But appart from the error message in analyzter i can't see issues in webpage.
Also the 3 sec refresh works well with my really slow laptop, also while compiling emsesp in the background.

proddy, 2024-11-04:
yes, it's harmless but I changed it anyway. I had already implemented it the same way you suggested before seeing your comment. Great minds think alike !

proddy, 2024-11-04:
> This is what is written in your MM100 telegram 01: only a singel `M`, then EOT marker (ascii 04). Maybe we should ignore the 01-telegram name if it is not a null-terminated string? On the other hand [...]

We could read the bytes from offset 27 until we hit a non-ASCII value, and then null-terminate the string. And ignore anything that is less than 3 characters long (like just "M").

MichaelDvP, 2024-11-04:
> We could read the bytes from offset 27 until we hit a non-ASCII value, and then null-terminate the string. And ignore anything that is less than 3 characters long (like just "M").

Done. In my dev2-testbuild.

tp1de, 2024-11-04:
This solution works good

tp1de, 2024-11-04:
@MichaelDvP something wron on your testbuild with mqtt:

[image]

MichaelDvP, 2024-11-04:
Can not reproduce and have not changed anything in mqtt code. Collision with a second mqtt client? 
Have you tried restarting mqtt broker?

proddy, 2024-11-04:
using the same MQTT clientID perhaps?

proddy, 2024-11-04:
on the same hardware or are you running multiple instances?

MichaelDvP, 2024-11-04:
> nd repeats the last mqtt commands every time.

The mqtt code is not changed, i've updated/merged latest changes to dev/dev2 and compared. 
What commands do you see? Maybe it's a change in commands?

MichaelDvP, 2024-11-05:
@tp1de The mqtt in my testbuild should work now, Was in HA config for the timer values, the default case results in a scrambled topic for the timer.

tp1de, 2024-11-05:
> @tp1de The mqtt in my testbuild should work now, Was in HA config for the timer values, the default case results in a scrambled topic for the timer.

Yes I can confirm it is working now.
I just added #2179 for a bug on ems-esp start. This is a general topic - not just for your testbuild.

proddy, 2024-11-05:
great, we can close this one now then.

closed 2024-11-05 by proddy