emsesp/EMS-ESP32 #2181: Switch boiler_wwonetime doesn't work

jimmyhenne, 2024-11-05:
### DESCRIPTION

I have a E32 V2 Gateway and recently updated from 3.6.5 to 3.7.0 and now I experienced that the switch _boiler_wwonetime_ from the boilder device doesn't trigger anything. It immediately turns to off again without any action. I think I remember that it has worked with 3.6.5

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0","uptime":"004+18:58:39.935","uptimeSec":413919,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":162,"maxAlloc":107,"freeCaps":4170,"usedApp":2376,"freeApp":3320,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4043,"model":""},"network":{"network":"Ethernet","hostname":"ems-esp","TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":131660,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":2,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":1,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":1687911,"busReads":420970,"busWrites":8,"busIncompleteTelegrams":41,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"de","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":100,"webLogBuffer":0,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i","deviceID":"0x08","productID":172,"brand":"Buderus","version":"01.21","entities":159,"handlersReceived":"0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04AA 0x04A7 0x02D6","handlersPending":"0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF 0x04A5","handlersIgnored":"0x049F 0x04A0 0x04A3 0xD2 0xE2 0xF7 0x0240 0x029B 0x04AB 0x04AC 0xE7 0x02E0 0x02EA 0x061E 0x3B 0x2E 0x17 0x36 0x0497 0xF6"},{"type":"thermostat","name":"RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"73.03","entities":112,"handlersReceived":"0x06 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02CC 0x0467 0x0291 0x02CE 0x0468 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E","handlersIgnored":"0x0239 0x0508 0x0509 0x050A 0x0507 0xBF 0xF6 0xF7 0x02C4"},{"type":"mixer","name":"MM100","deviceID":"0x21","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D8","handlersFetched":"0x02CE","handlersIgnored":"0x0255 0x02E2 0x02EC 0xE6 0x0292 0xBF 0xF7"},{"type":"gateway","name":"WiFi module","deviceID":"0x48","productID":252,"brand":"Buderus","version":"07.02","entities":0,"handlersIgnored":"0x0CBD 0xF7 0x125A 0x1259 0xF6 0xF9 0x03 0x04 0x17 0xD1 0xE3 0xE4 0x0269 0x05A7 0x026A 0x05A8 0x026B 0x05A9 0x026C 0x05AA 0x026D 0x05AB 0xE5 0xE6 0xE9 0xEA 0xEF 0x02E0 0x0484 0x048D 0x048F 0x0494 0x0495 0x0499 0x04A2 0x04A7 0x04AB 0x04AC 0x04AD 0x0CBC 0x0CBF 0x0CC3 0x1258 0x2040 0x04AA 0x023E 0x029B 0x029C 0x029D 0x029E 0x02B9 0x02BA 0x02BB 0x02BC 0x02A5 0x02A6 0x02F5 0x02FF 0x031D 0x0CC2 0x0291 0x0292 0x0293 0x0294 0x02C3 0x02C4 0x02C5 0x02C6 0x0449 0x044A 0x044B 0x044C 0x02AF 0x02B0 0x02B1 0x"}]}
```

### TO REPRODUCE

click/activate the switch _boiler_wwonetime_

### EXPECTED BEHAVIOUR
Clicking or activating the switch should initiate an onetime warmwater generation, but it doesn't. The switch immediately turns off and nothing happens. I think I remember that it worked with 3.6.5.

### ADDITIONAL CONTEXT

I also experienced that the switch _thermostat_wwcharge_ from the thermostat device, in german _WWK Laden_, initiated an extra warmwater generation which heats up water much faster and consumes much more energy. This could be named better.
There is also a sensor _thermostat_wwextra_ which has a wrong unit. It is a temperature in °C but I think it should be boolean as it is usually _0_ and just switches to _1_ when the extra warmwater _thermostat_wwcharge_ is activated.

proddy, 2024-11-05:
there are breaking changes in 3.7.0. But you would have seen that since you checked all those check-boxes above :-)

ww has been renamed to dhw. Since you mention 'switch' I'm guessing you're on Home Assitant. You can find the new name there too like:

[image]

if it still doesn't work, check both the HA and EMS-ESP logs.

jimmyhenne, 2024-11-05:
I'm aware of the changes coming with 3.7.0 and also noticed some entities changed from ww to dhw and some entities are unavailable and are now available with the new dhw spelling.
Yes, I use Home Assistant.
Should all ww-entities be renamed to dhw? I still have a lot with ww instead of dhw and they are still available. This is also for the boiler_ww_onetime. When I filter entities by onetime, HA just finds the ww one and it's available.
[image]

The strange thing is, that ww_onetime was also on when extra dhw charge _thermostat_wwcharge_ was running and changed to off when it has been charged. (see additional context)

As I didn't changed anything when updating to 3.7.0 because the I noticed all entities were available, are there any steps I have to do by hand?

proddy, 2024-11-05:
The cleanest way is to set the MQTT Entity Format ID to the standard "Short name, single instance" in EMS-ESP, then use something like MQTT Explorer to delete any left over entities (remove the homeassistant branch), and re-boot EMS-ESP. That way you'll be future safe in case another breaking change happens. If all of that is too many changes then we'll need to go through one by one and hack it. Right now you're in compatibility mode and its a mix of ww and dhw, and to be honest, I've never tested it thoroughly.

jimmyhenne, 2024-11-05:
I think I got you. So I already changed the Entity Format ID in EMS-ESP and restarted. In HA I have e.g. _boiler_wwonetime_ and _boiler_dhwonetime_
Now I'm running MQTT Explorer for the first time and I understood you to mean that I was deleting the entire home assistant branch here, right?
[image]
After restart I would have deleted all old and not longer available entities and just have the new ones would I?

proddy, 2024-11-05:
> I think I got you. So I already changed the Entity Format ID in EMS-ESP and restarted. In HA I have e.g. _boiler_wwonetime_ and _boiler_dhwonetime_ Now I'm running MQTT Explorer for the first time and I understood you to mean that I was deleting the entire home assistant branch here, right? ![grafik](https://private-user-images.githubusercontent.com/34960757/383241075-54f97945-a47d-41e8-8724-3e8c28bd4199.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NDAzMzksIm5iZiI6MTczMDg0MDAzOSwicGF0aCI6Ii8zNDk2MDc1Ny8zODMyNDEwNzUtNTRmOTc5NDUtYTQ3ZC00MWU4LTg3MjQtM2U4YzI4YmQ0MTk5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA1VDIwNTM1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ1NzgxMWY1ZDg0ZjIyODk2YzlhOWI4MTkzM2MyMDY0N2VlN2E0ZmNlOTdlYWJmY2RmNzQ1OWFhODQzNWI5MDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.K6VDJ4F-skLYjkY3KRHuBfgq520c1PJ0-0-xarDkJbw) After restart I would have deleted all old and not longer available entities and just have the new ones would I?
> 
> Edit: Sorry for my inconvenience

No inconvenience at all. I can imagine how frustrating this can be when you upgrade a product and it breaks all the automation. We do our best to build in migration scripts, but there will always be breaking changes that we can't handle, especially when it touches integrations. And there are really only 2 developers working in their spare time on this project.

Anyways...yes you would use MQTT Explorer, click on 'homeassistant' and hit the delete icon on the right next to the topic. The problem with this method, however is that you'll lose all the historic data which HA keeps. If this is an issue, then I suggest you stick to the old naming MQTT entity ID format.

jimmyhenne, 2024-11-05:
Historical data aren't an issue.
I did it 😄 And it looks good so far. I will try dhw_onetime again tomorrow if it works as intended.
Thanks for support!

Just one more thing I noticed: _thermostat_dhw_extra_ is defined as a temperature but it's simply a 0 and 1 value. Usually 0 and when extra warmwater is charged 1.

proddy, 2024-11-06:
> Just one more thing I noticed: _thermostat_dhw_extra_ is defined as a temperature but it's simply a 0 and 1 value. Usually 0 and when extra warmwater is charged 1.

good spot. It's incorrectly set to degrees in this line `register_device_value(tag, &dhw->wwExtra_, DeviceValueType::UINT8, FL_(wwExtra), DeviceValueUOM::DEGREES);`. I'll fix

jimmyhenne, 2024-11-06:
Unfortunately the switch _boiler_dhw_onetime_ still doesn't work.
When I click the switch in Home Assistant it goes on but after a second it switches to off again and the heat pump doesn't swich to warmwater charge.
In the HA log is no entry concerning the switch but nonetheless there is at least one message in the EMSESP log

> 000+15:04:29.402    INFO 29:     [command]    Called command boiler/onetime (Einmalladung) with value an and id 9 on device 0x08
000+15:04:29.435   TRACE 30:     [emsesp]     me(0x0B) -W-> boiler(0x08), UBAFlags(0x35), data: 22
000+15:04:31.561   TRACE 31:     [emsesp]     me(0x0B) -R-> boiler(0x08), UBAMonitorWWPlus(0xE9), length: 27
000+15:04:31.604   TRACE 32:     [emsesp]     boiler(0x08) -W-> me(0x0B), UBAMonitorWWPlus(0xE9), data: 34 01 C7 01 B5 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 34
000+15:04:31.604   TRACE 33:     [emsesp]     boiler(0x08) -B-> all(0x00), HPPower2(0x04AA), data: 03 DE 01 04 08 1E 00 00 00 00 00 00 00 00 00 00 00 00 03 EC 00 F0 00 00
000+15:04:32.005   TRACE 34:     [emsesp]     boiler(0x08) -B-> all(0x00), HPPower2(0x04AA), data: 00 00 (offset 24)
000+15:04:33.953   TRACE 35:     [emsesp]     boiler(0x08) -B-> all(0x00), HPPower2(0x04AA), data: 03 DE 01 04 08 1F 00 00 00 00 00 00 00 00 00 00 00 00 03 ED 00 F0 00 00
000+15:04:34.129   TRACE 36:     [emsesp]     boiler(0x08) -B-> all(0x00), HPPower2(0x04AA), data: 00 00 (offset 24)
000+15:04:37.016   TRACE 37:     [emsesp]     boiler(0x08) -B-> all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 30 48 00 CB 1A 01 00 2A 2B 02 00 02 B2 00 00 00 FE 00 00 00 00 01 32 01 00
000+15:04:37.205   TRACE 38:     [emsesp]     boiler(0x08) -B-> all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 00 2B 64 55
000+15:04:37.401   TRACE 39:     [emsesp]     boiler(0x08) -B-> all(0x00), UBAMonitorWWPlus(0xE9), data: 34 01 C7 01 B5 00 00 00 00 41 3C 00 00 00 00 00 00 00 00 00 00 00 00 34
000+15:04:37.830   TRACE 40:     [emsesp]     boiler(0x08) -B-> all(0x00), ?(0x049F), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+15:04:38.007   TRACE 41:     [emsesp]     boiler(0x08) -B-> all(0x00), ?(0x04A0), data: 00 00 01 6D
000+15:04:38.203   TRACE 42:     [emsesp]     boiler(0x08) -B-> all(0x00), ?(0x04A3), data: 00 01 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 08 00 00 00
000+15:04:39.052   TRACE 43:     [emsesp]     boiler(0x08) -B-> all(0x00), HPPower2(0x04AA), data: 03 DE 01 04 08 1F 00 00 00 00 00 00 00 00 00 00 00 00 03 ED 00 F0 00 00

I can't really interprete these lines but _value an_ in INFO 29 seems trange to me doesn't it? I would assume that it should at least be _on_.

jimmyhenne, 2024-11-06:
I'm pretty sure it worked in 3.6.5
Let me know if I can test, try or provide you with more information.

MichaelDvP, 2024-11-07:
The boiler onetime has a special logic. It's not recharging on call as most people expect. It checks the current temperature and only recharges if it is out of setpoint range (setpoint-hysteresis). If dhw is set to automatic recharge, it will never work, only in dhw eco mode and low dhw storage temperature. For just recharging you have to use the thermostat charge function. That sets the dhw setpoint temporary to a higher value,so the boiler triggers a recharge, and then back to old setpoint.

proddy, 2024-11-07:
Thanks Michael, didn't know that. I have added to the doc.

jimmyhenne, 2024-11-07:
> It checks the current temperature and only recharges if it is out of setpoint range (setpoint-hysteresis). If dhw is set to automatic recharge, it will never work, only in dhw eco mode and low dhw storage temperature.

Thanks for your explanation. So I must have been wrong with my assessment 😄 
Then I would rather use the thermostat charge, which does work as I experienced two days ago.
It charged dhw to the same temperature as the time programm in dhw eco (_boiler_dhw_seltempsingle_ and _boiler_dhw_ecooff_ set to 52°) but it ran in a let's call it performance mode and required almost 4 kW electricity (_boiler_hpcurrpower_) for charging while it usually just needs 1,5 kW. Do you have an idea why? I couldn't find a setting that I found conclusive for that.

MichaelDvP, 2024-11-07:
Don't know, maybe because the setpoint is raised for the charging. Check if the compressor is running on more power or the auxillary heater is triggert. I have only a oil burner, so i'm not familiar with the heatpump settings.

jimmyhenne, 2024-11-07:
Ok, since it behaves the same way when I activate extra dhw on the heat pump, it's not due to EMSESP.
Thanks for your support.

closed 2024-11-07 by jimmyhenne