emsesp/EMS-ESP32 #2182: enable/disable services via API

proddy, 2024-11-06:
Triggered by https://github.com/emsesp/EMS-ESP32/discussions/2168 two new commands were added in 3.7.1 to control the shower (`/api/system/showeralert` and `/api/system/showertimer`).

A more generic approach is to extend the new `action` command, moving this from /rest/ to a HTTP POST API `/api/system` so we can use a format like {"action":'enable|disable', param:'ntp|ap|syslog|showeralert...'}.`

This will remove 3 EMS-ESP Commands and free up some memory.

proddy, 2024-11-10:
Request to extend this to turn LED on and off as well

https://github.com/emsesp/EMS-ESP32/discussions/2194

MichaelDvP, 2024-11-28:
If i understand right, the appoach with `action` is API only and can not used via console or mqtt. 
I've made a alternative appoach withing the command structure. Please check and tell what do you think about it.
https://github.com/emsesp/EMS-ESP32/commit/30fca2a190f8c06692b19a4a63bf62ea5033b1b4

MichaelDvP, 2024-11-29:
I think the commands should always be the same as api read of the entity. E.g.
`api/system/settings/hideled` should show (there is a camelCase bug in my dev for writeable):
```
{
  "name": "hideLed",
  "circuit": "settings",
  "readable": true,
  "writeable": true,
  "visible": true,
  "value": false,
  "type": "boolean"
}
```
So the command also has to be `api/system/settings/hideled` with value.
Using different commands does not allow automatisation like thomas do it in his ioBroker adapter:
- reading the system/info
- reading `system/<path>/<entity>`
- create entities depending on readable/writeable settings
> as this makes it easier for the end user.

ot sure, as i implemented the shower settings without prefix the user extected them as `settings/showeralert` and i have to explain why this is not normal scheme.

proddy, 2024-11-29:
okay, that actually makes a lot of sense. What do you want me to do next, test?

proddy, 2024-12-04:
The doc also needs to be updated.

closed 2024-12-19 by proddy