emsesp/EMS-ESP32 #2188: heatpump entity values incorrect formatting/scaling

proddy, 2024-11-07:
_Originally posted by @Domoberry in https://github.com/emsesp/EMS-ESP32/issues/2177#issuecomment-2459670500_
           
  o	The heatpump (energy price gas) shows up as “13” for example, meaning 1.3 € in the Easycontrol App (note the scaling)
  o	The heatpump (energy price electric) shows up as “25” for example, meaning 0.25€ in the App
  o	The heatpump (energy feed in PV) price shows up as “24” for example, meaning 0.24€ in the App
  o	The heatpump (low noise starttime) is a number (afaik) in minutes
  o	The heatpump (low noise stoptime) is a number (afaik) in minutes

proddy, 2024-11-07:
@Michael, add some DIV10 and DIV100 will solve the first 3 I think. For the noise times we can change these to DeviceValueUOM:MINUTES ?

MichaelDvP, 2024-11-08:
Scaling depends on the unit used. 
The raw values uom is `ct/kWh`, should we add this as UOM. Difficult in countries with different currency.
The Easycontrol scales the pv/grid to `€/kWh` and the gas to `€/m³`. 
The price compare portals like check24.de  always show the prices in ct/kWh, also for gas (in germany). 
I prefere to leave it unscaled, but add the unit. 
If really needed we can add a currency setting (string + factor) to emsesp-settings and use this in uom output. 

The low noise times are the same as we have in boiler as `silentmode`, `silentfrom`, `silentto` and is a time of day `hh.mm`  scaled as uint8 in steps of 15 min. Actual we show minutes, but a time ago we promise to change output to clock. 
Should we add a DeviceValueType::CLOCK as uint8_t with web output as string `hh:mm` and api/mqtt output minutes? (we've  made similar for type::TIME). 
I think we rename to silentmode, saves also flashspace.

In the heatpump.cpp there is a `switchovertemp` without uom, i think this should be degrees, and `hybriddhw ` should be tagged with DHW.

MichaelDvP, 2024-11-08:
I'm wrong, the low noise times / silenttimes for the HM200 are in hours 0-23, not clock like the heatpumps with device 0x08.
see #500

MichaelDvP, 2024-11-08:
I've made some change in https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Plese check. Should show uoms for the prices a ct/kWh  and hours for the low noise times.

proddy, 2024-11-10:
@Domoberry would you mind checking please?

Domoberry, 2024-11-11:
Apologies for the somewhat later response due to (nice) personal circumstances :-)
I installed 3.7.1-dev.5 and took a look at the various entities in the Domoticz "hardware" setup:
[image]
Looks fine for electricity, but scaling and unit (kWh) for the gas price is not as I would expect:
the '13' is shown as '1.3 E/m3' in the Easycontrol app. I would expect to see for example '130 ct/m3'

[image]
This look great. Agree that the numbers refers to hours rather then minutes. 

[image]
The oum of degrees-C is perfect.
I do wonder about the fact that the entity is called 'dhw Outside switchover temperature' and not 'Outside switchover temperature'. If I change the 'Bivalent temperature' in the EasyControl app, 'dhw Outside switchover temperature' changes accordingly. But this settings is not (only) for dhw afaik, its setting determines the outside temperature below which the conventioal boiler kicks in (my hybrid system does not use the heatpump for dhw).

MichaelDvP, 2024-11-11:
>  I would expect to see for example '130 ct/m3'

The module reports in ct/kWh. Don't know why the Easycontrol changes the units to make comparing difficult. Maybe political to make gas looks more expensive to users not checking the units.
> I do wonder about the fact that the entity is called 'dhw Outside switchover temperature' and not 'Outside switchover 

That's a mistake, wanted to map `"hybriddhw` to the dhw circuit, and put it in the wrong line.

commit afc2afba (2024-11-14): heatpump uoms fix #2188
closed 2024-11-14 by proddy