emsesp/EMS-ESP32 #2189: Invalid JSON because of unescaped character "lastcode"

0Bu, 2024-11-08:
### DESCRIPTION

It looks like my ems-esp sends an invalid JSON because of unescaped character to the mqtt.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.0","uptime":"000+01:50:38.536","uptimeSec":6638,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":156,"maxAlloc":87,"freeCaps":107,"usedApp":1730,"freeApp":6398,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-46,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"pool.ntp.org","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":2376,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":2,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":0,"publishTimeThermostat":0,"publishTimeSolar":0,"publishTimeMixer":0,"publishTimeWater":10,"publishTimeOther":0,"publishTimeSensor":0,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":7367,"busReads":2092,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32","locale":"en","txMode":3,"emsBusID":11,"showerTimer":true,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":false,"maxWebLogBuffer":25,"webLogBuffer":19,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"18.01","entities":64,"handlersReceived":"0x10 0x11 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x15 0x1C 0x1A 0x35 0x26 0x2A 0x2E 0x3B","handlersIgnored":"0x23 0x05"},{"type":"thermostat","name":"RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"74.03","entities":56,"handlersReceived":"0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlersPending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E","handlersIgnored":"0xBF"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"18.01","entities":0}]}
```

### TO REPRODUCE

"lastcode" has an unescaped character

```json
{
    "reset": "",
    "heatingoff": "off",
    "heatingactive": "off",
    "tapwateractive": "off",
    "selflowtemp": 51,
    "heatingpumpmod": 100,
    "outdoortemp": 6.6,
    "curflowtemp": 50.3,
    "burngas": "off",
    "burngas2": "off",
    "fanwork": "on",
    "ignwork": "off",
    "oilpreheat": "off",
    "burnminpower": 0,
    "burnmaxpower": 100,
    "burnminperiod": 3,
    "boilhyston": -10,
    "boilhystoff": 0,
    "heatingactivated": "on",
    "heatingtemp": 55,
    "heatingpump": "on",
    "pumpmodmax": 100,
    "pumpmodmin": 10,
    "pumpmode": "deltaP-4",
    "pumpdelay": 3,
    "selburnpow": 100,
    "curburnpow": 0,
    "burnstarts": 48646,
    "burnworkmin": 1419876,
    "burn2workmin": 0,
    "heatworkmin": 1035952,
    "heatstarts": 44720,
    "ubauptime": 914469,
    "lastcode":"(256) 00.00.2000 00:00 (130 min)",
    "servicecodenumber": 0,
    "nompower": 14,
    "nrgtotal": 11489.68,
    "nrgheat": 5642.34,
    "dhw": {
        "settemp": 47,
        "seltemp": 47,
        "type": "buffer",
        "comfort": "hot",
        "flowtempoffset": 38,
        "circpump": "off",
        "chargetype": "chargepump",
        "hyston": 0,
        "hystoff": 0,
        "disinfectiontemp": 75,
        "circmode": "2x3min",
        "circ": "off",
        "curtemp": 46.1,
        "curtemp2": 46.1,
        "curflow": 0.0,
        "activated": "on",
        "onetime": "off",
        "disinfecting": "off",
        "charging": "off",
        "recharging": "off",
        "tempok": "off",
        "active": "off",
        "3wayvalve": "off",
        "starts": 4037,
        "workm": 383923,
        "nrg": 5847.34
    }
}
```

### EXPECTED BEHAVIOUR

remove invalid/control characters

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-11-08:
Please post the output for `read 08 10` and `read 08 11` to check what is wrong in these telegrams.

MichaelDvP, 2024-11-08:
Please test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
ignoring the telegram if it does not contain a valid errormessage.

0Bu, 2024-11-08:
It doesn’t look like the latest build has fixed the bug. The non-ascii character is still in the UI and in the JSON.

```json
{"system":{"version":"3.7.1-dev.4","uptime":"000+00:06:25.891","uptimeSec":385,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":157,"maxAlloc":83,"freeCaps":107,"usedApp":1733,"freeApp":6395,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-47,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"pool.ntp.org","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":255,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":false,"entityFormat":1,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":0,"publishTimeThermostat":0,"publishTimeSolar":0,"publishTimeMixer":0,"publishTimeWater":10,"publishTimeOther":0,"publishTimeSensor":0,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":533,"busReads":195,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32","locale":"en","txMode":3,"emsBusID":11,"showerTimer":true,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":false,"maxWebLogBuffer":25,"webLogBuffer":17,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"18.01","entities":64,"handlersReceived":"0x10 0x11 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x15 0x1C 0x1A 0x35 0x2A 0x2E 0x3B","handlersIgnored":"0x23"},{"type":"thermostat","name":"CW400/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"74.03","entities":56,"handlersReceived":"0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlersPending":"0xA3 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"18.01","entities":0}]}
```

0Bu, 2024-11-08:
Now I had to restart the s32 in order to activate telnet, the error is gone 🤔

proddy, 2024-11-10:
closing this one then, fix will be in Michael's upcoming PR and 3.7.1

0Bu, 2025-03-22:
It looks like, the error has came back into 3.7.2

[image]

```json
{"reset":"","heatingoff":"0","heatingactive":"0","tapwateractive":"0","selflowtemp":85,"heatingpumpmod":100,"outdoortemp":17.9,"curflowtemp":75.9,"burngas":"0","burngas2":"0","fanwork":"0","ignwork":"0","oilpreheat":"0","burnminpower":0,"burnmaxpower":100,"burnminperiod":3,"boilhyston":-10,"boilhystoff":0,"heatingactivated":"1","heatingtemp":50,"heatingpump":"1","pumpmodmax":100,"pumpmodmin":10,"pumpmode":"deltaP-4","pumpdelay":3,"selburnpow":100,"curburnpow":0,"burnstarts":82786,"burnworkmin":1472300,"burn2workmin":0,"heatworkmin":1076653,"heatstarts":77997,"ubauptime":1107995,"lastcode":"(256)","servicecodenumber":0,"maintenancemessage":"H00","maintenance":"Manuell","maintenancetime":2300,"nompower":14,"nrgtotal":18220.12,"nrgheat":10083.93,"dhw":{"settemp":50,"seltemp":50,"type":"Speicher","comfort":"Heiß","flowtempoffset":35,"circpump":"0","chargetype":"Ladepumpe","hyston":0,"hystoff":0,"disinfectiontemp":75,"circmode":"2x3min","circ":"0","curtemp":48.6,"curtemp2":48.6,"curflow":0.0,"activated":"1","onetime":"0","disinfecting":"0","charging":"1","recharging":"0","tempok":"0","active":"0","3wayvalve":"0","starts":4715,"workm":395646,"nrg":8136.19}}
```

MichaelDvP, 2025-03-24:
Please check if t is fixed with https://github.com/MichaelDvP/EMS-ESP32/releases

0Bu, 2025-03-24:
The error no longer occurs in version v3.7.3-test.1. Thanks for the quick fix!

commit df92e925 (2024-11-14): check lastcode, fix #2189, allow setting silentfrom/to as hh:mm
closed 2025-04-05 by proddy