emsesp/EMS-ESP32 #2190: Firmware upgrade notification is confusing

red-black-bonsai, 2024-11-08:
### DESCRIPTION

The firmware upgrade notification in http://ems-esp/settings/version is confusing: "Version: 3.7.0. There is a firmware upgrade available! Install v3.7.0"

To be fair, this is not a minor but a miniscule issue :smile: 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT) (Can't for the life of me join this chat :disappointed:  )
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json

{
  "system": {
    "version": "3.7.0",
    "uptime": "000+01:02:59.109",
    "uptimeSec": 3779,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 189,
    "maxAlloc": 147,
    "freeCaps": 8318,
    "usedApp": 2278,
    "freeApp": 3418,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8130,
    "model": "",
    "temperature": 55.2
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -64,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 1068,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp-1",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 15736,
    "busReads": 3456,
    "busWrites": 4,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "en",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 8,
    "txGPIO": 5,
    "dallasGPIO": 0,
    "pbuttonGPIO": 0,
    "ledGPIO": 38,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "webLogBuffer": 42,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i",
      "deviceID": "0x08",
      "productID": 172,
      "brand": "Buderus",
      "version": "02.02",
      "entities": 156,
      "handlersReceived": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlersFetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04AA 0x02D6",
      "handlersPending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF 0x04A5 0x04A7",
      "handlersIgnored": "0xF7 0x049F 0x04A0 0x04A3 0x0240 0xE7 0x02E0 0x02EA 0x061E 0xF6 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "73.04",
      "entities": 67,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E",
      "handlersIgnored": "0x0239 0x0508 0x0509 0x050A 0x0507 0xBF 0xF7"
    },
    {
      "type": "gateway",
      "name": "WiFi module",
      "deviceID": "0x48",
      "productID": 252,
      "brand": "Buderus",
      "version": "08.02",
      "entities": 0,
      "handlersIgnored": "0x2040 0xF9 0x0495 0x0494 0x3B 0xE5 0xE9"
    }
  ]
}

```

### TO REPRODUCE

1. Go to http://ems-esp/settings/version
2. Look at the notification
3. Wonder

### EXPECTED BEHAVIOUR

If a newer version is announced here, I would expect the new version number shown to be higher than the currently installed version.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

After installing a dev build, I'm told that I use the latest version.

[image]

So I suspect some part of this update check doesn't differentiate between stable and dev versions but the other part does?

proddy, 2024-11-08:
That shouldn't be the normal behaviour, I'll test it again. Thanks for reporting.

MichaelDvP, 2024-11-08:
No this PR is the fix for this issue. Just compile the actual dev with release version number and you'll see tht it's fixed.

proddy, 2024-11-08:
ah I see now. I'll need to think about a better logic here as the rest call is done before useDev is set (it'll always be false).

MichaelDvP, 2024-11-08:
Oh, i see, now the compare is always to the release from web. If you roll back the PR the dev will show the updates again, but we need to find another solution before next release.

proddy, 2024-11-08:
yeah, I'm thinking about a solution. Which is to chain the useRequest APIs to fetch both the dev and stable releases from GitHub, and then only call the WebStatusService::checkUpgrade() action. I can do this with a useEffect but I know alova has a nicer way - just need to remember how.

VlastiBroucek, 2024-11-08:
While you are talking about it, I think that:
- it would be good if it was showing next dev released when using dev too. Currently dev3 shows as latest...
- IMHO it belongs more to STATUS menu than to currently being in SETTINGS menu

proddy, 2024-11-08:
> While you are talking about it, I think that:
> 
> * it would be good if it was showing next dev released when using dev too. Currently dev3 shows as latest...
> * IMHO it belongs more to STATUS menu than to currently being in SETTINGS menu

your first point, that is how it should be. Working on a fix.
your second point, the decision came from https://github.com/emsesp/EMS-ESP32/issues/2118, where we originally showed the Version in the Status page, then redirected the user to the Settings page, and when they clicked on the back arrow, it would get confused. So the easiest solution was to keep everything together. Also, Settings is only for users with 'admin' rights. We have many users that are guests (think apartment block + landlord) where we don't want guest users to know about versions or upgrade, so this also shields it off. I'm open to think of other approaches. Enjoy the summer down under, bloody freezing here!

VlastiBroucek, 2024-11-09:
> your first point, that is how it should be. Working on a fix. your second point, the decision came from #2118, where we originally showed the Version in the Status page, then redirected the user to [...]

That makes sense - did not think about it that way...

Yes, it's very pleasant here - probably not as hot as people think - I am down in Tasmania...

MichaelDvP, 2024-11-09:
> yeah, I'm thinking about a solution. Which is to chain the useRequest APIs to fetch both the dev and stable release versions from GitHub first, and then only call the WebStatusService::checkUpgrade() action, which is done once. I can do this with a useEffect but I know alova has a nicer way - just need to remember how.

This sounds like a good solution. 
For now i have made a workaround, calling checkUpgrade twice and check there for dev prerelease.
Have not tested with all different combinations, but think it will work. 
Please check this commit: https://github.com/MichaelDvP/EMS-ESP32/commit/38efb0de67497e994c1bb427f3e8f2cc0afb5a8f

proddy, 2024-11-09:
I made a comment

proddy, 2024-11-10:
improved in 3.7.1-dev.5

closed 2024-11-10 by proddy