emsesp/EMS-ESP32 #2191: remote thermostat A11 error

quark654, 2024-11-08:
Hi all,

A while back I downloaded the latest ems-esp release to be able to emulate a virtual thermostat.
Unfortunately after a couple of minutes of running I receive an A11 error message.
I noticed a similar issue in [this ](https://github.com/emsesp/EMS-ESP32/issues/1611) thread, but I doubt it's the same cause as the aforementioned issue should have been resolved in the latest release.

I have also tried to install the latest development release (3.7.1-dev.4), alas to no avail.
Just to give you some more insight in the system we have, the Buderus burner is connected to a RC310 thermostat, acting as master, to which another RC200 remote thermostat is connected. There are 3 heating circuits and I'm trying to connect a virtual thermostat to hc1. The RC200 is installed to control hc2.  The RC310 controls both hc1 and hc3. After getting the virtual thermostat running on hc1, it would be nice if I can add another to hc2, although I faced some difficulties as we already have the RC200 there. I don't see the virtual thermostat appear in the device list, but let's keep that as a secondary issue.

Attached is also the log which starts just before submitting the remotetemp command and which ends shortly after receiving the A11 error.
[log.txt](https://github.com/user-attachments/files/17684070/log.txt)

Do you have any idea on what might be causing this issue?

Thanks!

MichaelDvP, 2024-11-09:
This is a different error, you have A11(3081). See manual : 
``` 
Konfigurationsfehler: Fernbedienung nicht verwendet (3081 = Heizkreis 1)
Konfiguration prüfen. Es ist eine Fernbedienung im System,die mit der gewählten Einstellung nicht verwendet wird
```
You have to set control to RC200 for hc1. `thermostat/hc1/control` value:1
And the real RC200 is configured to hc3. 
To also use a virtual on hc2 just set `thermostat/hc2/control` to RC200 and send a `thermostat/hc2/remotetemp`

quark654, 2024-11-09:
Thanks for your swift reply, that was a silly mistake.
Sting the control to RC200 indeed seems to do the trick.

Thanks a lot!

closed 2024-11-09 by quark654