emsesp/EMS-ESP32 #2192: "Aux heater status" problem

masterx1981, 2024-11-10:
I'm using the latest stable 3.7 firmware. On the previous firmware before 3.7 i've had an entity called on the web UI "Aux heater status" (on HA was binary_sensor.ems_esp_pdc_boiler_auxheaterstatus), and was turning on when the heatpump (in an hybrid system) requested hot water from the external gas boiler.
Now i see it as a percentage value, and not change state when the hp request hot water from the gas burner (the HA name now is sensor.ems_esp_pdc_boiler_auxheaterstatus)
On the hp there is a setting that can change the output from on/off to a linear 0-10v, but i've never changed it, and isn't working as was working before.

MichaelDvP, 2024-11-10:
Yes, this has been changed in https://github.com/emsesp/EMS-ESP32/commit/355ff5656aeb3dd6ecb96d900212e9a6faeabb90 due to #1160.
But seems we  need both values. 
@proddy Do you agree to add back the old bool auxheaterstatus and rename the newer percent entity to auxheaterlevel

masterx1981, 2024-11-10:
I think that both values are valid, based on how the output is configured.

proddy, 2024-11-10:
strange that the value can have two interpretations. Maybe just leave it as uint/DeviceValueUOM::NONE and let the user handle how it behaves.

masterx1981, 2024-11-10:
Also strange that in percentage i not see any change of the value. I was expecting at least 0% off 100% on or something similar. Instead, it remain fixed to 0%, also when i'm sure that it has changed.

masterx1981, 2024-11-10:
Maybe both can be mapped to the same auxheaterstatus field, instead of having two separate entities? 0% off, 100% on? I not think that can cause problems to anyone. Check a boolean, or check an int, doesn't change too much.

masterx1981, 2024-11-22:
Hi! I can confirm that auxheater status (bool) works as it should in 3.7.1.
Thanks!

closed 2024-11-22 by proddy