emsesp/EMS-ESP32 #2197: Lost entity after updates

gume, 2024-11-11:
This issue is on the release branch, I have not tested the dev branch. 

### DESCRIPTION

After update to the 3.7 release version, the previously working flame current sensor info (boiler_flamecurr) is lost. This is not a name change, as I cannot find any similar sensor under different name, At a previous update the same thing happened with ignition sensor (boiler_ignwork), but that wasn't a problem for me. I just know that these sensors are there.

My boiler is recognised as "Generic Boiler (34)", but it is a Bosch Condens GC7000iW
I use a custom adapter, where the TX is blocked (by hardware and software)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

{"system":{"version":"3.7.0","uptime":"001+13:56:52.346","uptimeSec":136612,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":163,"maxAlloc":91,"freeCaps":113,"usedApp":1730,"freeApp":254,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-52,"BSSID":"set","TxPowerSetting":78,"staticIP":false,"lowBandwidth":true,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":29689,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTConnects":1,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":3,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":6,"temperatureSensorReads":163913,"temperatureSensorFails":1,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected, tx issues - try a different Tx Mode","busProtocol":"HT3","busTelegramsReceived":52684,"busReads":0,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":57,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":0},"settings":{"boardProfile":"CUSTOM","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"phyType":0,"rxGPIO":23,"txGPIO":5,"dallasGPIO":18,"pbuttonGPIO":0,"ledGPIO":2,"hideLed":false,"noTokenApi":false,"readonlyMode":true,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":false,"maxWebLogBuffer":25,"webLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Generic Boiler","deviceID":"0x08","productID":0,"brand":"","version":"00.00","entities":34,"handlersReceived":"0x1C 0x18 0x19 0x34 0x2A","handlersPending":"0xBF 0x10 0x11 0xC2 0x14 0x15 0x1A 0x35 0x16 0x33 0x26 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x28 0x0494 0x0495 0x048D 0x048F 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x04A5 0x04AA 0x04A7 0x02D6"},{"type":"controller","name":"Generic Controller","deviceID":"0x09","productID":0,"brand":"","version":"00.00","entities":0},{"type":"temperaturesensor","name":"temperaturesensor","entities":6}]}
```

### TO REPRODUCE

Just switch on and it is (not) there

### EXPECTED BEHAVIOUR

I have used this value before, so I expect that this sensor is available and would transmit correct data.

### SCREENSHOTS

Not applicable.

### ADDITIONAL CONTEXT

I receive and process the sensor values in Home Assistant. 

**(Please remember to close the issue when it has been addressed)**

gume, 2024-11-11:
Thanks for the quick answer. The TX mode is disabled intentionally and because of the cold weather here, I would not experiment with turning it on until next summer. Is there any option to override the default "generic boiler" and set up the type manually? Could this be a feature in the next release?

MichaelDvP, 2024-11-11:
You have set readonlymode, that's ok. But tx-mode is set to 1 and read requests all give error, so there is something wrong with the tx connection. I think a wrong floating gpio on the tx-line is much more dangerous to accidential block the bus, than to use readonly mode and allow read requests. I fyou really want to block any activity on the tx line, set tx-mode 0 in terminal, holding the tx-gpio on the fixed idle level. 
For generic boiler comment out this line:
https://github.com/emsesp/EMS-ESP32/blob/9b91faba1015fa6a902c8c33c6f40e4ddb1ee7eb/src/emsesp.cpp#L1347
and compile your own version.

gume, 2024-11-11:
Thanks, I will comment out that line and see if that is better.

On the hardware, the 1K resistor is not placed. This resistor would pull down the bus. Without that the module cannot influence the bus signal, so I guess there could be no accidents, even the MCU tx-line is floating. Of course I will put it there, just I would like to wait for the end of the heating season.

closed 2024-11-11 by gume