emsesp/EMS-ESP32 #2200: the analog sensor does not work in the "counter" mode

KCH1963, 2024-11-14:
Water and gas meters are connected according to this diagram.
[image]
In EMS-ESP it was written like this
[image]
In the Home assistant  I see the meters like this, but they don't count the readings.
[image]
I installed pull-up resistors on the GPIO counters
in the open state of the reed switch at the output of the ESP 32 (GPIO 25/26/27) = 3v3, in the closed state = 0v
signal levels on gpio 25 when water is consumed
[image]
Soldered a resistor, did not help. In the "counter" mode, the readings are not counted. When setting the GPIO to the "Digital in" mode, the ESP-EMS begins to see pulses and output them to "on / off" No miracle happened
[image]
It doesn't work with these settings.
[image]
[image]
[image]

### DESCRIPTION

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2024-11-14:
Please add your system info (download from help page). 
Have you tried with different gpio? On esp32 pin 25 and 26 are also dac output, maybe this conflicts with the input-pullup config.

proddy, 2024-11-14:
This is from the Discord chat https://discord.com/channels/816637840644505620/816958041345884180/1306610790375100499

I can't reproduce it. I think it's because of the GPIO 25 and dac