emsesp/EMS-ESP32 #2206: EMS-ESP s32 v2.0 (2022) bricked after Firmware Update

teabox93-git, 2024-11-15:
### DESCRIPTION

I have EMS-ESP s32 v2.0 (2022) and I have tried to update it via WEB UI, downloaded firmware that showed up after checking new version (it is 3_7_0 16MB). It started the update.
Since then the device does not stand up, reconnecting it to the cable is not showing blue LED.

I have tried to connect it via USB and upload firmware from PC, but it is not recognized by PC, no serial port is showing up.

proddy, 2024-11-15:
What version did you upgrade from?

MichaelDvP, 2024-11-15:
> but it is not recognized by PC, no serial port is showing up.

Try another usb cable and check usb/serial driver. The usb is a CH340 chip. Independend of the processor state it should show up on USB.

teabox93-git, 2024-11-15:
Guys, the issue was the cable. On different cable I can flash the firmware. @Proddy sorry I don't remember the original version but I think it was 3.6.0.

Now I have flashed the 3.6.0, 3.6.5 and 3.7.0, on every version I have the same issue. When I connect my external temperature sensor - controller (Bosch) to the bus, the EMS-ESP goes down, led goes down and website is down like there is no power. When I connect only ESP-EMS to my Buderus all is working.

It seems that it cannot power up together with sensor connected to the bus. It was working fine before the firmware upgrade so it is not a power issue. Maybe I need to set some settings? Can you please help? On board there is S32 V 2.0

teabox93-git, 2024-11-15:
[image]
[image]
[image]

MichaelDvP, 2024-11-15:
You have a S32 gateway not a E32, to set board profile to S32.

proddy, 2024-11-16:
he got it working - it was a bad connection

closed 2024-11-16 by proddy