emsesp/EMS-ESP32 #2216: Free Memory decreased by HTTP API requests

semmel3k, 2024-11-17:
### DESCRIPTION

With frequent HTTP API requests every minute (for example with "/api?device=boiler&cmd=info") , the free memory continuously decreases until the module freezes after 45 minutes.
No other services active (MODBUS off; MQTT off, etc.). This occured with latest 3.7.0. and 3.7.1. developer version with ESP32-S3.

### TO REPRODUCE

frequent HTTP API requests

### EXPECTED BEHAVIOUR

That the free memory not decreases and no freeze

proddy, 2024-11-17:
Thanks for reporting. We'll try and reproduce it. Can you send your Support Info please so we can see what hardware, memory etc is being used. Did you notice the same behaviour with v3.6.5?

proddy, 2024-11-17:
There's a benchmark load test I wrote a while back at https://github.com/emsesp/EMS-ESP32/blob/dev/scripts/run_memory_test.py . I'm not sure it works with the latest 3.7.0, so probably needs updating.

I don't know how 'techie' you are, but it'll help if you could try and reproduce this using a script (maybe with curl or a Python equivalent) and help trace where it's breaking. Or look into advancing that script above.

proddy, 2024-11-17:
It's ok. It's just a script that uses curl and makes frequent API calls while monitoring the memory. I'll can work on that. The Gira homeserver is KNX based, you may be interested in the great work @Ing-Dom has been doing at https://github.com/emsesp/EMS-ESP32/discussions/1737

MichaelDvP, 2024-11-18:
I don't think it's the api call. I have ioBroker adapter running with >50000 api reads the last day with no effect to memory.
Could you please add the system info from the help page.

semmel3k, 2024-11-18:
Yes, but currently I use Modbus.
[emsesp_system_info.json.txt](https://github.com/user-attachments/files/17798805/emsesp_system_info.json.txt)

proddy, 2024-11-18:
I improved the memory load tests (in my fork) and ran it for 1hr calling boiler info, and saw no decrease in memory. I'll close this. If the issue is with Modbus (which doesn't use the API) then please create a new issue.

MichaelDvP, 2024-11-18:
> es, but currently I use Modbus.

You have still 38 API calls and 2 API fails. Could it be another API call, or a API fail that have the memory leak. 
Can you reproduce or remember the AP calls/command you have used?

semmel3k, 2024-11-18:
Now with active HTTP API requests. 
[emsesp_system_info.json](https://github.com/user-attachments/files/17803552/emsesp_system_info.json)

MichaelDvP, 2024-11-18:
Does this setting mean that you always cut the connection before all data are received (3382 bytes)?  
I think then some data stuck in sending buffer and waiting.

semmel3k, 2024-11-18:
Yes but without this setting I have same behaviour. This was just a try to solve the problem without success.

MichaelDvP, 2024-11-18:
Oh, just tested, if i use the v2 style command i can reproduce a memory leak. 
Try v3 API: URL: `api/boiler/info`

MichaelDvP, 2024-11-18:
Please check 3.7.1-dev.9 from here https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Should work now also with v2 api commands.

proddy, 2024-11-18:
great find @MichaelDvP - I was only testing v3 style.

commit b7548476 (2024-11-19): Merge pull request #2222 from MichaelDvP/dev

fix #2216, add #2212, #2217
closed 2024-11-18 by semmel3k