emsesp/EMS-ESP32 #2218: NTP Status exists in MQTT but not set

VlastiBroucek, 2024-11-18:
### DESCRIPTION

NTP Status entity appears as unavailable in HA. It is in MQTT, but no value is ever set? Started with dev 8

[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
    "system": {
        "version": "3.7.1-dev.8",
        "uptime": "000+00:08:04.659",
        "uptimeSec": 484,
        "platform": "ESP32",
        "cpuType": "ESP32-D0WD-V3",
        "arduino": "ESP32 Arduino v2.0.17",
        "sdk": "v4.4.7-dirty",
        "freeMem": 159,
        "maxAlloc": 107,
        "freeCaps": 4184,
        "usedApp": 2383,
        "freeApp": 3313,
        "partition": "app0",
        "resetReason": "Software reset CPU / Software reset CPU",
        "psram": true,
        "psramSize": 4093,
        "freePsram": 4060,
        "model": ""
    },
    "network": {
        "network": "Ethernet",
        "hostname": "ems-esp",
        "TxPowerSetting": 0,
        "staticIP": false,
        "lowBandwidth": false,
        "disableSleep": true,
        "enableMDNS": true,
        "enableCORS": false,
        "APProvisionMode": "disconnected",
        "APSecurity": "wpa2",
        "APSSID": "ems-esp"
    },
    "ntp": {
        "NTPStatus": "connected",
        "enabled": true,
        "server": "time.cloudflare.com",
        "tzLabel": "Australia/Hobart"
    },
    "mqtt": {
        "MQTTStatus": "connected",
        "MQTTPublishes": 288,
        "MQTTQueued": 0,
        "MQTTPublishFails": 0,
        "MQTTConnects": 1,
        "enabled": true,
        "clientID": "esp32-d767b4fc",
        "keepAlive": 60,
        "cleanSession": false,
        "entityFormat": 1,
        "base": "ems-esp",
        "discoveryPrefix": "homeassistant",
        "discoveryType": 0,
        "nestedFormat": 1,
        "haEnabled": true,
        "mqttQos": 0,
        "mqttRetain": false,
        "publishTimeHeartbeat": 60,
        "publishTimeBoiler": 10,
        "publishTimeThermostat": 10,
        "publishTimeSolar": 10,
        "publishTimeMixer": 10,
        "publishTimeWater": 10,
        "publishTimeOther": 60,
        "publishTimeSensor": 10,
        "publishSingle": false,
        "publish2command": false,
        "sendResponse": true
    },
    "syslog": {
        "enabled": false
    },
    "sensor": {
        "temperatureSensors": 3,
        "temperatureSensorReads": 288,
        "temperatureSensorFails": 0
    },
    "api": {
        "APICalls": 0,
        "APIFails": 0
    },
    "bus": {
        "busStatus": "connected",
        "busProtocol": "HT3",
        "busTelegramsReceived": 685,
        "busReads": 249,
        "busWrites": 0,
        "busIncompleteTelegrams": 0,
        "busReadsFailed": 0,
        "busWritesFailed": 0,
        "busRxLineQuality": 100,
        "busTxLineQuality": 100
    },
    "settings": {
        "boardProfile": "E32V2",
        "locale": "en",
        "txMode": 3,
        "emsBusID": 11,
        "showerTimer": false,
        "showerMinDuration": 180,
        "showerAlert": false,
        "hideLed": false,
        "noTokenApi": false,
        "readonlyMode": false,
        "fahrenheit": false,
        "dallasParasite": false,
        "boolFormat": 1,
        "boolDashboard": 1,
        "enumFormat": 1,
        "analogEnabled": false,
        "telnetEnabled": true,
        "maxWebLogBuffer": 100,
        "webLogBuffer": 19,
        "modbusEnabled": false,
        "forceHeatingOff": false,
        "developerMode": true
    },
    "devices": [
        {
            "type": "boiler",
            "name": "Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3",
            "deviceID": "0x08",
            "productID": 95,
            "brand": "",
            "version": "12.16",
            "entities": 66,
            "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
            "handlersFetched": "0x14 0x16 0x33",
            "handlersPending": "0xBF 0xC2 0x1A 0x35 0x2A 0x2E 0x3B",
            "handlersIgnored": "0x23"
        },
        {
            "type": "thermostat",
            "name": "CR50/RC200, CW100, CR120, CR50",
            "deviceID": "0x18",
            "productID": 157,
            "brand": "",
            "version": "43.02",
            "entities": 46,
            "handlersReceived": "0x06 0xA2 0x031D",
            "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A",
            "handlersPending": "0xA3 0x12 0x13 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0x0267",
            "handlersIgnored": "0xBF"
        },
        {
            "type": "controller",
            "name": "HT3",
            "deviceID": "0x09",
            "productID": 95,
            "brand": "",
            "version": "12.16",
            "entities": 0
        },
        {
            "type": "temperaturesensor",
            "name": "temperaturesensor",
            "entities": 3
        }
    ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-11-18:
good spot, we need to also remove it from HA as in dev-8 we removed it from the heartbeat MQTT topic (it served no purpose). it was discussed here https://github.com/emsesp/EMS-ESP32/pull/427#issuecomment-2478911039

proddy, 2024-11-19:
fixed in latest dev

closed 2024-11-19 by proddy