emsesp/EMS-ESP32 #2219: add comand delay in scheduling on change/condition

johkn, 2024-11-18:
example of an automation in HA

If boiler_heating_active = on
delay xy sec/min
then set boilhystoff = value

my question: would it be possible to add a delay function?

proddy, 2024-11-19:
hmm, been thinking how to do this. It ain't easy!

MichaelDvP, 2024-11-19:
For on_change i have an idea for implementation, but if there is a second change within the delay it get's chaotic and does not work.
For condition-trigger it's really hard to implement because we want it single action and only reactivate on condition change.

proddy, 2024-11-19:
I was thinking of an on_change with a `time+n` in the `condition` that calls another Immediate scheduler command as the `command`

proddy, 2024-11-21:
@johkn Since you're using Home Assistant, why not do the logic there in a trigger? It'll be so much easier, and it's also what HA is designed to do.