emsesp/EMS-ESP32 #2229: Modbus problem with ID server for two MM100 modules

Zembol76, 2024-11-19:
### DESCRIPTION
I have two MM100 modules (json, configuration below). How do I read the data from the second MM100.
Reading from the first ID 7 offset 1000 (HC1) works fine, I cannot read data from the second module, theoretically it should be ID 7 and offset 2000 (HC2) but it does not work.

```json
{
  "system": {
    "version": "3.7.1-dev.8",
    "uptime": "000+01:13:44.248",
    "uptimeSec": 4424,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 162,
    "maxAlloc": 151,
    "freeCaps": 8296,
    "usedApp": 2286,
    "freeApp": 5842,
    "partition": "app1",
    "resetReason": "Power on reset / Power on reset",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 43.9
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-buderus",
    "RSSI": -62,
    "WIFIReconnects": 1,
    "TxPowerSetting": 78,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": true,
    "server": "tempus1.gum.gov.pl",
    "tzLabel": "Europe/Warsaw"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 2721,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-buderus",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 5,
    "publishTimeThermostat": 5,
    "publishTimeSolar": 10,
    "publishTimeMixer": 5,
    "publishTimeWater": 10,
    "publishTimeOther": 5,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 2,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 8529,
    "busReads": 2251,
    "busWrites": 0,
    "busIncompleteTelegrams": 3,
    "busReadsFailed": 6,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "pl",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "webLogBuffer": 14,
    "modbusEnabled": true,
    "forceHeatingOff": false,
    "developerMode": true
  },
  "devices": [
    {
      "type": "boiler",
      "name": "CBS/GB*72, Trendline, Cerapur, Greenstar Si",
      "deviceID": "0x08",
      "productID": 123,
      "brand": "",
      "version": "07.00",
      "entities": 82,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33 0x26 0x28",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B",
      "handlersIgnored": "0x25 0x27 0x2D 0x1E"
    },
    {
      "type": "thermostat",
      "name": "RC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "18.05",
      "entities": 93,
      "handlersReceived": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x21",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D8",
      "handlersFetched": "0x02CE",
      "handlersIgnored": "0x16 0x33 0x0255 0x0292 0x03BE 0x02E2 0x02EC 0xF6 0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x20",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0x03BD 0x0255 0x02E1 0x02EB 0xF6 0xF7 0x16 0x33 0x0291 0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "deviceID": "0x09",
      "productID": 125,
      "brand": "",
      "version": "03.05",
      "entities": 0
    }
  ]
}

Zembol76, 2024-11-19:
I have noticed a pattern.
When the BBQ Gateway S3 is switched off/on (plugged into the boiler).
This module which I read which first one,  can continue to read example
I can read the second MM100 ID 7 offset 2000 but ID7 offset 1000 does not work.
When BBQ Gateway S3 (boiler plug) is switched off/on.
I can read the second MM100 ID 7 offset 1000 but ID7 offset 2000 does not work.

Logs 
ID 7 offset 2000 works, offset 1000 generates this error
2024-11-19 22:24:19.747 ERROR 43: [modbus] Unable to read raw device value flowtemphc for tag=1 - error_code = -1
2024-11-19 22:24:24.748 ERROR 44: [modbus] Unable to read raw device value flowtemphc for tag=1 - error_code = -1
2024-11-19 22:24:29.746 ERROR 45: [modbus] Unable to read raw device value flowtemphc for tag=1 - error_code = -1

ID 7 offset 1000 works, offset 2000 generates this error
2024-11-19 22:33:00.589 ERROR 14: [modbus] Unable to read raw device value flowtemphc for tag=2 - error_code = -1
2024-11-19 22:33:05.594 ERROR 15: [modbus] Unable to read raw device value flowtemphc for tag=2 - error_code = -1
2024-11-19 22:33:10.595 ERROR 16: [modbus] Unable to read raw device value flowtemphc for tag=2 - error_code = -1

Zembol76, 2024-11-20:
It works perfectly. Thank you very much.
By the way.
There is a small difference (not a huge error) between the documentation and the physical reading of the registers
Example
For the MM100, according to the documentation (https://docs.emsesp.org/Modbus-Entity-Registers/#mm100) flowtempoffset has a register offset of 6
In reality the offset is 7 (as in MM200)
Example log (reading IDs 7 register 1006):
2024-11-20 23:32:32.838 ERROR 70: [modbus] Unable to read raw device value flowtempvf for tag=1 - error_code = -1
2024-11-20 23:32:37.841 ERROR 71: [modbus] Unable to read raw device value flowtempvf for tag=1 - error_code = -1

proddy, 2024-11-21:
Documentation has been updated. @Zembol76 please feel free to close this if it all works, or comment back on further changes needed

closed 2024-11-21 by Zembol76