emsesp/EMS-ESP32 #2230: IMPROVEMENT? WiFi Reconnects is redundant for ethernet based systems

VlastiBroucek, 2024-11-20:
### DESCRIPTION

I am using BBQKeys V2 connected to the ethernet.

system_wifireconnects is defined for HA in MQTT

```json
{
  "uniq_id": "system_wifireconnects",
  "obj_id": "system_wifireconnects",
  "name": "WiFi reconnects",
  "stat_t": "ems-esp/heartbeat",
  "val_tpl": "{{value_json['wifireconnects'] if value_json['wifireconnects'] is defined else 0}}",
  "avty": [
    {
      "t": "ems-esp/heartbeat",
      "val_tpl": "{{'online' if value_json['wifireconnects'] is defined else 'offline'}}"
    }
  ],
  "avty_mode": "all",
  "ent_cat": "diagnostic",
  "ic": "mdi:counter",
  "stat_cla": "measurement",
  "dev": {
    "name": "ems-esp",
    "ids": [
      "ems-esp"
    ]
  }
}

homeassistant/sensor/ems-esp/system_wifireconnects/config

```
however, since being on ethernet, it is then not included in heartbeat (?), hence producing unavailable entity in HA.

```json
{
  "bus_status": "connected",
  "uptime": "000+04:31:00.471",
  "uptime_sec": 16260,
  "rxreceived": 12761,
  "rxfails": 0,
  "txreads": 3157,
  "txwrites": 0,
  "txfails": 0,
  "mqttcount": 5302,
  "mqttfails": 0,
  "mqttreconnects": 0,
  "apicalls": 0,
  "apifails": 0,
  "sensorreads": 9753,
  "sensorfails": 0,
  "freemem": 162,
  "max_alloc": 107
}

ems-esp/heartbeat

```

I realise this might be tricky, but ideally, on systems that are using ethernet, the variable/entity should not be defined at all.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-11-20:
I did consider that, and you're right as it is tricky. On Ethernet enabled boards you can also run WiFi and this can be configured real-time in the EMS-ESP settings so the check would need to be at start-up to only when there is a valid ssid set for example.

MichaelDvP, 2024-11-20:
I think it's only to skip the HA create, see https://github.com/MichaelDvP/EMS-ESP32/commit/de7bd38850502ebadaa06fa0649b7510d58889cc
This is called on every mqtt connect, so changing the connection should work. The heartbeat already checks eth connection status.

VlastiBroucek, 2024-11-20:
You guys rock!

closed 2024-11-22 by proddy