emsesp/EMS-ESP32 #2232: BBQKees S32 V2 keeps forgetting settings since v3.6

M1SJE, 2024-11-20:
### DESCRIPTION

My S32 V2 keeps forgetting it's settings, sometimes MQTT, sometimes WIFI, sometimes everything.

Seeing as this has been reported multiple times across multiple people for years, is there an actual fix?

I've reflashed the firmware more times than I care to remember (due to the partition change) and everytime I change the settings and restart, settings get lost.

### REQUESTED INFORMATION

[emsesp_system_info (1).json](https://github.com/user-attachments/files/17830022/emsesp_system_info.1.json)

### TO REPRODUCE

Restart module.

### EXPECTED BEHAVIOUR

For it not to lose settings.

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

Works fine when you don't restart it, and will do so for months, but the second a power cut happens, or you accidentally pull the plug, you now have to spend 5-10 minutes setting it back up.

That's if you even realise that it's forgot its settings because you weren't aware there was a power cut, and the 1st time you find out is possibly days later while Home Assistant just keeps repeating the same stale data.

M1SJE, 2024-11-20:
Following the advice from https://github.com/emsesp/EMS-ESP32/issues/1930#issuecomment-2433253001

[flashlog.txt](https://github.com/user-attachments/files/17830967/flashlog.txt)

Configured Wifi and MQTT only, powered with USB and not connected to boiler.

[emsesp_system_info (1).txt](https://github.com/user-attachments/files/17831025/emsesp_system_info.1.txt)

Pressed restart button, MQTT settings lost.

[emsesp_system_info (2).txt](https://github.com/user-attachments/files/17831029/emsesp_system_info.2.txt)

So I reconfigured MQTT, then yanked the power, no restart button... And it seems to have stuck this time. Multiple restarts and it's holding onto it.

It's plugged back into the boiler now so I'll keep an eye on it. But I've had this before where it seems to be fine, but will then forget again.

We'll see though. :)

proddy, 2024-11-20:
This issue is annoying but very uncommon and only seems to happen on BBQKees S32 boards. I think it's perhaps a bootloader timing issue where the filesystem tries to mount before it's ready. What I suggest is that you send the board back so we can run some tests (since you're the only one now to have a reproducible case) and see if you can get a nice discount on an S3 board, which will set you up for the future.

closed 2024-11-20 by M1SJE

M1SJE, 2024-11-20:
Well it totally lost the settings now, so I'll just remove it from the network.

I'm also not the only person, I linked to an issue where someone is having the same issue.