emsesp/EMS-ESP32 #2242: Confusing MQTT Status

VlastiBroucek, 2024-11-23:
### DESCRIPTION

Being myself former developer, I sort of understand ``Connected (0)``, but I think it might be rather confusing for an "average" user:

[image]

Not sure if it was like that before, this is on v3.7.1-dev.10, but I think I would have noticed...

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-11-23:
It's kinda always been there. We just suppressed showing the number of reconnects if it was 0. Now with the new "Track number of WiFi reconnects" #2207 we also show if it's zero to be consistent.

Do you have a better suggestion?

MichaelDvP, 2024-11-23:
It shows the number of mqtt-reconnects. Before it was number of connects starting with 1 and shown only for values >1.
With #2207 the wifi network always shows the number of reconnects starting with 0, so mqtt was adapted to same logic.