emsesp/EMS-ESP32 #2245: Unable to enable Modbus 3.7.1-dev.10

Zembol76, 2024-11-23:
### DESCRIPTION

After uploading 3.7.1-dev.10 such errors 
2024-11-23 19:56:06.000 ERROR 7: [modbus] Error in modbus parameters: temp must be listed before requiredtemp.
2024-11-23 19:56:06.000 ERROR 8: [modbus] Unable to enable Modbus - the parameter list order is corrupt. This is a firmware bug.
In 3.7.1-dev.9 everything is ok

MichaelDvP, 2024-11-24:
I've fixed in https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest and made a PR.

We have added the water devices in dev10. SM100 can only act as DHW3 circuit, but modbus script generation needs all mapped to DHW1, so it has generated doulettes.

Zembol76, 2024-11-24:
Ok works, tested with plc s7 1200 Siemens.

closed 2024-11-24 by Zembol76