emsesp/EMS-ESP32 #2247: Last Will (LWT) not set on MQTT Connect

andatche, 2024-11-24:
### DESCRIPTION

I recently upgraded from 3.6.2 to 3.7.0 and since, EMS-ESP is not setting the Last Will properties on the MQTT Connect command. As a result, Home Assistant continues to perceive it as online after disconnecting/crashing etc.

I tested downgrading to 3.6.5 and that appears to be affected too.

Example PCAP from a connection attempt:
```
Frame 9: 118 bytes on wire (944 bits), 118 bytes captured (944 bits)
Ethernet II, Src: <snip> (<snip>), Dst: <snip> (<snip>)
Internet Protocol Version 4, Src: 192.168.5.2, Dst: 192.168.1.10
Transmission Control Protocol, Src Port: 57354, Dst Port: 1883, Seq: 1, Ack: 1, Len: 64
MQ Telemetry Transport Protocol, Connect Command
    Header Flags: 0x10, Message Type: Connect Command
        0001 .... = Message Type: Connect Command (1)
        .... 0000 = Reserved: 0
    Msg Len: 62
    Protocol Name Length: 4
    Protocol Name: MQTT
    Version: MQTT v3.1.1 (4)
    Connect Flags: 0xc2, User Name Flag, Password Flag, QoS Level: At most once delivery (Fire and Forget), Clean Session Flag
        1... .... = User Name Flag: Set
        .1.. .... = Password Flag: Set
        ..0. .... = Will Retain: Not set
        ...0 0... = QoS Level: At most once delivery (Fire and Forget) (0)
        .... .0.. = Will Flag: Not set
        .... ..1. = Clean Session Flag: Set
        .... ...0 = (Reserved): Not set
    Keep Alive: 60
    Client ID Length: 7
    Client ID: ems-esp
    User Name Length: 7
    User Name: ems-esp
    Password Length: 32
    Password: <snip>
```

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
  "system": {
    "version": "3.7.0",
    "uptime": "000+00:29:19.658",
    "uptimeSec": 1759,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 180,
    "maxAlloc": 107,
    "freeCaps": 130,
    "usedApp": 1730,
    "freeApp": 126,
    "partition": "app0",
    "resetReason": "Power on reset / APP CPU reset by PRO CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -58,
    "TxPowerSetting": 78,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": false,
    "enableCORS": false,
    "APProvisionMode": "never",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "cr1.caw.andatche.com",
    "tzLabel": "Europe/London"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 720,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTConnects": 1,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 0,
    "publishTimeThermostat": 0,
    "publishTimeSolar": 0,
    "publishTimeMixer": 0,
    "publishTimeWater": 0,
    "publishTimeOther": 0,
    "publishTimeSensor": 0,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 2,
    "temperatureSensorReads": 702,
    "temperatureSensorFails": 0,
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 1230,
    "busReads": 214,
    "busWrites": 29,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32",
    "locale": "en",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 25,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "GB*72, Trendline, Cerapur, Greenstar Si",
      "deviceID": "0x08",
      "productID": 123,
      "brand": "Worcester",
      "version": "06.02",
      "entities": 83,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33 0x26 0x28",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B",
      "handlersIgnored": "0x24 0x25 0x27"
    },
    {
      "type": "controller",
      "name": "Controller",
      "deviceID": "0x09",
      "productID": 152,
      "brand": "",
      "version": "02.09",
      "entities": 0
    },
    {
      "type": "temperaturesensor",
      "name": "temperaturesensor",
      "entities": 2
    }
  ]
}
```

### EXPECTED BEHAVIOUR

EMS-ESP should set the `Will Topic`, `Will Payload`, `Will Retain` etc. properties on the Connect message appropriately.

proddy, 2024-11-24:
thanks for reporting, never knew it didn't work! I made a fix in dev-12. You can pull it from https://github.com/proddy/EMS-ESP32/releases/tag/latest or wait until the PR is approved and merged.

andatche, 2024-11-25:
Awesome. Thanks for such a quick fix!

I've upgraded to 3.7.1-dev.12 and confirmed that LWT is now being set correctly on connect.

```
Frame 20: 143 bytes on wire (1144 bits), 143 bytes captured (1144 bits)
Ethernet II, Src: <snip> (<snip>), Dst: <snip> (<snip>)
Internet Protocol Version 4, Src: 192.168.5.2, Dst: 192.168.1.10
Transmission Control Protocol, Src Port: 56090, Dst Port: 1883, Seq: 1, Ack: 1, Len: 89
MQ Telemetry Transport Protocol, Connect Command
    Header Flags: 0x10, Message Type: Connect Command
    Msg Len: 87
    Protocol Name Length: 4
    Protocol Name: MQTT
    Version: MQTT v3.1.1 (4)
    Connect Flags: 0xee, User Name Flag, Password Flag, Will Retain, QoS Level: At least once delivery (Acknowledged deliver), Will Flag, Clean Session Flag
    Keep Alive: 60
    Client ID Length: 7
    Client ID: ems-esp
    Will Topic Length: 14
    Will Topic: ems-esp/status
    Will Message Length: 7
    Will Message: 6f66666c696e65
    User Name Length: 7
    User Name: ems-esp
    Password Length: 32
    Password: <snip>
```

closed 2024-11-25 by andatche