emsesp/EMS-ESP32 #2254: Leak memory when modbus TCP is activated.

Zembol76, 2024-11-26:
### DESCRIPTION

Firmware 3.7.1-dev.12  Platform ESP32S3-16MB+

Switching on the modbus causes a systematic loss of memory. I did not send any modbus queries (I switched off the communication to check if the problem was not on my side), in my case, just switching on causes memory leaks.

It is likely that this problem was also present in previous versions, but a similar issue was reported (related to API, memory leak).

### TO REPRODUCE

modbus TCP is activated.

### SCREENSHOTS

[image]
[image]
[image]

proddy, 2024-11-27:
It looks like a very slow leak, ~30KB heap lost in 24hrs.

Could you send over the required System Info so we can see what else is running on your setup to accurately reproduce this?

@mheyse we may need your help on this one.

Zembol76, 2024-11-27:
Me setup:
{
  "system": {
    "version": "3.7.1-dev.12",
    "uptime": "000+01:10:13.359",
    "uptimeSec": 4213,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 163,
    "maxAlloc": 151,
    "freeCaps": 8298,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 40.9
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-buderus",
    "RSSI": -62,
    "WIFIReconnects": 0,
    "TxPowerSetting": 78,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": " 194.146.251.100 ",
    "tzLabel": "Europe/Warsaw"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 2597,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-buderus",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 5,
    "publishTimeThermostat": 5,
    "publishTimeSolar": 10,
    "publishTimeMixer": 5,
    "publishTimeWater": 10,
    "publishTimeOther": 5,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 8068,
    "busReads": 2184,
    "busWrites": 0,
    "busIncompleteTelegrams": 3,
    "busReadsFailed": 4,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "pl",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 12,
    "modbusEnabled": true,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "CBS/GB*72, Trendline, Cerapur, Greenstar Si",
      "deviceID": "0x08",
      "productID": 123,
      "brand": "Buderus",
      "version": "07.00",
      "entities": 82,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33 0x26 0x28",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B",
      "handlersIgnored": "0x1E"
    },
    {
      "type": "thermostat",
      "name": "RC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "18.05",
      "entities": 93,
      "handlersReceived": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x20",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0xBF 0x16 0x33 0x0255 0x0291 0x02E1 0x02EB"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x21",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D8",
      "handlersFetched": "0x02CE",
      "handlersIgnored": "0x16 0x33 0x0255 0x0292 0x02E2 0x02EC 0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "deviceID": "0x09",
      "productID": 125,
      "brand": "",
      "version": "03.05",
      "entities": 0
    }
  ]
}

mheyse, 2024-11-27:
I'm running and using Modbus/TCP for some months now, reading registers about once per minute, without noticing any issues. However, I did not track memory usage. I'm still on `3.7.0-dev.38`. Current uptime is 18 days, so for me it is definitely less than 30k/day or the system would have run out of memory by now.

I'll have a look at the code to check if I missed any dynamic memory allocations. If no Modbus reads/writes are made, none of the code should run though except for the initialization, it's all modbus callbacks.

Here my system info for comparison:

```
"system": {
    "version": "3.7.0-dev.38",
    "uptime": "018+01:50:17.956",
    "uptimeSec": 1561817,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 145,
    "maxAlloc": 103,
    "freeCaps": 4167,
    "usedApp": 2333,
    "freeApp": 5795,
    "partition": "app1",
    "resetReason": "Power on reset / APP CPU reset by PRO CPU",
    "psram": true,
    "psramSize": 4093,
    "freePsram": 4057,
    "model": ""
  }
  ```

MichaelDvP, 2024-11-27:
@Zembol76 Please post another system info at a state with lower memory left (after some hours). You are using very short mqtt intervals of 5 sec, check if mqtt is stable or sometimes the queue fills (eating memory). 

@proddy I think this is 30kB per hour, but i don't think this is linear. 

I'm not using modbus, but have enabled the service and can't see any memory leak.

@mheyse Do you have a suggestion how to test, I've found [openModScan](https://github.com/sanny32/OpenModScan), but don't know what i need to configure to test emsesp-modbus.

Zembol76, 2024-11-30:
Welcome,
What I have done yet : 
1. uploaded firmware 3.7.2-dev 0
2. factory reset
3. new configuration (minmalistic, no mqtt, only modbus)
4. memory problem is still there (does not perform any queries: api, mqtt, modbus)
5. in 7-8 hours all memory has been used up 

{
  "system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+00:18:42.771",
    "uptimeSec": 1122,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 192,
    "maxAlloc": 179,
    "freeCaps": 8326,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Power on reset / Power on reset",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -66,
    "WIFIReconnects": 1,
    "TxPowerSetting": 0,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": false,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "disconnected",
    "enabled": false,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 0,
    "base": "ems-buderus",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 2518,
    "busReads": 679,
    "busWrites": 0,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 5,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "pl",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 50,
    "webLogBuffer": 26,
    "modbusEnabled": true,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "CBS/GB*72, Trendline, Cerapur, Greenstar Si",
      "deviceID": "0x08",
      "productID": 123,
      "brand": "",
      "version": "07.00",
      "entities": 82,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33 0x26 0x28",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B",
      "handlersIgnored": "0x25 0x27 0x2D 0x1E"
    },
    {
      "type": "thermostat",
      "name": "RC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "18.05",
      "entities": 93,
      "handlersReceived": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x21",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D8",
      "handlersFetched": "0x02CE",
      "handlersIgnored": "0x03BE 0x0255 0x02E2 0x02EC 0xF6 0x16 0x33 0x0292 0xBF"
    },
    {
      "type": "mixer",
      "name": "MM100/MM100",
      "deviceID": "0x20",
      "productID": 160,
      "brand": "",
      "version": "24.05",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0x0255 0x03BD 0x02E1 0x02EB 0x16 0x33 0x0291 0xF6 0xF7 0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "deviceID": "0x09",
      "productID": 125,
      "brand": "",
      "version": "03.05",
      "entities": 0
    }
  ]
}
[image]

MichaelDvP, 2024-11-30:
@Zembol76 Please post another system info at a state with lower memory left (after some hours)

Zembol76, 2024-11-30:
another system info - what do you mean?

MichaelDvP, 2024-11-30:
What you have posted: 
```
{
"system": {
"version": "3.7.2-dev.0",
"uptime": "000+00:18:42.771",
"uptimeSec": 1122,
"platform": "ESP32S3",
"cpuType": "ESP32-S3",
"arduino": "ESP32 Arduino v2.0.17",
"sdk": "v4.4.7-dirty",
"freeMem": 192,
"maxAlloc": 179,
"freeCaps": 8326,
````

is with full mem and 18 minutes after start. We need a second complete system info (help-page, download) to see what happend.
I have setup a single esp32s3 chip with 3.7.2-dev.0, no bus connection and let it run with modbus enabled. No leak withing the first half hour. 
To find the leak we have to know where to search and reproduce it.

Zembol76, 2024-11-30:
After 2 hours (i continue to observe).

"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+00:00:11.218",
    "uptimeSec": 11,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 209,
    "maxAlloc": 195,
    "freeCaps": 8380,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8172,
    "model": "",
    "temperature": 41.9
  }
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+00:59:08.758",
    "uptimeSec": 3548,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 175,
    "maxAlloc": 163,
    "freeCaps": 8309,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
{
  "system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+02:00:26.310",
    "uptimeSec": 7226,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 151,
    "maxAlloc": 139,
    "freeCaps": 8285,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },

MichaelDvP, 2024-11-30:
Please the complete info.

Zembol76, 2024-11-30:
Maximum time (and results every hour):
‘uptime": ‘000+07:47:03.553’

"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+00:00:11.218",
    "uptimeSec": 11,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 209,
    "maxAlloc": 195,
    "freeCaps": 8380,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8172,
    "model": "",
    "temperature": 41.9
  }
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+00:59:08.758",
    "uptimeSec": 3548,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 175,
    "maxAlloc": 163,
    "freeCaps": 8309,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
{
  "system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+02:00:26.310",
    "uptimeSec": 7226,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 151,
    "maxAlloc": 139,
    "freeCaps": 8285,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+03:00:27.236",
    "uptimeSec": 10827,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 126,
    "maxAlloc": 115,
    "freeCaps": 8261,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 41.9
  },
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+04:00:41.400",
    "uptimeSec": 14441,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 104,
    "maxAlloc": 91,
    "freeCaps": 8238,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 41.9
  },
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+05:03:53.367",
    "uptimeSec": 18233,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 79,
    "maxAlloc": 65,
    "freeCaps": 8213,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+06:11:19.018",
    "uptimeSec": 22279,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 55,
    "maxAlloc": 40,
    "freeCaps": 8189,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
},
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+06:58:02.073",
    "uptimeSec": 25082,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 40,
    "maxAlloc": 26,
    "freeCaps": 8174,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 42.9
  },
"system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+07:47:03.553",
    "uptimeSec": 28023,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 23,
    "maxAlloc": 7,
    "freeCaps": 8158,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 43.9
  },

proddy, 2024-12-01:
@Zembol76 what we need is the full system into after 2hrs, so we can see how many API calls, MQTT calls, Tx/Rx line quality, telegrams sent etc, not just memory. It will help us trace what is causing the leak. It would also be good to compare your setup with @mheyse 's.

Zembol76, 2024-12-01:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SYSTEM INFO

  "system": {
    "version": "3.7.2-dev.0",
    "uptime": "000+02:05:14.980",
    "uptimeSec": 7514,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 147,
    "maxAlloc": 135,
    "freeCaps": 8282,
    "usedApp": 2292,
    "freeApp": 5836,
    "partition": "app1",
    "resetReason": "Power on reset / Power on reset",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 43.9
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -61,
    "WIFIReconnects": 1,
    "TxPowerSetting": 0,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": false,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "disconnected",
    "enabled": false,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 0,
    "base": "ems-buderus",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
[... 262 lines cut ...]
  "type": "customizations",
  "Customizations": {
    "ts": [],
    "as": [],
    "masked_entities": []
  }
}

ENTITIES

{
  "type": "entities",
  "Entities": {
    "entities": []
  }
}

SCHEDULE

{
  "type": "schedule",
  "Schedule": {
    "schedule": []
  }
}

mheyse, 2024-12-01:
> @mheyse Do you have a suggestion how to test, I've found [openModScan](https://github.com/sanny32/OpenModScan), but don't know what i need to configure to test emsesp-modbus.

I was hoping the documentation is sufficient to answer what needs to be configured :-) I'd really appreciate your feedback about what is missing and how the docs can be improved!

I was testing with a small C# program using the NModbus library (because then I can read whatever registers I want, tools like OpenModScan are very limited in that they can only read continuous register blocks from a single device), but here is what you need to do for example to read `hc1 current room temperature` from a RC310 remote using OpenModScan:

1. Connection -> Connect, configure ESP32 IP address

[image]

2. Configure register settings:
  - Set `Device ID` to `6` (for thermostat, see https://docs.emsesp.org/Modbus-Server-IDs/)
  - `Address Base` -> `0-based`
  - `Start Address` -> 1001 (see https://docs.emsesp.org/Modbus-Register-Blocks/ and https://docs.emsesp.org/Modbus-Entity-Registers/#rc30-moduline-30001010h-cw400-sense-ii-hpc410)
  - `Length` -> 1 (just read a single register)
  - `MODBUS Point Type` -> `04: INPUT REGISTER` (see https://docs.emsesp.org/Modbus/#reading-and-writing-entity-values)

3. That's it, it should continuously read the value. In my case it displays

```
31001: <  206>
```

The 206 needs to be multiplied with the scale factor 0.1 for this register, so it is 20.6°C (see https://docs.emsesp.org/Modbus-Entity-Registers/#rc30-moduline-30001010h-cw400-sense-ii-hpc410).

[image]

mheyse, 2024-12-01:
>     "freeMem": 145,

I just checked again, `freeMem` did not change on my system (full system info attached) in the last 4 days. Not sure what is going on. When I have time I'll upgrade to `3.7.1-dev.12` to see if this changes anything.

[emsesp_system_undefined (5).txt](https://github.com/user-attachments/files/17970286/emsesp_system_undefined.5.txt)

Zembol76, 2024-12-01:
Hi,
I decided to check once again that the problem isn't on my end. Today, installed a switch (with port mirroring) and a dedicated AP for the BBQ Geteway S3. I used Wireshark to search for packets on port 502. I found a device in my network that was sending Modbus TCP requests.

I would like to express my sincere gratitude and apologize for taking up your time on a non-existent issue.

By the way, you can add modbus TCP information in the Status / Data Traffic menu.
[image]

proddy, 2024-12-02:
Hang on, I did look like the ems-esp was loosing memory, maybe accidentally, but it shouldn't. Maybe we need to add a fail safe to the code for rogue incoming tcp packets not intended for this client?

MichaelDvP, 2024-12-04:
@mheyse 
>I'd really appreciate your feedback about what is missing and how the docs can be improved!

Thanks for the detailed instruction for using openmodscan. I have not used modbus before, so get lost about the many different numbers. With this simple example as dummy instruction it works now. 
I also tested with https://github.com/ioBroker/ioBroker.modbus, working too.

So far:
+ got error logs for all tries to misconfigure a value request
+ got debug logs for each request (compiled with debug enabled)
+ no memory leak after an hour.

I case we get similar issues we can send the user a ems-esp(debug) to log the modbus. This should show all requests.
I have not tested with non-modbus-packages to the port, maybe this  causes the leak?

@Zembol76 
> I found a device in my network that was sending Modbus TCP requests.

Please specify what was sent by this device, so we can reproduce.

MichaelDvP, 2024-12-04:
When playing with modbus parameters i've seen the memory leak too: 
When setting poll intervall to 10 sec and default ems-esp modbus timeout is 10 sec, it give arbitary a disconnect/reconnect and memory goes down. Did not see this with with different values, seems to be a race condition if next poll comes in within timeout disconnect procedure. 
I think this is in emodbus library, difficult to find/fix. Maybe we should make the default timeout a bit longer, 10 sec seems very short for a slow changing thing like heating. 
But with normal parameters all seems ok.

mheyse, 2024-12-04:
The `timeout` parameter specifies when eModbus closes the connection if no new request is sent by the client within this time (https://emodbus.github.io/modbusserver-tcp#bool-startuint16_t-port-uint8_t-maxclients-uint32_t-timeout), so it is not directly related to an entity change rate, but it is  related to the polling rate.

How did you test, what software do you use? If I use Open ModScan to poll every 11 seconds the second request just fails with an (expected) error "The remote host closed the connection" and it does not try to re-establish the connection.

MichaelDvP, 2024-12-04:
I've seen this with ioBroker and poll-time 10 sec. Sometimes the connection is closed and ioBroker reconnects, sometimes not. But over time the the memory goes down. 
With openModScan i can't reproduce. After a connection closes i have to reconnect manualy, makes it nasty to test over long time.
For poll-time > timeout (e.g. 11sec, 10sec) i see only disconnect/reconnect and no memory issue, 
For poll-time < timeout connection stays and no memory issue. 
Only for poll-time = timeout i have seen it once, but takes a long time to test and reproduce, have not done a second test yet.

mheyse, 2024-12-04:
I'm now running this small C# program to test. It reads a value every 10.1 seconds (which at least on my setup sometimes fails and sometimes succeeds), automatically reconnecting if disconnected by server. Let's see what happens with memory.

```csharp
using System.Net.Sockets;
using NModbus;

const string address = "wp.mh-internal.net"; // EMS-ESP address
const int port = 502;                        // modbus port
const byte serverId = 5;                     // boiler
const int startAddress = 7;                  // curflowtemp @WLW196i
const int dataLength = 1;                    // read 1 register
const int readIntervalMillis = 10100;        // read interval

var factory = new ModbusFactory();
var tcpClient = new TcpClient(address, port);
var modbusClient = factory.CreateMaster(tcpClient);

for (;;)
{
    try
    {
        var delayTask = Task.Delay(readIntervalMillis);
        var data = await modbusClient.ReadInputRegistersAsync(serverId, startAddress, dataLength);
        Console.WriteLine($"{data[0]}");
        await delayTask;
    }
    catch (IOException e)
    {
        Console.WriteLine($"{e.Message} - reconnecting");
        modbusClient.Dispose();
        tcpClient.Close();
        tcpClient.Dispose();
        tcpClient = new TcpClient(address, port);
        modbusClient = factory.CreateMaster(tcpClient);
    }
}
```

mheyse, 2024-12-04:
OK, I can confirm the issue. After 1 hour I permanently lost approx. 25kB of RAM. I see the following options:
- Upgrade the eModbus library (1.7.0 to 1.7.2) and see if this resolves the issue although I don't see anything obvious in the change log (https://github.com/eModbus/eModbus/commits/v1.7.2stable)
- switch to a different modbus library (probably quite a lot of work)
- set the timeout to something very low or very high
- try something hacky like periodically restarting the modbus server and see if this works around the issue

MichaelDvP, 2024-12-04:
I've made a version to test:
- update eModbus to v.1.7.2
- fix API prefix for commands
- set default timeout to 300 sec, change timeout setting to seconds (older saved values will give 10000 sec = 2,75 h)

Check https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

The test version also update AsyncTCP library to a threadsafe  v3.2.14

I think this is still mainly a misconfiguration. We should keep this in mind and can give tips to avoid the leak, but for well configured system this is no issue.

Zembol76, 2024-12-04:
@proddy 
@mheyse 
@MichaelDvP 
Gentlemen, I have a moment now to reply. The device I was talking about is a Raspberry Pi with Node-RED. The luck or misfortune (it helped to detect this memory leak) was that in the test flow, I had a configured Modbus server with an IP address the same as the BBQ V3 address. Unfortunately, I am unable to reproduce this configuration. The only thing I remember for sure is that the query time was set to 20 seconds.

closed 2024-12-19 by proddy