emsesp/EMS-ESP32 #2267: After adding many custom entities, the complete list was lost

proddy, 2024-12-01:
As reported on [Discord](https://discord.com/channels/816637840644505620/816958087160266775/1312370816415043696) by JHa

> Just for your info - because I dont think i can create a "reproducable" issue on github: Today I added some individual entities in 3.7.2-dev.0 according to "watch unknown" output  just to try to see the values and to see if I can correlate found values with the real world. after I entered one (I think it was approx the 9th or 10th new entry) suddenly they all have gone and I dont't know why. 

Let's see if we can reproduce, with and without the Apply/Save in between adding each new entry.

gh47110815, 2024-12-01:
Ok, I tried to reproduce (still with 3.7.2-dev.0) and produced the following state as shown with screenshots ...
[image]
[image]

Now, if I click on "custom entities" I am not able to see the entities (e.g. to add/delete or change one). If I click on "Geräte" (Devices) I see **14/13** Custom Entities .... see screenshots .... maybe due to duplicates.

Maybe helpful - and most important: I don't think I did everything write ;-) I am pretty sure that I configured entities, which not necessarily do really fit the real world inside my Bosch Hybrid System.

proddy, 2024-12-01:
so those custom entities are shown in the Dashboard and Devices pages, and you also see them when you download the JSON file from Upload/Download but they're not showing in the Custom Entities Module ?

MichaelDvP, 2024-12-01:
@gh47110815 Have you tried `Shift-CTRL-R`?

proddy, 2024-12-01:
@gh47110815 glad it works, and you did nothing wrong. the browser should automatically refresh and this is an issue we have open and looking into (https://github.com/emsesp/EMS-ESP32/issues/2162).

MichaelDvP, 2024-12-01:
> I assume you wanted me to reload the browser .... yes this worked - thanks.

Yes, reload page and clear cache entries for this pages. I also often ran in to this issue that parts of 3.6.5 are still cached on disk and the browser takes them after a ems-esp restart.

gh47110815, 2024-12-01:
Maybe NOT only a cache issue. Seems, that I was able to reproduce the situation again: No custom entities  have been listed again - but on download the json file contained a list of them and under devices -> custom entities have also been available. I double checked again ... I restarted the client (my macBook) and opened safari, chrome and firefox. NONE of the browsers listed the custom entities. see screenshot .... (and meanwhile i have updated to latest 3.7.2-dev.1) - then I restarted the E32V2 - and the custom entities are listed again. Good info: The reboot of the E32V2 fixed the issue and editing of the entities is possible again. While typing these lines - the entities disappeared again. ok .... rebooting again with log set to all .... immediately after reboot - the entities are listed - then after approx 1 min. they disappeared again.
**Maybe** a wrong configured entity is causing this "display" issue as soon as the first telegram is received? 
Logfile (role ALL after the reboot) is added also ....
[241201-2108-log.txt](https://github.com/user-attachments/files/17970214/241201-2108-log.txt)

[241201-2048-emsesp_entities.json](https://github.com/user-attachments/files/17970153/241201-2048-emsesp_entities.json)
[image]
[image]

gh47110815, 2024-12-01:
Update approx 1 hour later - without changing or touching anything - just having another look into the ESP ..... the "individuellen Entitäten" are visible again and they contain a  list of 18 entities which are editable again. 

For me it looks like "faulty" entries I have created contain some errors which lead to this strange behavior in the GUI.

proddy, 2024-12-01:
This looks like a bug. Could you attach the custom_entities.json exported file here so I can simulate your setup. Also, so I'm crystal clear, the custom entities are in EMS-ESP and working, but just not shown in the 'individuellen Entitäten' web page. In the Dashboard and Devices/Gerate they are always shown?

We'll also take a look into 21.474.836.

gh47110815, 2024-12-01:
"Could you attach the custom_entities.json exported file ...." - done (see emsesp_entities-3.json attached)
"Also, so I'm crystal clear, the custom entities are in EMS-ESP and working, but just not shown in the 'individuellen Entitäten' web page. In the Dashboard and Devices/Gerate they are always shown?" - YES - that's right - they are sometimes shown in  "individuellen Entitäten" and sometimes they are NOT showing there.

[emsesp_entities-3.json](https://github.com/user-attachments/files/17970430/emsesp_entities-3.json)

Please also have in mind: I have just "tried" to throw some light onto the "watch unknown" outputs from my installation. So it's trail & error on my side (I hoped I will see values which I can then somehow align to values I will see on the "BoschCom App" or on the Console of my Bosch GC7000) and I personally think, that this is the real issue and a "wrong way". 
I think I SHOULD NOT "trail & error" with entries, I do not really understand - like setting "offsets" which are just a "best guess". 

Your suggested way ("Have a concrete value from the system you are missing in the esp and try to find it using the "watch unknown" function to let it be integrated in a next release) seems to be the much better way than just "playing around" like I did.

MichaelDvP, 2024-12-02:
Could you please make a log (log all) of > 1 minute to see the telegram fetch for these entities and post the output of
`http://ems-esp.local/api/custom` and  `http://ems-esp.local/api/custom/entities`

gh47110815, 2024-12-02:
@MichaelDvP : done as zip....

[241202-ems-esp.zip](https://github.com/user-attachments/files/17980837/241202-ems-esp.zip)

MichaelDvP, 2024-12-03:
Thanks, the output of api looks ok. Also no values of 21.474.836
I think we have a issue with the large values. Devices uses messagepack and we transform the numbers to float, seems a calculation limit to  21.474.836, maybe we should change Helpers::transformNumberFloat to double
In custom entities page the number is the same as output of API and this seems to crash the number rendering in the react script, i don't know why.

MichaelDvP, 2024-12-03:
The empty page is a ems-esp crash when the number gets too long. The Helpers::renderValue()-buffer has to be enlarged.
It's a bug from adding the UINT32, before we only had 24 bit values.

MichaelDvP, 2024-12-03:
Please test with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

closed 2024-12-19 by proddy