emsesp/EMS-ESP32 #2268: Add additional heating types if external remote is installed

SpaceTeddy, 2024-12-01:
### DESCRIPTION

If an external Remote is addtached to the controller i.E. RC20 or es72 ... you can choose with the Sieger BK15 boiler and installed es73 aka RC30 controller two additional heating types. Standard heating types are: radiator, floor, off. The two additional are "Room Flow" and "Room Load". 
These two I can choose in the HCx Menu of my es73, but not with EMS-ESP. 

### REQUESTED INFORMATION

Please add these two heating types, if an remote device is attached. Probably usable also for "remote Temp." emulation.

Thanks a lot, 
If you need logs, please advice which log level you prever and which device i should watch.

thanks a lot.

Chris

proddy, 2024-12-01:
Think we need a `read` from your Thermostat for telegram `165` (you can do this in the System Log webUI now). From the code the 3rd byte is 0-off, 1-curve, 2-radiator, 3-convector, 4-floor. We'll need the values for Room Flow and Load.

SpaceTeddy, 2024-12-01:
@proddy 
I took some a log from WebUI with LogLevel ALL and switched the Heating Type from Room Flow -> Room Load -> Room Flow.
not sure which is the exact telegram, becasue didn't find 165 -> 0xA7

'
2024-12-01 12:52:12.111   TRACE 210:    [telegram]   Rx: 08 00 34 00 0A 02 42 02 42 20 00 00 03 00 01 6F 98 00 0E 1C 00 BA
2024-12-01 12:52:12.369   TRACE 211:    [telegram]   Rx: 08 00 18 00 32 01 67 07 00 01 01 20 60 80 00 02 42 01 4D 00 00 10 30 45 01 09 00 00 00 9A
2024-12-01 12:52:13.116   TRACE 212:    [telegram]   Rx: 08 00 18 00 32 01 67 07 00 01 01 20 60 80 00 02 42 01 4D 00 00 10 30 45 01 09 00 00 00 9A
2024-12-01 12:52:13.988   TRACE 213:    [telegram]   Rx: 10 00 06 00 18 0C 0C 01 33 39 06 00 E7
2024-12-01 12:52:14.277   TRACE 214:    [telegram]   Rx: 18 00 AF 00 00 C7 00 04 06 09
2024-12-01 12:52:20.728   TRACE 215:    [telegram]   Rx: 10 00 A3 00 FF 00 01 78
2024-12-01 12:52:21.113   TRACE 216:    [telegram]   Rx: 0B 90 FF 00 19 00 67 3D
2024-12-01 12:52:21.217   TRACE 217:    [telegram]   Rx: 10 0B FF 00 4B
2024-12-01 12:52:21.217   TRACE 218:    [telegram]   Rx: 0B 90 FF FE 01 00 67 E2
2024-12-01 12:52:21.342   TRACE 219:    [telegram]   Rx: 10 0B FF FE B5
2024-12-01 12:52:22.063   TRACE 220:    [telegram]   Rx: 08 00 34 00 0A 02 42 02 42 20 00 00 03 00 01 6F 98 00 0E 1C 00 BA
2024-12-01 12:52:22.294   TRACE 221:    [telegram]   Rx: 08 00 18 00 32 01 5F 07 00 01 01 20 60 80 00 02 42 01 4D 00 00 10 30 45 01 09 00 00 00 FB
2024-12-01 12:52:24.044   TRACE 222:    [telegram]   Rx: 0B 88 14 00 1B DF
2024-12-01 12:52:24.061   TRACE 223:    [telegram]   Rx: 08 0B 14 00 29 16 0E 94
2024-12-01 12:52:24.343   TRACE 224:    [telegram]   Rx: 0B 88 16 00 1B D7
2024-12-01 12:52:24.369   TRACE 225:    [telegram]   Rx: 08 0B 16 00 FF 5A 5A 00 06 FA 0A 01 05 55 37 02 28
2024-12-01 12:52:24.591   TRACE 226:    [telegram]   Rx: 0B 88 33 00 1B 43
2024-12-01 12:52:24.616   TRACE 227:    [telegram]   Rx: 08 0B 33 00 08 FF 3C FB 00 0F 00 02 41 00 FF 6E
2024-12-01 12:52:24.940   TRACE 228:    [telegram]   Rx: 0B 90 3E 00 1B B7
2024-12-01 12:52:25.001   TRACE 229:    [telegram]   Rx: 10 0B 3E 00 04 02 28 00 C7 5C 3C 32 32 32 00 00 07 11 32 F7
2024-12-01 12:52:25.312   TRACE 230:    [telegram]   Rx: 0B 90 3D 00 1B BB
2024-12-01 12:52:25.412   TRACE 231:    [telegram]   Rx: 10 0B 3D 00 05 25 28 22 00 28 00 02 00 05 05 2D 01 01 04 32 07 4B 01 FF 3C FF 11 00 05 02 01 70
2024-12-01 12:52:25.412   TRACE 232:    [telegram]   Rx: 0B 90 3D 1B 1B 8D
2024-12-01 12:52:25.487   TRACE 233:    [telegram]   Rx: 10 0B 3D 1B 00 01 0F 32 0A A2
2024-12-01 12:52:25.785   TRACE 234:    [telegram]   Rx: 0B 90 3F 00 1B B3
2024-12-01 12:52:25.912   TRACE 235:    [telegram]   Rx: 10 0B 3F 00 01 1F 00 39 01 57 00 87 21 1F 20 39 21 57 20 87 41 1F 40 39 41 57 40 87 61 1F 60 C5
2024-12-01 12:52:25.912   TRACE 236:    [telegram]   Rx: 0B 90 3F 1B 1B 85
2024-12-01 12:52:26.041   TRACE 237:    [telegram]   Rx: 10 0B 3F 1B 39 61 57 60 87 81 1F 80 39 81 57 80 8A A1 24 A0 8D C1 27 C0 87 E7 90 E7 90 E7 90 73
2024-12-01 12:52:26.041   TRACE 238:    [telegram]   Rx: 0B 90 3F 36 1B DF
2024-12-01 12:52:26.166   TRACE 239:    [telegram]   Rx: 10 0B 3F 36 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 DD
2024-12-01 12:52:26.166   TRACE 240:    [telegram]   Rx: 0B 90 3F 51 1B 11
2024-12-01 12:52:26.265   TRACE 241:    [telegram]   Rx: 10 0B 3F 51 90 E7 90 00 00 00 06 01 00 01 01 00 01 01 00 01 01 00 FF
2024-12-01 12:52:26.532   TRACE 242:    [telegram]   Rx: 0B 90 A5 00 1B E9
2024-12-01 12:52:26.626   TRACE 243:    [telegram]   Rx: 10 0B A5 00 02 00 00 00 FF F6 01 06 00 01 0D 02 00 54
2024-12-01 12:52:26.938 WARNING 244:    [telegram]   Incomplete Rx: 10 00 3E 00 04 02 28 00 C7 5C 3C 32 32 32 89
2024-12-01 12:52:27.228   TRACE 245:    [telegram]   Rx: 0B 90 37 00 1B 93
2024-12-01 12:52:27.367   TRACE 246:    [telegram]   Rx: 10 0B 37 00 FF 00 02 02 00 00 04 00 3C C3
2024-12-01 12:52:27.498   TRACE 247:    [telegram]   Rx: 10 00 3E 00 04 02 28 00 C7 5C 3C 32 32 32 00 00 07 11 32 C3
2024-12-01 12:52:28.065   TRACE 248:    [telegram]   Rx: 08 00 18 00 32 01 5D 07 00 01 01 20 60 80 00 02 42 01 4D 00 00 10 30 45 01 09 00 00 00 5C
2024-12-01 12:52:30.224   TRACE 249:    [telegram]   Rx: 10 08 35 00 01 11 10
2024-12-01 12:52:31.986   TRACE 250:    [telegram]   Rx: 08 00 34 00 0A 02 42 02 42 20 00 00 03 00 01 6F 98 00 0E 1C 00 BA
2024-12-01 12:52:32.223   TRACE 251:    [telegram]   Rx: 10 00 3D 00 04 E9
2024-12-01 12:52:32.444   TRACE 252:    [telegram]   Rx: 08 00 18 00 32 01 5C 07 00 01 01 20 60 80 00 02 42 01 4D 00 00 10 30 45 01 09 00 00 00 83
2024-12-01 12:52:33.474   TRACE 253:    [telegram]   Rx: 10 08 1A 00 26 64 64 00 F9
2024-12-01 12:52:35.015   TRACE 254:    [telegram]   Rx: 08 00 18 00 26 01 5B 5A 00 01 01 2C 60 80 00 02 42 01 4C 00 00 10 30 43 01 1B 00 02 00 EC
2024-12-01 12:52:38.001   TRACE 255:    [telegram]   Rx: 08 00 18 00 26 01 5B 5A 00 01 01 2D 60 80 00 02 42 01 4D 00 00 11 30 4C 01 1C 00 02 00 79
2024-12-01 12:52:38.995   TRACE 256:    [telegram]   Rx: 08 00 18 00 26 01 5B 5A 00 09 01 2D 60 80 00 02 42 01 4D 00 7D 11 2D 48 00 C8 00 02 00 EE
2024-12-01 12:52:39.474   TRACE 257:    [telegram]   Rx: 10 08 1A 00 32 07 64 00 CC
2024-12-01 12:52:41.011   TRACE 258:    [telegram]   Rx: 08 00 18 00 32 01 5B 07 30 09 01 25 60 80 00 02 42 01 4E 00 A6 11 2D 48 00 C8 00 02 00 69
[... cut ...]
2024-12-01 12:52:53.558   TRACE 272:    [telegram]   Rx: 18 08 35 00 00 00 1A
2024-12-01 12:52:53.974   TRACE 273:    [telegram]   Rx: 10 00 3D 00 04 E9
2024-12-01 12:52:54.279   TRACE 274:    [telegram]   Rx: 08 00 18 00 26 01 61 5A 00 01 01 2D 60 80 00 02 42 01 4E 00 00 11 30 4C 01 1C 00 02 00 13
2024-12-01 12:52:55.000   TRACE 275:    [telegram]   Rx: 08 00 18 00 26 01 5F 5A 00 09 01 2D 60 80 00 02 42 01 4E 00 80 11 30 4C 01 1C 00 02 00 71
2024-12-01 12:52:55.753 WARNING 276:    [telegram]   Incomplete Rx: 9D 00 89 00 09

MichaelDvP, 2024-12-01:
It's telegram 0x3D offset 0 with new values 4 and 5.
We need a function to dynamic adapt the enum list if control is set to RC20. 
I'll check if my RC35 have the same new options.

SpaceTeddy, 2024-12-01:
No worries Michael,
Usually you have to set it once at the Controller and basically no need to add this feature if it will take risk of implementation. I just mention this, because to have this heating type complete, if you like :)
Sers, 
Chris

MichaelDvP, 2024-12-01:
Please try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Changing the options dynamically is also for RC300 usefull, first appoach now. For my RC35 (compatible to ES73) it looks ok.
If you're using HA, check if it the config recreated correctly.

SpaceTeddy, 2024-12-01:
@MichaelDvP ,
I updated to 3.7.2-dev.2 and via WebUI I can recognize the new, correct setting:

[image]

It is also possible to change the heating types (tested with roomload). 
If roomload is choosen, the target flow temperature changes to max. flow temperature. This works correct. 

[image]

In HA and telnet console, the heating type is also updated:

[image]

[image]

If I change the control type from RC20 to OFF, the list of heating type is changed that roomflow and roomload is not available:
[image]

so, from my point of view it looks very optimistic ;)

thx and sers,
Chris

closed 2024-12-03 by SpaceTeddy