emsesp/EMS-ESP32 #2271: Night setback not working properly

mattreim, 2024-12-02:
### DESCRIPTION

Hello everyone,
The night setback has not been working properly for some time now. It switches to eco and the heater shows off but it continues to heat. The seltemp remains at the comfort or eco temperature and the value cannot be manually adjusted to zero.  The only remedy is to restart of EMS-ESP.  When the remote thermostat is created, the value of seltemp jumps to zero and the heating switches off.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
  "system": {
    "version": "3.7.2-dev.1",
    "uptime": "000+22:42:10.336",
    "uptimeSec": 81730,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 164,
    "maxAlloc": 103,
    "freeCaps": 114,
    "usedApp": 1740,
    "freeApp": 244,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "EMS-ESP",
    "RSSI": -82,
    "WIFIReconnects": 0,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": false,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "192.168.178.1",
    "tzLabel": "Europe/Berlin"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 59056,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 1,
    "enabled": true,
    "clientID": "esp32-422a13d8",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 0,
    "publishTimeThermostat": 0,
    "publishTimeSolar": 0,
    "publishTimeMixer": 0,
    "publishTimeWater": 0,
    "publishTimeOther": 0,
    "publishTimeSensor": 0,
    "publishSingle": false,
    "publish2command": true,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {},
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 180536,
    "busReads": 33424,
    "busWrites": 0,
    "busIncompleteTelegrams": 2,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 120,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 4,
    "txGPIO": 5,
    "dallasGPIO": 0,
    "pbuttonGPIO": 0,
    "ledGPIO": 0,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 13,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Logamax plus GB172i.2-15",
      "deviceID": "0x08",
      "productID": 195,
      "brand": "Buderus",
      "version": "05.06",
      "entities": 45,
      "handlersReceived": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlersFetched": "0xE6 0xEA 0x28",
      "handlersPending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlersIgnored": "0xE7 0x02E0 0x02EA 0xD7 0xD8 0x02D6 0x029B 0x04B4"
    },
    {
      "type": "thermostat",
      "name": "Logamatic BC400",
      "deviceID": "0x10",
      "productID": 4,
      "brand": "",
      "version": "49.04",
      "entities": 25,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E 0x0269",
      "handlersIgnored": "0xC4 0x047B 0xBF 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC100H, CR10H",
      "deviceID": "0x38",
      "productID": 200,
      "brand": "",
      "version": "40.04",
      "entities": 3,
      "handlersReceived": "0x042B 0x047B",
      "handlersPending": "0x0273 0x0A6A",
      "handlersIgnored": "0xF7"
    },
    {
      "type": "scheduler",
      "name": "scheduler",
      "entities": 2
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

mattreim, 2024-12-03:
Now I have a log of the event with a minute before and after.
[image]

```lua
001+05:41:58.806 T 2: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 31 01 EC 64 14 03 02 01 E1 00 00 01 BB 00 9F 11 00 01 ED 00 00
001+05:41:58.982 T 3: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorFastPlus(0xE4), data: 00 02 3C 00 9A 14 64 (offset 35)
001+05:41:59.171 T 4: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 ED 14 64 41 00 31 00 00
001+05:41:59.375 T 5: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorWWPlus(0xE9), data: 05 01 41 01 41 00 00 00 00 46 3C 00 00 02 00 07 ED 00 00 40 00 00 00 05 02 00
001+05:42:06.272 T 6: [emsesp] thermostat(0x10) -> all(0x00), B, RCTime(0x06), data: 18 0C 16 03 3B 17 01 00 10 FF 00
001+05:42:07.212 T 7: [emsesp] me(0x0B) -> boiler(0x08), R, UBAParametersPlus(0xE6), length: 27
001+05:42:07.250 T 8: [emsesp] boiler(0x08) -> me(0x0B), W, UBAParametersPlus(0xE6), data: 01 41 00 50 64 0E 00 00 06 FA 0A 01 02 64 0A 02 00 00 00 1E 01 00 01 00 01 00 00
001+05:42:07.250 T 9: [emsesp] me(0x0B) -> boiler(0x08), R, UBAParametersPlus(0xE6), length: 27 (offset 27)
001+05:42:07.285 T 10: [emsesp] boiler(0x08) -> me(0x0B), W, UBAParametersPlus(0xE6), data: 00 08 0D 78 00 00 00 00 00 3C (offset 30)
001+05:42:07.285 T 11: [emsesp] me(0x0B) -> boiler(0x08), R, UBAParametersPlus(0xE6), length: 27 (offset 40)
001+05:42:07.304 T 12: [emsesp] boiler(0x08) -> me(0x0B), W, UBAParametersPlus(0xE6), data: <empty> (offset 40)
001+05:42:07.487 T 13: [emsesp] me(0x0B) -> boiler(0x08), R, UBAParameterWWPlus(0xEA), length: 27
001+05:42:07.539 T 14: [emsesp] boiler(0x08) -> me(0x0B), W, UBAParameterWWPlus(0xEA), data: 00 00 00 00 00 01 3C FB 00 0C 00 02 46 00 00 00 37 3C 28 23 41 64 00 00 00 00 01
001+05:42:07.539 T 15: [emsesp] me(0x0B) -> boiler(0x08), R, UBAParameterWWPlus(0xEA), length: 27 (offset 27)
001+05:42:07.544 T 16: [emsesp] boiler(0x08) -> me(0x0B), W, UBAParameterWWPlus(0xEA), data: <empty> (offset 27)
001+05:42:07.760 T 17: [emsesp] me(0x0B) -> boiler(0x08), R, WeatherComp(0x28), length: 27
001+05:42:07.773 T 18: [emsesp] boiler(0x08) -> me(0x0B), W, WeatherComp(0x28), data: 00 5A 14 12 FF 05
001+05:42:07.850 T 19: [emsesp] thermostat(0x38) -> thermostat(0x10), W, RemoteTemp(0x042B), data: 00 D6
001+05:42:08.372 T 20: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Monitor(0x02A5), length: 25
001+05:42:08.450 T 21: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Monitor(0x02A5), data: 00 D6 03 2C 31 00 2C 22 00 01 03 03 01 00 01 03 DD 00 00 11 01 00 FF FF 00
001+05:42:08.450 T 22: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Monitor(0x02A5), length: 25 (offset 25)
001+05:42:08.542 T 23: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Monitor(0x02A5), data: 00 00 00 12 00 00 0B 64 41 00 3C 01 33 01 02 00 00 00 01 5C 0B (offset 25)
001+05:42:08.681 T 24: [emsesp] thermostat(0x38) -> thermostat(0x10), W, RemoteHumidity(0x047B), data: 0B 33 00 6C
001+05:42:09.143 T 25: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 31 01 EC 64 14 03 02 01 E1 00 00 01 BA 00 9A 11 00 01 ED 00 00
001+05:42:09.319 T 26: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorFastPlus(0xE4), data: 00 02 3C 00 9A 14 64 (offset 35)
001+05:42:09.510 T 27: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 EC 14 64 41 00 31 00 00
001+05:42:09.714 T 28: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorWWPlus(0xE9), data: 05 01 41 01 41 00 00 00 00 46 3C 00 00 02 00 07 ED 00 00 40 00 00 00 05 02 00
001+05:42:09.930 T 29: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Set(0x02B9), length: 25
001+05:42:10.030 T 30: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Set(0x02B9), data: FF 00 2C 00 22 01 00 00 FF 00 00 01 EC 20 01 0F 05 2E 00 01 22 02 24 00 01
001+05:42:10.030 T 31: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Set(0x02B9), length: 25 (offset 25)
001+05:42:10.068 T 32: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Set(0x02B9), data: 30 24 00 00 2E 00 00 (offset 25)
001+05:42:10.285 T 33: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Summer(0x02AF), length: 25
001+05:42:10.339 T 34: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Summer(0x02AF), data: 03 00 00 00 3C 23 11 01 19 14 00 00 00 00
001+05:42:10.498 T 35: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Curves(0x029B), length: 25
001+05:42:10.542 T 36: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Curves(0x029B), data: 01 01 00 00 FF 02 05 28 41 00
001+05:42:10.725 T 37: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Set2(0x02CC), length: 25
001+05:42:10.749 T 38: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Set2(0x02CC), data: 01 (offset 3)
001+05:42:10.749 T 39: [emsesp] me(0x0B) -> thermostat(0x10), R, RC300Set2(0x02CC), length: 25 (offset 4)
001+05:42:10.792 T 40: [emsesp] thermostat(0x10) -> me(0x0B), W, RC300Set2(0x02CC), data: <empty> (offset 4)
001+05:42:11.008 T 41: [emsesp] me(0x0B) -> thermostat(0x10), R, HPMode(0x0291), length: 25
001+05:42:11.044 T 42: [emsesp] thermostat(0x10) -> me(0x0B), W, HPMode(0x0291), data: 01 00 00 03 00 00
001+05:42:11.203 T 43: [emsesp] me(0x0B) -> thermostat(0x10), R, HPMode(0x0292), length: 25
001+05:42:11.239 T 44: [emsesp] thermostat(0x10) -> me(0x0B), W, HPMode(0x0292), data: 00 00 00 00 00 00
001+05:42:11.421 T 45: [emsesp] me(0x0B) -> thermostat(0x10), R, HPMode(0x0293), length: 25
[... cut ...]
001+05:43:29.461 T 189: [emsesp] boiler(0x08) -> all(0x00), B, UBAMonitorWWPlus(0xE9), data: 05 01 41 01 41 00 00 00 00 46 3C 00 00 02 00 07 ED 00 00 40 00 00 00 05 02 00
001+05:43:29.663 T 190: [emsesp] me(0x0B) -> thermostat(0x38), R, DeviceName(0x01), length: 27 (offset 27)
001+05:43:29.697 T 191: [emsesp] thermostat(0x38) -> me(0x0B), W, DeviceName(0x01), data: <empty> (offset 27)
001+05:43:30.830 T 192: [emsesp] thermostat(0x10) -> all(0x00), B, SystemError(0xBF), data: 10 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

mattreim, 2024-12-03:
After restarting.
[image]

MichaelDvP, 2024-12-05:
Looks all ok to me. The log starts with
0x2A5 offset 6/7: 2C 22 actual setpoint 22°, next 17°
then switches to 
0x2A5 offset 6/7: 00 2C actual setpoint 17°, next 22°
But you have set
0x029B offset 0: 01 optimize switchtimes
so the optimized setpoint stay on 0x2A5, offset 3: 2C

I don't know what the optimization do in this thermostat, but seems that it want to stay at daytemp for a while.
RTFM

mattreim, 2024-12-05:
A function that could be suitable is called “Einschaltoptimierung” and I have now switched it off. I'm also trying to stop the remote thermostat's communication during this time (this night it worked).

mattreim, 2024-12-08:
You were right, it was because of this attitude. The heating now switches off reliably. Thank you very much for the help

[image]

closed 2024-12-08 by mattreim