emsesp/EMS-ESP32 #2273: Improve error handling for Tx write and TX Read errors

tp1de, 2024-12-02:
I do have some errors on my EMS+ system. 

For write commands there are sometimes TX write timeouts and errors:  
`Last Tx write timeout / Last Tx write rejected by host`

While sending raw read request I get failures:
`
2024-12-02 11:11:43.393 I 360: [command] Called command system/send (Sende EMS-Telegramm) with value 0B 90 FF 00 0C 02 09
2024-12-02 11:11:44.771 E 361: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B 10 FF 00 02 09 0C
`

This happens on all type of thermostat commands - setting tempautotemp or switchprog and others.
While setting single values it happens on 5-10% of all command. On longer commands (e.g. switchprog) it happens much more often.
Using the testbuild from @MichaelDvP commands are > 24 bytes and errors are more often (>40% of all cases).
When I send short raw commands with 4 bytes (2 switchTimes) the error rate is < 10% but still existing.

Since these errors do not exist while using the Buderus gateway, this might look like a timing issue on ems-esp.

Can't these errors be corrected within ems-esp while resending the commands?
Or is there any other way to recognize these errors rather then looking at status log? (e.g. within response topic)

MichaelDvP, 2024-12-03:
> Can't these errors be corrected within ems-esp while resending the commands?

If we know what causes these errors we can send the first time in the right way. But we have 4 different TX-modes and don't know why one or another works better in some systems. (in my system all 4 modes work without any issue, i always use hardware).

tp1de, 2024-12-03:
There is no 1:1 cause since most of the same commands work without any issue.
Actually I am resending important commands up to 3 times to make sure that at least one is working.
Couldn't the command with TX write errors being mentioned within the response topic?

Or do you have any idea what I can test next to the different tx-modes? (all of them produces these errors)

tp1de, 2024-12-04:
My km200 gateway is requesting 38 to 40 read requests from boiler or thermostat within every minute. These requests are replied from boiler / thermostat to gateway. 

There are 2 breaks in every minute where no telegram is send/received of approx. 11-12 seconds.
ems-esp requests a similar amount of data but within 15 seconds on every minute and then having an break of 35 secs.

When I try to do changes (tx write requests) during one of this breaks, it looks like that I do not get errors.( Otherwise I do get some)

Could it be that the coexistence of ems-esp and buderus gateway reduces the chance to execute tx writes correctly?
I recognized a similar behavior when I had 2 active ems-esp gateways for testing purposes active.

closed 2025-03-22 by proddy