emsesp/EMS-ESP32 #2274: API http://[IP]/api/thermostat/info gives error

RikBast, 2024-12-02:
I tried to use http://[IP]/api/thermostat/info in v3.7.1 and it gives an error:

{"message":"unknown device thermostat"}

In v3.6.5 it gives the correct response.

proddy, 2024-12-02:
support info?

RikBast, 2024-12-04:
> support info?

Sorry for the late response in this.

What happens is the following under v3.7.1:

When I typ: http://192.168.0.50/api/thermostat into any browser window I now get this response:

{"message":"unknown device thermostat"}

This used to be in v3.6.5

{"hc1 Streeftemperatuur kamer (seltemp)":15.0,"hc1 Huidige kamertemperatuur (currtemp)":16.4,"hc1 Discovery huidige kamertemperatuur (haclimate)":"kamertemperatuur"}

It's also reflected in my LUA code with a Fibaro Home Center 3 where I use the 'GET' function

function QuickApp:httpGetJsonThermostat(url, body, func)
  self.http:request(url, { 
  
      options = { 
        method = 'GET',
        
        headers = {
          
          --['x-access-token'] = self.token,
          ['Content-Type']='application/json'
        },
        
      },
        success = function(response) 
                    self:debug("EMS API status :", response.status)
                    self:debug("EMS API output :", response.data)
                    --datareadable = json.encode(response.data)
                    --self:debug("Huawei Token API output :",datareadable)
                    -- continue if HTTP status code is 200
                    if (tonumber(response.status) == 200) then
                        local jsontable = json.decode(response.data)
                        local Trimmed_output = tostring(response.data)
                        Trimmed_output = string.gsub(Trimmed_output, "[^%S\n]+", "")
                        Trimmed_output = string.gsub(Trimmed_output, "[()]", "")
                        --self:debug("Trimmed output: ", Trimmed_output)
                        Trimmed_output_decoded = json.decode(Trimmed_output)
                        data.Huidigekamertemperatuur = Trimmed_output_decoded.hc1Huidigekamertemperatuurcurrtemp
                        data.Streeftemperatuurkamer = Trimmed_output_decoded.hc1Streeftemperatuurkamerseltemp
                        self:debug("hc1 Huidige kamertemperatuur EMS-data = " .. tostring(data.Huidigekamertemperatuur) .. " °C")   
                        self:debug("hc1 Streeftemperatuur kamer EMS-data = " .. tostring(data.Streeftemperatuurkamer) .. " °C")
                        self:updateChildDevices() -- Update the Child Devices
                        self:setPropertiesLabels() -- Update properties and labels
                        --self:updateProperty("value", { value= data.Huidigekamertemperatuur, unit= "C" })                  
                    end
                end,
        error = function(message)
                    self:debug("EMS API error:", message)
                end
    })   
end

proddy, 2024-12-04:
can you send your system info please? https://docs.emsesp.org/Support/

RikBast, 2024-12-04:
> can you send your system info please? https://docs.emsesp.org/Support/

I have a Gateway E32 (Ethernet + WiFi Edition) from BBQKees, is that what you need? Bought in jan 2023.  I contacted BBQKees already and he suggested to open an issue here at GitHub

MichaelDvP, 2024-12-04:
For the system info navigate to http://192.168.0.50/help and download the support information, attach it here.
Or call http://192.168.0.50/api/system/info and copy the output.

After update and restart it takes around minute to find the devices on the bus, in this time you will get the `unknown device` message.

RikBast, 2024-12-04:
Here is system info:

```
{"system":{"version":"3.7.1","uptime":"000+00:00:46.842","uptimeSec":46,"platform":"ESP32","cpuType":"ESP32-D0WD","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":174,"maxAlloc":107,"freeCaps":124,"usedApp":1740,"freeApp":244,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-78,"WIFIReconnects":0,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":99,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"domoticz","discoveryType":1,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":2,"APIFails":1},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":102,"busReads":37,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32","locale":"nl","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":0,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"EASY CONNECT/Ecomline Excellent","deviceID":"0x08","productID":206,"brand":"Nefit","version":"02.01","entities":67,"handlersReceived":"0x10 0x11 0x1C 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x15 0x1A 0x35 0x2A 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B","handlersIgnored":"0x06"},{"type":"thermostat","name":"Logamatic TC100, Moduline Easy","deviceID":"0x18","productID":202,"brand":"","version":"02.22","entities":3,"handlersFetched":"0x0A","handlersPending":"0xA3 0x06 0xA2 0x12 0x13","handlersIgnored":"0x16 0x1C 0x33 0x19"},{"type":"controller","name":"EASY CONNECT/Ecomline","deviceID":"0x09","productID":206,"brand":"Nefit","version":"02.01","entities":0}]}

proddy, 2024-12-04:
that's very strange, I can't think of what could cause that. Do any of the other API calls work, like `api/boiler` or `api/system` ? And if you go to the Telnet console and do a `su` with ems-esp-neo and then `call thermostat info` does it work?

RikBast, 2024-12-05:
boiler info is returning values:

```
function QuickApp:startGetBoilerInfo()
  self:httpGetJsonBoiler("http://192.168.0.50/api/boiler/info","",processResult)
  --self:checkState()
  --self:startGetThermostatInfo()
end
```

```

[05.12.2024] [06:19:40] [DEBUG] [QUICKAPP548]: EMS API output : {"force heating off (heatingoff)":"uit","Verwarming actief (heatingactive)":"uit","Warm water actief (tapwateractive)":"uit","Ingestelde aanvoertemperatuur (selflowtemp)":0,"Modulatie verwarmingspomp (heatingpumpmod)":0,"Huidige aanvoertemperatuur (curflowtemp)":15.0,"Retourtemperatuur (rettemp)":15.0,"Mixer temperatuur (switchtemp)":0.0,"Gas (burngas)":"uit","gas fase 2 (burngas2)":"uit","Vlammenstroom (flamecurr)":0.0,"Ventilator (fanwork)":"uit","Ontsteking (ignwork)":"uit","Olie voorverwarming (oilpreheat)":"uit","Minimaal brandervermogen (burnminpower)":0,"Maximaal brandervermogen (burnmaxpower)":83,"Antipendeltijd (burnminperiod)":10,"ketel aan hysterese verschil (boilhyston)":-6,"ketel uit hysterese verschil (boilhystoff)":6,"Verwarmen geactiveerd (heatingactivated)":"aan","Verwarmingstemperatuur (heatingtemp)":56,"Verwarmingspomp (heatingpump)":"uit","Ketelpomp max vermogen (pumpmodmax)":100,"Ketelpomp min vermogen (pumpmodmin)":100,"Ketelpomp modus (pumpmode)":"proportioneel","Pomp nalooptijd (pumpdelay)":3,"Ingestelde aanvoertemperatuur (setflowtemp)":0,"Ingestelde maximale brandervermogen (selburnpow)":0,"Brandervermogen (curburnpow)":0,"Aantal brander starts (burnstarts)":0,"Totale branderlooptijd (burnworkmin)":"0 Dagen 0 Uren 0 Minuten","Totale looptijd brander fase 2 (burn2workmin)":"0 Dagen 0 Uren 0 Minuten","Totale looptijd verwarming (heatworkmin)":"0 Dagen 0 Uren 0 Minuten","Aantal brander starts verwarming (heatstarts)":0,"totale looptijd branderautomaat (UBA) (ubauptime)":"0 Dagen 0 Uren 0 Minuten","Statuscode (servicecode)":"0H","Status codenummer (servicecodenumber)":0,"Onderhoudsmelding (maintenancemessage)":"H00","dhw zet aan/uit (wwtapactivated)":"aan","dhw Streeftemperatuut (wwsettemp)":60,"dhw Geselecteerd temperatuur (wwseltemp)":60,"dhw type (wwtype)":"volumestroom","dhw Comfort (wwcomfort)":"heet","dhw Aanvoertemperatuur offset (wwflowtempoffset)":40,"dhw Circulatiepomp aanwezig (wwcircpump)":"aan","dhw Buffer laadtype (wwchargetype)":"3-weg klep","dhw Inschakeltemperatuurverschil (wwhyston)":-5,"dhw Uitschakeltemperatuurverschil (wwhystoff)":-1,"dhw Desinfectietemperatuur (wwdisinfectiontemp)":70,"dhw Modus circulatiepomp (wwcircmode)":"2x3min","dhw Circulatiepomp actief (wwcirc)":"uit","dhw Huidige interne temperatuur (wwcurtemp)":51.0,"dhw Hudige warmwater doorstroming (wwcurflow)":2.5,"dhw geactiveerd (wwactivated)":"aan","dhw Buffer eenmalig laden (wwonetime)":"uit","dhw Desinfectie (wwdisinfecting)":"uit","dhw Laden (wwcharging)":"uit","dhw herladen (wwrecharging)":"uit","dhw Temperatuur OK (wwtempok)":"aan","dhw Actief (wwactive)":"uit","dhw 3-wegklep actief (ww3wayvalve)":"uit","dhw Aantal starts (wwstarts)":0,"dhw Actieve tijd (wwworkm)":"0 Dagen 0 Uren 0 Minuten","nominal Power (nompower)":0,"total energy (nrgtotal)":0.00,"energy heating (nrgheat)":0.00,"dhw energy (nrgww)":0.00}
'''

With telnet I will try later today.

proddy, 2024-12-05:
@MichaelDvP any idea what could cause this?

MichaelDvP, 2024-12-05:
I've already mentioned:
>>After update and restart it takes around minute to find the devices on the bus, in this time you will get the unknown device message.

But only get back: `"uptimeSec":46,` with `"api":{"APICalls":2,"APIFails":1}` indicating the api fail was before. 
How often should i mention to wait longer?

proddy, 2024-12-05:
please test and re-open (with support info) if you still have issues.

RikBast, 2024-12-05:
I have been running v3.7.1 for more than 1,5 hours:

```
{"system":{"version":"3.7.1","uptime":"000+01:35:22.976","uptimeSec":5722,"platform":"ESP32","cpuType":"ESP32-D0WD","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":175,"maxAlloc":107,"freeCaps":125,"usedApp":1740,"freeApp":244,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-78,"WIFIReconnects":0,"TxPowerSetting":78,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":1330,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"domoticz","discoveryType":1,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":172,"APIFails":86},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":3716,"busReads":553,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32","locale":"nl","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":0,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"EASY CONNECT/Ecomline Excellent","deviceID":"0x08","productID":206,"brand":"Nefit","version":"02.01","entities":67,"handlersReceived":"0x10 0x11 0x1C 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x15 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x28 0x2E 0x3B","handlersIgnored":"0x06"},{"type":"thermostat","name":"Logamatic TC100, Moduline Easy","deviceID":"0x18","productID":202,"brand":"","version":"02.22","entities":3,"handlersFetched":"0x0A","handlersPending":"0xA3 0x06 0xA2 0x12 0x13","handlersIgnored":"0x18 0x16 0x1C 0x33 0x19"},{"type":"controller","name":"EASY CONNECT/Ecomline","deviceID":"0x09","productID":206,"brand":"Nefit","version":"02.01","entities":0}]}

<img width="1464" alt="Scherm­afbeelding 2024-12-05 om 20 35 40" src="https://github.com/user-attachments/assets/95759983-464c-4196-9508-bd31ac2d8c83">

http://192.168.0.50/api/thermostat still gives this message:

{"message":"unknown device thermostat"}

proddy, 2024-12-05:
Oh!

MichaelDvP, 2024-12-05:
Oh, this is an easy thermostat, not writeable, so if fails here:
https://github.com/emsesp/EMS-ESP32/blob/473cf7c8af7a3502bb187ad60d7d37e3177fe9b8/src/command.cpp#L68
This is a bug.

RikBast, 2024-12-06:
Glad you found the issue, thank you for helping

proddy, 2024-12-06:
good find Michael

MichaelDvP, 2024-12-06:
@RikBast Please check with the new dev version if it is fixed.

RikBast, 2024-12-06:
@MichaelDvP Works! Thank you

closed 2024-12-06 by MichaelDvP