emsesp/EMS-ESP32 #2276: modbus write permission

GLT-IN, 2024-12-03:
### DESCRIPTION

Hi,
writing of a single entity per request doesn´t work for Buderus Topline GB162, RC35 and MM10
request
FF 00 00 00 00 06 06 06 03 E8 00 0A
answer
FF 00 00 00 00 03 06 86 03

Reading works.
Do I need to catch up a write permission first?
How does it work?

Best regards,
Markus

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2024-12-04:
@mheyse , any idea?

mheyse, 2024-12-04:
There should be an error in the log, could you please post it?

GLT-IN, 2024-12-04:
[image]

MichaelDvP, 2024-12-04:
Oh, seems modbus is hardcoded to use mqtt-base `ems-esp`, we should better use `path = std::string("api/") + ` here:
https://github.com/emsesp/EMS-ESP32/blob/473cf7c8af7a3502bb187ad60d7d37e3177fe9b8/src/modbus.cpp#L417-L419

@GLT-IN As workaround set back in mqtt settings the base to `ems-esp`

MichaelDvP, 2024-12-05:
@GLT-IN Please try with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Check and adapt in application setting the modbus timeout, this is changed to seconds (instead of milliseconds).

closed 2024-12-05 by GLT-IN