emsesp/EMS-ESP32 #2277: Temperature and set Temperature reading of bosh easycontrol CT200

Disantyk, 2024-12-04:
Good morning,

I purchased the Bosh easycontrol thermostat as a preperation for floor heating (this thermostat supports up to 25 temp sensors including a floor temp sensor)

Then i discoverd that the temperature, set temperature and humidity was missing.

So I did some debugging and after an few hours I could send and receive the temperature and set temperature (read only sadly enough)

I showed it to Kees and he asked me to post here so it can be build in

You have to devide the temperature by 10 and set temp by 2 (to have correct values in HA

Hope this is usefull and I wil try to get more info out of it.

Here is a screenshot of how I did it

[image]

MichaelDvP, 2024-12-04:
Is this the known CT200,product-id 203?
Please add the support information (emsesp ->help page) and attach it here, so we can see the device info.
Telegram 0x2A5 is normally not writeable, does this really work on this thermostat?
Can you make a log (log all) of > 1 minute and attach it here.

Disantyk, 2024-12-04:
That's the one, I have no knowledge about ems but I work alot with plc software and I learned myself about python.

When I'm back home I'll make a log file and try to get the humidity also

I have both values in my home assistant  and they get updated

If you can tell me what the usual write command are, then I'll play around a bit with that to and post here if I found something

Disantyk, 2024-12-04:
device info: 

Device Gegevens
Type
Thermostat

Naam
EasyControl, CT200

Merk
Bosch

Apparaat ID
0x18

Product ID
203

Versie
05.02
[log (2).txt](https://github.com/user-attachments/files/18015349/log.2.txt)

MichaelDvP, 2024-12-05:
As for other Easy thermostatsthis is not writeable, sorry. Also the bus communication is very limited, the 0x0A telegram is answered, but only zeros, the 0x02A5 is only broadcasted for roomTemp and selTemp, no other values. Reads to this ems+ telegram are answered with `<empty>` with ems telegram style. Don't know what bosch engineers have smoked when implementing this.
Try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
This should read the right values, but no way to change settings.

Disantyk, 2024-12-05:
It is also broadcasted by the thermostat

HEX value B0 = 176 (17.6)

See screenshot

[image]

MichaelDvP, 2024-12-06:
Yes, this should be shown now in devices/thermostat as roomtemperature (0x02A5, offset0, int16) and seltemp (0x02A5, offset6, int8)

Isn't it working with latest dev version? Please make a log (log all) of > 1 min, download and attach here.

Disantyk, 2024-12-06:
[image]

Vochomurka, 2024-12-10:
I have a Bosch Condens 2300i boiler and a CT200 Thermostat
I tried the above and it doesn't work for me.
I'm not getting any usable responses from the thermostat.
Just empty data.
[image]
[image]
[image]
[image]
[image]
[image]

proddy, 2024-12-10:
@Vochomurka `UINT8`?

Vochomurka, 2024-12-10:
that is the same
[image]
[image]

proddy, 2024-12-10:
and a read on 2A5, what does that show (set developer mode and go to System Log in webUI)

proddy, 2024-12-10:
strange that it doesn't work as you have the same EMS device as @Disantyk

Vochomurka, 2024-12-10:
The only thing that comes to mind is that I have a K20RF wireless module in the boiler and the CT200 communicates with the boiler through it. I also see only empty responses there. Whereas where it works there are broadcasts to everyone from the thermostat with that temperature.

Disantyk, 2024-12-13:
I was looking to get a bit more control and there seem to be an API for it so that you can control the easy control

https://github.com/robertklep/bosch-xmpp

https://github.com/bosch-thermostat/bosch-thermostat-client-python/tree/dev

Maybe this can be implemented in the ems-esp code?

When I have more time again I'll have a better look

Also a integrations for Home assistant

https://github.com/bosch-thermostat/home-assistant-bosch-custom-component

Also found in hacs adon

Hope this is helpfull information

Vochomurka, 2024-12-13:
I use this integration.
It works, but only via the Bosch cloud.
It does not communicate with the boiler, it only reads data from the Cloud that the CT200 sends there.
Unfortunately, it does not work reliably for me.
It randomly stops reading items.
Sometimes only some and sometimes all.
Unfortunately, it behaves in such a way that the item stops changing and so the problem is not visible at first glance.
There are probably the biggest problems with energy items.
In addition, the energy in the Cloud apparently changes only once a day at midnight.
After restarting the integration, it works for a while.
It seems that when the integration is set Use new statistic API
to ON, it is more reliable, but again, for example, energy values ​​are always read as -17kWh.
That is why I bought EMS ESP.
Now I have reliable reading of all data from the boiler, including gas (energy) consumption, continuously at all times.
I use the integration mainly for the remaining data that I cannot obtain otherwise, such as smart thermostatic heads that only communicate with the CT200.

proddy, 2024-12-14:
That's correct. I looked into those HA solutions a few years ago (there's an EMS-ESP GitHub issue somewhere on this) and @Vochomurka is correct. It uses the XMPP protocol to communicate to the Bosch cloud with private keys so we can't add this to EMS-ESP. Best to keep using EasyControl and paying Bosch's subscription fees :-)

MichaelDvP, 2024-12-14:
@Vochomurka Please make a full log of 2-3 minutes and attach it as text file. Maybe there is something from thermostat or wireless we can use.

Vochomurka, 2024-12-14:
> That's correct. I looked into those HA solutions a few years ago (there's an EMS-ESP GitHub issue somewhere on this) and @Vochomurka is correct. It uses the XMPP protocol to communicate to the Bosch [...]

As far as I know, there is no subscription fee.

Vochomurka, 2024-12-14:
Here is the log
[log.txt](https://github.com/user-attachments/files/18135950/log.txt)
[emsesp_system_info.json](https://github.com/user-attachments/files/18135967/emsesp_system_info.json)

MichaelDvP, 2024-12-14:
The thermostat send very few messages. Only writes to the boier
```
thermostat(0x18) W boiler(0x08), ?(0xE7), data: 00 00 00 64
thermostat(0x18) W boiler(0x08), ?(0x02E0), data: 01 28 64 00 01
thermostat(0x18) W boiler(0x08), ?(0x02EA), data: 00
```
looks like some commands, 0x64 could be burner or pump with 100%, the 01 28 from 0x02E0 could be a temperature,
You can try to create custom entities and for 0xE7 offset 3 UINT and 0x02E0, offset 0, UINT16 and 0x02E0 offset 2 UINT
and log the changes. But i can not see any roomtemp or setpoint.

The wireless base only sends the clock in telegram 0x06. We can add this to connect device.
You can try if you get a reply from connect for the telegram 0x2A5, with `read 50 2A5`

Disantyk, 2024-12-14:
When I had the custom entity, it also showed up empty (screenshot) 

[image]

But if I look at the devices, custom the value was there.

Best is to try te latest beta

Vochomurka, 2024-12-14:
CT200 responds to 0x2E0
so far I have only reliably identified offset 1 - the desired water temperature from the boiler by the thermostat
It also responds to 0x0E7 and 0x0EF, but I have no idea what values ​​it returns
[image]
[image]

Disantyk, 2024-12-14:
> That's correct. I looked into those HA solutions a few years ago (there's an EMS-ESP GitHub issue somewhere on this) and @Vochomurka is correct. It uses the XMPP protocol to communicate to the Bosch [...]

isn't it possible to add fields where you have to add the Device serial number, Device access key and Password to setup the connection?

@Vochomurka 

for the room temp, you nee offset 1 on 0X2A5
seltemp devide by 2?

Vochomurka, 2024-12-15:
I have the latest version 5.02
My CT200 does not respond to 0x2A5.
It does not return anything. I don't have anything to move.
I use the integration from your link.
But it is not reliable and some things do not work or load incorrectly.
I only use it for what cannot be obtained through EMS ESP
[image]

Disantyk, 2024-12-15:
My log was also <empty> with the 0x2a5,

But if you look at the devices tab.

You see at least 3:
Your thermostat
Your boiler
Your custom items

If you look in costum items you see your roomtemp and seltemp and any other items made. Wait for about a minute it should show the values

bittylicious, 2024-12-15:
Hey. I'm also trying to get things working (see my discussion [here](https://github.com/emsesp/EMS-ESP32/discussions/2293)), but it seems as though my boiler now permanently thinks it has a CT200 attached to it, even with the key no longer attached. It won't respond to **selflowtemp**.

However, if I simulate the CT200 commands, it seems to control the boiler.

I've so far found that this:

```
call system send "0B 08 FF 00 01 E0 01 00 00 00 01"
000+12:51:26.283 T 904: [emsesp] me(0x0B) -> boiler(0x08), W, ?(0x02E0), data: 01 00 00 00 01
```

Sets the flow temperature to zero, and this:

```
call system send "0B 08 FF 00 01 E0 01 4B 64 00 01"
000+12:39:36.539 T 273: [emsesp] thermostat(0x18) -> boiler(0x08), W, ?(0x02E0), data: 01 4B 64 00 01
```

Sets it to 75°C (which must be the 4B bit) - this is the flow temperature my CT200 seems to use. I have no idea what the 64 00 01 bits are though - anyone else have any ideas?

closed 2025-01-25 by proddy