emsesp/EMS-ESP32 #2278: Retain modus offsets when new EMS entities are added

proddy, 2024-12-04:
Reported in [Discord](https://discord.com/channels/816637840644505620/816652040305901628/1313212836008296539)   

> Hi, I´m using Modbus TCP, and I wonder if it's possible to avoid frequent changes of the offset register. With each update, most of the offset numbers change, so I have to change all my numbers in the logic. 
Maybe you could keep the existing offsets, at least for a while and create new information with a higher offset number.

proddy, 2024-12-04:
@mheyse any suggestions how to do this in the scripts?

mheyse, 2024-12-04:
The offsets are already kept if new registers are added with the `generate_csv_and_headers.sh` script as documented here: https://docs.emsesp.org/Modbus-Dev/

They only change if all registers have to be re-generated when breaking changes are introduced (like modifying the tags, change data types as has happended during 3.7.0 development). This requires a dev to manually delete all mappings in `modubs_entity_parameters.hpp` .

proddy, 2024-12-06:
That makes perfect sense, thanks @mheyse

closed 2024-12-06 by proddy