emsesp/EMS-ESP32 #2284: Bosch Compress 5800i AW: Power Modulation with PV Integration Not Working After Update to v3.7.1

tomatensuppe, 2024-12-08:
### DESCRIPTION

With EMS-ESP v3.7.1, modulation options for heat pump and PV system owners have been introduced, as highlighted in the release notes. This includes features like Silent Mode and % power reduction settings. Unfortunately, after updating my system, these features do not appear to work or show up in my configuration.

I own a Bosch Compress 5800i AW heat pump and a PV system. After applying the update, none of the expected modulation settings or parameters for power reduction are visible in the EMS-ESP web interface or API.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

	1.	Update EMS-ESP to v3.7.1.
	2.	Connect a Bosch Compress 5800i AW heat pump and PV system.
	3.	Access the EMS-ESP web interface or API.
	4.	Look for power modulation settings (Silent Mode, power reduction percentages).

### EXPECTED BEHAVIOUR

After updating to v3.7.1, I expected the following:
	•	Silent Mode settings (on, auto, off) to be visible.
	•	Power reduction options (30%, 40%, 50%, 60%) to be available for configuration.

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

The modulation features appear to have been tested on Bosch CS 6800i AW heat pumps. Could this be why the Bosch Compress 5800i AW does not show the relevant parameters? If additional configuration steps are needed, they are not clear in the documentation.

[emsesp_system_info.json](https://github.com/user-attachments/files/18053509/emsesp_system_info.json)

proddy, 2024-12-21:
Those entities should still be there. Can you provide the output of `http://<ip>/api/boiler`

Also check the Customizations model to see if there present, perhaps without values

tomatensuppe, 2025-01-14:
`{"reset":"","heatingoff":"off","heatingactive":"on","tapwateractive":"off","selflowtemp":27,"heatingpumpmod":30,"outdoortemp":1.0,"curflowtemp":27.5,"rettemp":27.3,"syspress":1.7,"headertemp":0.0,"heatingactivated":"on","heatingtemp":60,"heatingpump":"on","pumpmodmax":0,"pumpmodmin":0,"pumpcharacter":"proportional","selburnpow":0,"curburnpow":0,"burnstarts":0,"burnworkmin":0,"burn2workmin":0,"heatworkmin":0,"heatstarts":0,"ubauptime":159210,"emergencyops":"off","emergencytemp":0,"nrgtotal":3822.97,"nrgheat":2685.02,"nrgcool":0.00,"metertotal":1271.43,"metercomp":1206.33,"metereheat":65.09,"meterheat":773.23,"metercool":0.00,"uptimetotal":159264,"uptimecontrol":89085,"uptimecompheating":75736,"uptimecompcooling":0,"uptimecomppool":0,"totalcompstarts":854,"heatingstarts":764,"coolingstarts":0,"poolstarts":0,"nrgconstotal":1271,"nrgconscomptotal":1206,"nrgconscompheating":773,"nrgconscompcooling":0,"nrgconscomppool":0,"auxelecheatnrgconstotal":65,"auxelecheatnrgconsheating":19,"auxelecheatnrgconspool":0,"nrgsupptotal":3822,"nrgsuppheating":2685,"nrgsuppcooling":0,"nrgsupppool":0,"hppower":0.0,"hpsetdiffpress":300,"hpcompon":"off","hpactivity":"none","hpbrinepumpspd":0,"hpswitchvalve":"off","hpcompspd":0,"hpcircspd":30,"hpbrinein":0.0,"hpbrineout":0.0,"hptc0":27.3,"hptc1":27.4,"hptc3":27.2,"hptr1":39.9,"hptr3":21.0,"hptr4":13.8,"hptr5":13.9,"hptr6":26.9,"hptr7":0.0,"hptl2":-0.4,"hppl1":0.7,"hpph1":6.1,"hpta4":16.1,"hptw1":46.9,"hp4way":"heating & dhw","hpin1":"off","hpin1opt":"000000000000000","hpin2":"off","hpin2opt":"000000000000000","hpin3":"off","hpin3opt":"100000000000000","hpin4":"off","hpin4opt":"000000000000","maxheatcomp":"9 kW","maxheatheat":"9 kW","mandefrost":"on","pvcooling":"off","auxheateronly":"off","auxheateroff":"off","auxheaterstatus":"off","auxheaterlevel":0,"auxheaterdelay":300,"auxmaxlimit":2.0,"auxlimitstart":2.0,"auxheatrmode":"eco","silentmode":"on","mintempsilent":-10,"tempparmode":10,"auxheatmix":0,"tempdiffheat":4.5,"tempdiffcool":3.0,"vpcooling":"off","heatcable":"off","vc0valve":"off","primepump":"off","primepumpmod":0,"hp3way":"off","elheatstep1":"off","elheatstep2":"off","elheatstep3":"off","hpea0":"on","hppumpmode":"auto","shutdown":"","hpcurrpower":115,"pc0flow":-1,"pc1flow":-1,"pc1on":"on","pc1rate":0,"dhw":{"nrg":1137.94,"meter":433.09,"uptimecomp":13349,"startshp":90,"nrgconscomp":433,"auxelecheatnrgcons":45,"nrgsupp":1137,"maxheat":"9 kW","alternatingop":"on","altopprioheat":30,"altopprio":120,"ecoplusoff":0,"comfdiff":13,"ecodiff":11,"ecoplusdiff":9,"comfstop":60,"ecostop":58,"ecoplusstop":53,"settemp":15,"seltemp":55,"seltemplow":51,"tempecoplus":44,"seltempsingle":70,"comfort1":"high comfort","flowtempoffset":0,"circpump":"off","hyston":0,"hystoff":0,"disinfectiontemp":70,"circmode":"1x3min","circ":"off","curtemp":46.9,"curtemp2":46.9,"curflow":0.0,"activated":"on","onetime":"off","disinfecting":"off","charging":"off","recharging":"off","tempok":"off","3wayvalve":"off","starts":0,"workm":0}}`

proddy, 2025-01-18:
indeed, I see powerReduction is missing. Let's see if it's in the telegram. do a `read 8 484` from the console or via the WebUI's System Log (blue arrow, needs Developer Mode enabled in Application Settings)

tomatensuppe, 2025-02-02:
nothing happens

[image]

proddy, 2025-02-02:
remove the 'read' if using the webUI

ChrisDeBmw, 2025-04-21:
[image]

proddy, 2025-04-22:
> [image]

`powerReduction` is at offset 64 in the x484 telegram and I see here it's missing, so perhaps stored somewhere else.

MarkusMit, 2025-05-10:
Hi, I already once started [a discussion about silent mode / powerreduction](https://github.com/emsesp/EMS-ESP32/discussions/2215), but then I realized, I wasn't on the latest EMS-ESP firmware, and after updating it looked fine, so I closed it.

But recently I realized again, that the values for the silent mode times were not available in EMS-ESP / HomeAssisant, although configured on my _Bosch Compress 6800i AW_:
[image]
```
INFO 28: [command] Called command system/read (send read request) with value 8 484
NOTICE 29: [emsesp] boiler(0x08) W me(0x0B), HPSilentMode(0x0484), data: 01 00 1E 78 00 00 00 00 00 02 01 F6 00 00 02
```
This is can be reproduced by restarting the EMS-ESP gateway.

However, the values become available again on the EMS-ESP entities when the settings are changed in the device's settings menu:
```
TRACE 84: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: 00 (offset 10)
TRACE 85: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: 00 (offset 10)
TRACE 107: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: 01 (offset 10)
TRACE 108: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: 01 (offset 10)

TRACE 137: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: 5D (offset 52)
TRACE 138: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: 5D (offset 52)
TRACE 192: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: 19 (offset 53)
TRACE 213: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: 19 (offset 53)

TRACE 248: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: F5 (offset 11)
TRACE 254: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: F5 (offset 11)

TRACE 280: [emsesp] thermostat(0x10) W boiler(0x08), HPSilentMode(0x0484), data: 06 (offset 64)
TRACE 281: [emsesp] boiler(0x08) B all(0x00), HPSilentMode(0x0484), data: 06 (offset 64)

INFO 297: [command] Called command system/read (send read request) with value 8 484
NOTICE 298: [emsesp] boiler(0x08) W me(0x0B), HPSilentMode(0x0484), data: 01 00 1E 78 00 00 00 00 00 02 01 F5 00 00 02
```

[image]