emsesp/EMS-ESP32 #2285: Unrecognized EMS device (deviceID 0x30, no productID)

nikeles1, 2024-12-08:
Hello,
I recently purchased an EMS-ESP from BBQKees. Unfortunately, I'm experiencing an issue with one module that doesn't seem to be recognized, which in turn causes my heating system to lose connection with the device. The module in question is the SM10 module.

This is the error message:
2024-12-08 20:09:31.300 E 2624: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 02 00 FF
2024-12-08 20:09:31.300 W 2625: [emsesp] Unrecognized EMS device (deviceID 0x30, no productID). Please report on GitHub.

proddy, 2024-12-19:
> "which in turn causes my heating system to lose connection with the device"

can you elaborate more on what is happening?

nikeles1, 2024-12-26:
When the heating system runs without the EMS-ESP module, everything works without any issues. However, when the EMS-ESP is plugged in via the service jack, the communication with the SM10 no longer works. I have only tested the SM10 myself and unplugged everything else, but the heating system always loses the connection to the SM10 as soon as the EMS-ESP module is plugged in.

nikeles1, 2025-01-01:
On the bus is a GB142 (BC10), RC30, MM10, WM10 and the SM10. My neighbour has the exact same setup like me and it works just fine. The only difference I can tell is, that the modules and the GB142 has a slightly newer version of the firmware because his heating system has been built one year past mine.

Multiple buses can't be the problem, because I have no heatpump or anything else. Different TxMode doesn't change anything and I tried all of these. Same with the Bus IDs.

proddy, 2025-01-04:
I don't know what could cause it, sorry. Maybe @MichaelDvP has an idea?

nikeles1, 2025-01-11:
[image]
Does this help anything?

proddy, 2025-01-25:
not really. the problem you describe is that when you plug in in the Gateway board to the service jack, the SM10 stops functioning. Does it work if you don't use the service jack? What do you see in the logs? Perhaps try a different bus id? I can't see why EMS-ESP would impact the EMS network as it just sits there and listens as a 'slave' device.

nikeles1, 2025-01-27:
I already tried installing the esp direcly on the bus and not via service jack. Its always the same error as I mentioned in the beginning, but here again the error message from right now:
2025-01-27 22:06:29.277 E 2281535: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 02 00 FF
2025-01-27 22:06:29.278 T 2281536: [emsesp] 30(0x30) W me(0x0B), Version(0x02), data: <empty>
2025-01-27 22:06:29.278 W 2281537: [emsesp] Unrecognized EMS device (deviceID 0x30, no productID). Please report on GitHub.

I understand that its just working as a slave and dont get it.

proddy, 2025-02-22:
If I understand correctly, when EMS-ESP is plugged into the EMS bus, your SM10 stops communicating to the boiler? Have you tried
- different Bus IDs?
- setting EMS-ESP to read-only mode
- turning on Developer Mode, go to Status->System Log, blue triangle and typing '30 2' and see what comes back in the log

Also without the System Information we can't really help you further.

nikeles1, 2025-03-02:
I have already tried every single Bus ID in read-only mode as I mentioned above. But i tried it now again and nothing changes.

When I type "30 2" it replies with an error:
2025-03-02 16:59:52.526 E 261: [system] Invalid device ID (48) in read command
2025-03-02 16:59:52.526 W 262: [command] Command read: 30 2 failed (Error)

For System Information: I got the BBQKees Gateway E32 V2

proddy, 2025-03-02:
Ok, the read command failed because the SM10 is not found. You'll need to use the Console (telnet) and do a `read 30 2` (https://docs.emsesp.org/Console/)

nikeles1, 2025-03-02:
In Telnet the same error is coming:

[image]

proddy, 2025-03-02:
> In Telnet the same error is coming:
> 
> [image]

oh, silly me, it's the same command.

nikeles1, 2025-03-02:
now i tried custom entities and they function. But the error:
2025-03-02 17:46:04.197 E 2066: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 0A B0 02 00 FF
2025-03-02 17:46:04.198 T 2067: [emsesp] 30(0x30) W me(0x0A), Version(0x02), data: <empty>
2025-03-02 17:46:04.198 W 2068: [emsesp] Unrecognized EMS device (deviceID 0x30, no productID). Please report on GitHub.

is still coming on the protocol

MichaelDvP, 2025-03-02:
Please check the cables between the modules. Try a different order. I have a nearly similar system with GB125, BC10, RC35, MM10, SM10 and two ems-esp.
My connection is: RC35 is in BC10, Bus from BC10 to first(left) ems on SM10, second ems connector (on the edge) to MM10 white ems, MM10 orange connector to ems-esp.  The other ems-esp is on service jack.
[image]

nikeles1, 2025-03-02:
I tried the exact same setup and the same errors appear :(

MichaelDvP, 2025-03-03:
Where is the WM10 connected. Just try a different order. Have you all in a row, or somewhere a Y-connection?

nikeles1, 2025-03-03:
I did try every possible order and no there is no y connection anywhere

MichaelDvP, 2025-03-04:
Strange that it's only and always the SM10 with no connection, also if you change order and/or cables.
(BTW: cable length should always be > 20cm).
I'm out of ideas now, sorry, can't help.

proddy, 2025-03-23:
Moving to a discussion