emsesp/EMS-ESP32 #2287: Away-Mode for BC400 with GB192i*2

mehlior680, 2024-12-10:
Find out by "watch".

The away mode (at home, not at home) can be added by 
0x10 0x16E 0 INT8 

Value 0 will be at home (00)
Value -1 will be not at home/away (FF)

[image]

It would be nice, if you could add it to the code. 

Many thanks!

proddy, 2024-12-10:
thanks. Is the official name (used by Bosch/Buderus) also home? We need the shortname and longname so we can add along with the language translations

MichaelDvP, 2024-12-11:
See #1957 
But still unclear if this is a hc specific setting with hc1: 0x16E, hc2: 0x16F, ... or global for all heating circuits.

closed 2025-03-22 by proddy