emsesp/EMS-ESP32 #2288: Missing Entity "Absenkschwelle" for switchPrograms with mode "Absolute"

tp1de, 2024-12-11:
@Marcwa19197 mentioned a missing entity within #1594

Within switchprogmode 2 modes can be selected: "Levels" and "Absolute".

Using exact temperatures ("Absolute") you need to also define the "Absenkschwelle" under which temperature nightmode / reduced mode will be activated. This value is also missing right now in EMS-ESP, 

[image]

MichaelDvP, 2024-12-11:
We need device, telegram, offset, type, name to implement.
- help-page: support-info download button, attach it here
- status-logger: log all for more than a minute
- change the `Absenkschwelle` while logging
- attach the log here.

Marcwa19197, 2024-12-12:
1.)
[emsesp_system_info.json](https://github.com/user-attachments/files/18115067/emsesp_system_info.json)

2.)
[log.txt](https://github.com/user-attachments/files/18115091/log.txt)

i changed Absenkschwelle from 20 to 19.5 and back to 20.

MichaelDvP, 2024-12-12:
Looks good, try if this works:
[image]

If this is ok we'll add it to thermostat.

Marcwa19197, 2024-12-12:
Works! Thank you very much!

Marcwa19197, 2024-12-13:
min. Value is 12
max. Value is 22

proddy, 2024-12-21:
Think we just need to add the mix/max values and this issue can be closed.

commit ba3ae5ea (2024-12-23): Merge pull request #2305 from MichaelDvP/dev

fixes #2288, #2299, #2295 (new  FLAG)
closed 2024-12-23 by proddy