emsesp/EMS-ESP32 #2290: CW100 thermostat - telegram 0x241

SoftVienka, 2024-12-13:
While investigating my CW100 thermostat I came across several entities in telegram 0x241. 
More details in the attached pdf.

[18_241_telegramm.pdf](https://github.com/user-attachments/files/18124555/18_241_telegramm.pdf)

MichaelDvP, 2024-12-13:
Strange, we read these settings from 0x0240:
https://github.com/emsesp/EMS-ESP32/blob/8c424c7a64029f2d88a976c14d983f95db63a23e/src/devices/thermostat.cpp#L1275-L1282
Please post output for `read 18 240` and `read 18 241`

SoftVienka, 2024-12-13:
[log.txt](https://github.com/user-attachments/files/18126299/log.txt)

MichaelDvP, 2024-12-13:
Should work with with: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
Please check.

SoftVienka, 2024-12-13:
Checked, works well, including change via EMS-ESP

VlastiBroucek, 2024-12-13:
My Australian CR50 shows:
```
ems-esp:$ read 18 240
004+06:15:19.660 N 1: [emsesp] thermostat(0x18) -> me(0x0B), W, RC300Settings(0x0240), data: <empty>
ems-esp:$ read 18 241
004+06:15:23.480 N 2: [emsesp] thermostat(0x18) -> me(0x0B), W, ?(0x0241), data: 32 00 02 00 00 00 00 00 FF 02 F6 06 FF 00 00
ems-esp:$
```

closed 2024-12-23 by MichaelDvP