emsesp/EMS-ESP32 #2294: int8 data type

GLT-IN, 2024-12-18:
### DESCRIPTION

Hi,
int8 values are difficult to interpret by second devices, because they are transmitted as int16.
For example minexttemp (register offset: 51) with value "-16" is transmitted as 00 F0, but interpreted as "240".
Is it possible to send as full int16?

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

[image]

MichaelDvP, 2024-12-18:
I think it's only one small change, please try https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
and check if all is solved.

GLT-IN, 2024-12-19:
Hi,
int8 values are represented correctly now, but the whole system does not run stable.
Devices and entities getting lost and found randomly.

GLT-IN, 2024-12-19:
every minute the device restarts with v3.7.2-dev.5

MichaelDvP, 2024-12-19:
Do the restarts happen only in my version, or also in official v3.7.2-dev.5? 
What happen every minute? Commands/reads sent to emsesp?

GLT-IN, 2024-12-19:
it happens in both versions
[image]

GLT-IN, 2024-12-19:
i changed the register addresses and there are no device restarts any more

closed 2024-12-19 by GLT-IN

MichaelDvP, 2024-12-19:
> i changed the register addresses and there are no device restarts any more

Wrong address or length should only give an errormessage, but no crash. 
I'v found a candidate, but not sure if it is fixed now.
Please report if you see this crash again.