emsesp/EMS-ESP32 #2295: CR11 thermostat not recognized by EMS-ESP (Nefit Proline)

shostakovits, 2024-12-18:
### DESCRIPTION

My new CR11 thermostat is not recognized by EMS-ESP (Nefit Proline).
The log gives me the following line:

2024-12-18 20:57:22.388 NOTICE 31: [emsesp] Unrecognized EMS device (deviceID 0x18, productID 10). Please report on GitHub.

I also searched, but there is very little info on the CR11 thermostat...

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [X] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

Paste System information here....
[emsesp_info.json](https://github.com/user-attachments/files/18188307/emsesp_info.json)

### TO REPRODUCE

Install a CR11 thermostat on my Nefit Proline.

### EXPECTED BEHAVIOUR

I expected the thermostat would be recognized by EMS-ESP, so i could use it in Home Assistant

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

Log file, when the CR11 was connected. Including setting temperature to -.5 & -1 and +.5 & +1

Also the GUI of EMS-ESP shows it's 'seeing' an "CR11/unknown (0)"

The CR11 thermostat IS working correctly on my Nefit Proline. Only the reported(measured) temperature is about 4 degrees higher than the actual temperature of the room.

[log.txt](https://github.com/user-attachments/files/18188335/log.txt)

proddy, 2024-12-18:
We can add the device (deviceID 0x18, productID 10) and see how far we get

shostakovits, 2024-12-18:
That was fast.
Eh, do you need any more information from my side?
Or is the log file enough?

proddy, 2024-12-18:
Either me or Michael can add it to the next PR (it's just one line to add) and then you'll need to tell us what works and doesn't work.

MichaelDvP, 2024-12-19:
Wow, first post with system info and a log, Also the change of setpoint in the log was a very good idea and makes it easy for us. Thank you.

Please check https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Can you also give me the result of `read 18 241` and `read 18 240`. You can use the readcommand  in telnet or with developer mode in the web logger. (There was a change for #2290, maybe this affects also your thermostat).

shostakovits, 2024-12-19:
I've tested the new version. The thermostat is recognized. Thank you.

But there are some other issue's which are popping up...

First let me tell you the results of the 2 command's you've requested:

`read 18 241 --> 000+00:09:12.376 N 2: [emsesp] thermostat(0x18) W me(0x0B), RC300Settings(0x0241), data: <empty>`
`read 18 240 --> 000+00:09:14.915 N 3: [emsesp] thermostat(0x18) W me(0x0B), ?(0x0240), data: <empty>`

Now my new issue's:

With my old RC10/Moduline100 thermostat I could set the room temperature with "hc1 selected room temperature" or the "hc1 day temperature". As seen in the following screenshot.
[image]

Also the thermostat entity was working in HA:
[image]

With the new CR11 i'm unable to set the temperature with "hc1 selected room temperature"
But i have to use the "hc1 manual temperature". See screenshot.
[image]

When I try to add the thermostat to HA, it won't let me. Hc1 is 'unavailable'
What I also notice, "Hc1 cooling on" is turned on. I don't have cooling in my Nefit Proline.
I don't know if this supposed to happen, but maybe you can tell me.
[image]

If you want any new logs, please tell me if you want me to do some specific actions/things so you can analyze correctly.

shostakovits, 2024-12-19:
Here you go, 2 minutes log. (included 1 extra minute, because of boot time & detection)

[log_3_minutes.txt](https://github.com/user-attachments/files/18199380/log_3_minutes.txt)

MichaelDvP, 2024-12-19:
Oh, a lot of empty fields. The setting telegram 0x2B9 has only a single value for manual setpoint. No mode or other settings.
Does this thermostat allows to change the mode?  Does not look like. Also there is no clock to provide auto mode.
I think we need a new setting flag for this thermostat to set the mode fixed to manual for HA widget and seltemp.
As quick workaround i set the mode to manual if not set in the telegram 0x2B9. This should work, but still shows  silly values for modetype, summermode, cooling, etc.

Updated build: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

shostakovits, 2024-12-19:
I've just tested the new build, and it's working like a charm.
I can set temperature, including in HA.

The CR11 is the successor of the CR10. It has 4 modes but I can only use it as a thermostat. ("old" Nefit)
You have to set the mode prior to using it, and it cannot easily be changed after. (I have to reset the unit to do that)

So for me it's usable, but if there is anything I can assist to better implement this thermostat, please let me know.

Thank you very much.

So do I close this issue for now, or do you want it left open, so I can maybe assist? Let me know.

proddy, 2024-12-20:
We'll close it. Thanks for the collaboration, and thanks to Michael for the quick implementation.

closed 2024-12-20 by proddy

MichaelDvP, 2024-12-22:
@shostakovits Please check the build https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
I've added a flag for this thermostat and show only the relevant values i've found in the telegrams.
This is a much cleaner solution. It should work, but i can't test the HA integration.

shostakovits, 2024-12-22:
I've tested the build. It looks like it's working, but it isn't. (sorry)

I'm unable to change the temperature in the GUI of the EMS-ESP.
I've created a log, but it only says "it fails". So i hope this makes sense to you

[log_4.txt](https://github.com/user-attachments/files/18222945/log_4.txt)

After I did a couple of tries, I wanted to view the status page, logs. But unfortunately I was presented with this error:

[image]

The best way to reproduce this is (with Firefox, 133.0.03):
- Open tab to EMS-ESP dashboard page
- Replicate same tab
- In replicated tab, click the status 'link' on bottom screen
- Boom, error.

I'm doubting this error is related to my issue, but if you want I can make a new issue with this error.
For now I will leave it here.

Just to be sure, I restarted the EMS-ESP, en repeated the above. Same outcome.
Gui showed me a quick message that "Request failed" when changing the temperature in the GUI.

Log included, just to be sure:

[log_5.txt](https://github.com/user-attachments/files/18222962/log_5.txt)

MichaelDvP, 2024-12-23:
I've fixed the seltemp command.
For the web-page error try to press Ctrl-Shift-R to refresh brower cache.  
I can not reproduce this error.

shostakovits, 2024-12-23:
I'm able to set the temperature in the EMS gui & in HA with the new build.

Thank you very much for the swift fix & communications.

MichaelDvP, 2024-12-23:
The datetime bug (reproduced it)  is also fixed half a hour ago in this build.