emsesp/EMS-ESP32 #2304: Unexpected Application Error!  date value is not finite in DateTimeFormat format()

bakboord4, 2024-12-22:
### DESCRIPTION

Unexpected Application Error!

date value is not finite in DateTimeFormat format()

format@[native code]
@http://192.168.2.148/assets/index-CKWZUDcq.js:1:187359
Go@http://192.168.2.148/assets/index-CKWZUDcq.js:1:187409
L@http://192.168.2.148/assets/vendor-BWDhLb9H.js:1:6392
P@http://192.168.2.148/assets/vendor-BWDhLb9H.js:1:1624
💿 Hey developer 👋

You can provide a way better UX than this when your app throws errors by providing your own ErrorBoundary or errorElement prop on your route.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
 {"system":{"version":"3.7.2-dev.6","uptime":"000+00:16:16.985","uptimeSec":976,"platform":"ESP32","cpuType":"ESP32-D0WD","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":174,"maxAlloc":107,"freeCaps":124,"usedApp":1747,"freeApp":237,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-31,"WIFIReconnects":0,"BSSID":"set","TxPowerSetting":34,"staticIP":true,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"ntp.time.nl","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":313,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"ems-esp","keepAlive":60,"cleanSession":true,"entityFormat":0,"base":"NefitCV","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":863,"busReads":153,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":1,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32","locale":"nl","txMode":2,"emsBusID":11,"showerTimer":true,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"CBS/GB*72, Trendline, Cerapur, Greenstar Si","deviceID":"0x08","productID":123,"brand":"Nefit","version":"07.00","entities":83,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x14 0x16 0x33 0x26 0x28","handlersPending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B","handlersIgnored":"0x23"},{"type":"thermostat","name":"RRC2/EasyControl, CT200","deviceID":"0x18","productID":203,"brand":"Bosch","version":"05.02","entities":3,"handlersReceived":"0x06 0x0A 0x02A5","handlersPending":"0xA3 0xA2 0x12 0x13","handlersIgnored":"0x33 0x18"},{"type":"controller","name":"BC10","deviceID":"0x09","productID":190,"brand":"","version":"01.04","entities":0}]}

```

### TO REPRODUCE

DEV6 + NTP adjustment

### EXPECTED BEHAVIOUR

As intended.

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

Systems development is not my habitat

proddy, 2024-12-22:
Thanks for reporting. We'll try and reproduce.

MichaelDvP, 2024-12-23:
I can not reproduce, looks like a browser cache issue. Try `ctrl-shift-R` to update cache.

shostakovits, 2024-12-23:
I'm still getting the same errors.
I've refreshed with `ctrl-shift-R` but no luck.

I've cleared cookies & history, open the GUI in an incognito tab, but still the same error.
The strange thing is: I can only reproduce this in the "status" tab. I'm unable to do so in the NTP settings page like @bakboord4 . 

Then I enabled the `webontwikkelaarshulpmiddelen` (CTRL+SHIFT+I or F12) in firefox. Still the same error, but with more details (I think).

I hope this helps.

See screenshot:

[image]

MichaelDvP, 2024-12-23:
Ah, updated my local build to latest commit https://github.com/emsesp/EMS-ESP32/commit/b0c29b57c7085ba0d27a0a17cf800132af088ccd and can reproduce now.
Seems status shows data.ntp_time, but this is not send by emsesp.

bakboord4, 2024-12-23:
@MichaelDvP additional info. MacOS 16 (ARM), iPadOS & IOS 18 are showing the same issue.

proddy, 2024-12-23:
that's my bad, sorry, should have tested my last PR better. We'll fix it.

MichaelDvP, 2024-12-23:
I fixed to local time here: https://github.com/MichaelDvP/EMS-ESP32/commit/4c69c9e445ef01168f70d5416afe04b8f1fe3a98
@proddy Is it the way you wanted it?

I also realized that we once had a local time input if ntp is disabled, but this is gone after refactoring the web. (sending `rest/time`).

proddy, 2024-12-23:
> I fixed to local time here: [MichaelDvP@4c69c9e](https://github.com/MichaelDvP/EMS-ESP32/commit/4c69c9e445ef01168f70d5416afe04b8f1fe3a98) @proddy Is it the way you wanted it?

Thanks for fixing this. Yes, I though having a Status page that just showed "Active" for MQTT and NTP was very helpful. Next would be to show some MQTT meaningful metrics, for example queue or errors.

bakboord4, 2024-12-23:
@proddy @MichaelDvP  Downloaded a new Dev 6 version (#2295). Status is showing ntp values. For me it looks my issue is solved and will close this issue.
Thanks for your efforts put in.

closed 2024-12-23 by bakboord4