emsesp/EMS-ESP32 #2306: API errors for boiler.silentmode

tp1de, 2024-12-26:
@UlfHRO opened on my ioBroker adapter a GH issue for boiler.silentmode:  https://github.com/tp1de/ioBroker.ems-esp/issues/159

We recognized that with API/MQTT settings on "Value" the values are not send correctly with API call on .../api/boiler.
The result is a mix on index and values which gave errors on the ioBroker adapter.
Using the selector "Index" works fine, "Value" not.

This should be corrected.

proddy, 2024-12-27:
I assume you mean enums. Can you provide an example? I can't reproduce. If I look at api/boiler and take maintenance for example it correctly shows either the value or the index

tp1de, 2024-12-27:
It does not work correctly for one entity: boiler.silentmode (heat pump) - yes enum entity
All other enums are correctly send with settings "Value" and .../api/boiler reads.

I can't test since I do not have a heatpump.
"Index" is fine but "Value" returns a mix of Index and Values.

@UlfHRO logged the entity with my ioBroker adapter: 
(these values are from .../api/boiler polling - with API/MQTT settings to Value and then changing between Off/Auto/On)

Auto: ems-esp.0 2024-12-26 17:36:09.731 info boiler.silentmode auto 
Off:  ems-esp.0 2024-12-26 17:36:39.745 info boiler.silentmode 0
On:  ems-esp.0 2024-12-26 17:37:09.732 info boiler.silentmode 1

Auto is correct (auto : index 1) 
Off is the index - 0 is correct but but value missing
On is index 1 - value missing and correct index should be 2

This was reported to be an ioBroker adapter bug. It took me some efforts to find the reason of missing functionality.

MichaelDvP, 2024-12-30:
Looks to me as the adapter handle "on" as boolean and send a 1 instead of enums 0:off, 1:auto,  2:on
maybe here: https://github.com/tp1de/ioBroker.ems-esp/blob/fe053678b5f7b62ba3d83cfb08aa682628cebc39/lib/ems.js#L1131

tp1de, 2024-12-30:
Good remark Michael. I will check.

closed 2024-12-30 by tp1de

tp1de, 2024-12-30:
Anyhow this mapping does not work anymore since values for ON/OFF are translated.
The boolean select options for API/MQTT are not taken 1:1.