emsesp/EMS-ESP32 #2308: Setting custom temperature via remotetemp not working realiably

Vippis2000, 2024-12-28:
### DESCRIPTION
When using EMS-ESP internal command remotetemp the temperature value (source is a HMIP thermostat) will be set in 1 out of 10 trials and gets overwritten pretty soon. I have a physical CR10H thermostat wired to the HP and located in the living room, but I want the HP to heat until a target temperature of another room (Master bathroom) is reached, which has the far most distance in the HC from the source.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {
  "system": {
    "version": "3.7.2-dev.6",
    "uptime": "000+18:42:11.594",
    "uptimeSec": 67331,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 183,
    "maxAlloc": 147,
    "freeCaps": 8318,
    "usedApp": 2301,
    "freeApp": 5827,
    "partition": "app1",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8135,
    "model": "",
    "temperature": 41.8
  },
  "network": {
    "network": "WiFi",
    "hostname": "EMS-ESP-Bosch",
    "RSSI": -51,
    "WIFIReconnects": 0,
    "TxPowerSetting": 78,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 35249,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": false,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": true
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0,
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 4,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 285451,
    "busReads": 55330,
    "busWrites": 283,
    "busIncompleteTelegrams": 1161,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "de",
    "txMode": 2,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": true,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "CUH/Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i",
      "deviceID": "0x08",
      "productID": 172,
      "brand": "",
      "version": "01.16",
      "entities": 159,
      "handlersReceived": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlersFetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x02D6",
      "handlersPending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF 0x04A5 0x04AA 0x04A7",
      "handlersIgnored": "0x049F 0x04A0 0x04A3 0xE7 0x02E0 0x02EA 0x061E 0x0240 0x0291 0x029B 0x0497 0xF6 0xF7"
    },
    {
      "type": "thermostat",
      "name": "HPC400/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "13.07",
      "entities": 68,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x0269",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x016E",
      "handlersIgnored": "0xBF 0x02C3 0x02FF 0xF7 0xF6"
    },
    {
      "type": "thermostat",
      "name": "CR10H/RC100H, CR10H",
      "deviceID": "0x38",
      "productID": 200,
      "brand": "",
      "version": "40.04",
      "entities": 4,
      "handlersReceived": "0x042B 0x047B 0x0273",
      "handlersPending": "0x0A6A",
      "handlersIgnored": "0xBF 0x10"
    },
    {
      "type": "gateway",
      "name": "KM200, MB LAN 2",
      "deviceID": "0x48",
      "productID": 189,
      "brand": "",
      "version": "04.08",
      "entities": 0,
      "handlersIgnored": "0xF9 0x0488 0x02C3 0x02FF 0x0CBD"
    }
  ]
}

```

### TO REPRODUCE

Use remotetemp command

### EXPECTED BEHAVIOUR

The entered temperature should be set as currtemp

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

Vippis2000, 2024-12-29:
System Protocol:

`001+19:23:15.305    INFO 673175: [command]    Called command thermostat/remotetemp (Raumtemperatur Remote) with value 20 and id 1 on device 0x10
001+19:23:15.379   TRACE 673176: [telegram]   Rx: 8B 90 FF 00 19 01 92 C1
001+19:23:15.428   TRACE 673177: [telegram]   Rx: 90 0B FF 00 01 92 00 00 00 00 FF 00 04
001+19:23:15.500   TRACE 673178: [telegram]   Incomplete Rx: B8 10 FF 00 E0 2B 00 C8 C5
001+19:23:15.744   TRACE 673179: [telegram]   Rx: C8 90 FF 04 01 01 A8 F2
001+19:23:15.761   TRACE 673180: [telegram]   Rx: 90 48 FF 04 01 A8 52
001+19:23:15.911   TRACE 673181: [telegram]   Rx: 88 00 FF 00 03 8F 01 B1 01 B5 01 9B 01 8A 00 00 00 00 FF FD 01 93 01 81 FF F7 FF F0 00 09 5C
001+19:23:16.105   TRACE 673182: [telegram]   Rx: 88 00 FF 18 03 8F 01 89 FF E1 01 B9 F2
001+19:23:16.323   TRACE 673183: [telegram]   Rx: 88 00 FF 00 03 94 08 E4 68 64 00 00 E1 CD 00 00 44 6E 00 00 99 32 00 00 04 1B 00 00 00 00 2D
001+19:23:16.523   TRACE 673184: [telegram]   Rx: 8B 90 FF 00 19 01 93 C0
001+19:23:16.553   TRACE 673185: [telegram]   Rx: 90 0B FF 00 01 93 00 00 00 00 FF 00 44
001+19:23:16.704   TRACE 673186: [telegram]   Rx: C8 90 FF 00 02 01 A8 DE
001+19:23:16.720   TRACE 673187: [telegram]   Rx: 90 48 FF 00 01 A8 42
001+19:23:16.720   TRACE 673188: [telegram]   Rx: C8 90 FF 07 03 01 A8 E2`

MichaelDvP, 2024-12-30:
You can't use a virtual remte thermostat if you alrrady have a physical connected.

Vippis2000, 2024-12-30:
So I need to unplug the physical CR10H if I'd like to use a virtual thermostat?

Vippis2000, 2024-12-30:
I can confirm that by unplugging the physical CR10H thermostat, I can now use remotetemp as intended!

To make it easier to roll back I have just applied some tesa tape on the electrical contacts inside the CR10H, so that i remains electrically disconnected.

closed 2024-12-30 by Vippis2000