emsesp/EMS-ESP32 #2309: Unexpected Application Error! - date value is not finite in DateTimeFormat format()

HansRemmerswaal, 2024-12-28:
When clicking on **Status** I'm getting the following error;

[image]

The same issue happens on my iPhone, iPad and Mac. 
I have restarted the EMS-ESP but the issue is still there

I'm using 3.7.2-dev.6, see attached emsesp_system_info for more details

[emsesp_system_info.json](https://github.com/user-attachments/files/18267896/emsesp_system_info.json)

proddy, 2024-12-29:
it could be an error left over from dev.5. Have you tried refreshing the cache. In Chrome it's CTR + R then CTRL + F5. Or just removing the cache with `chrome://settings/clearBrowserData`

HansRemmerswaal, 2024-12-29:
What I did this morning;

- I cleared the cache and the issue was still there
- I took the power off from the EMS-ESP and after about 30 sec I put the power on again, the issue was still there
- I downgraded to version 3.7.1 and the issue was gone
- I upgraded to version 3.7.2 dev 6 and the issue is still gone

So somehow downgrading and upgrading has solved the problem...

HansRemmerswaal, 2024-12-29:
Closing the issue as it's resolved for me, but the root cause is unknown.

HansRemmerswaal, 2024-12-29:
The only data time in the Status tab is the NTP. So I guess that somehow there was an invalid date...?

closed 2024-12-29 by HansRemmerswaal

MichaelDvP, 2024-12-30:
dev6 was published with the bug, a few hours later this was reported in #2304 and fixed without counting up the dev number, so actual dev6 is the fixed version.