emsesp/EMS-ESP32 #2310: Temperature entity with 1 decimal making jumps of more than 1 degrees

HansRemmerswaal, 2024-12-29:
### Issue

The entity **Boiler Air inlet temperature (TL2)** on my heat pump is showing 1 decimal, but when I look at the data over time I see that the sensor is making jumps of 1, 1.1, 1.2... degrees. E.g the temperature jumps from 4.6 - 3.6 - 5.0 - 3.9 - 2.9 - 4.3 ...

When I check the value between the heat pump itself and the EMS-ESP the values are in sync, but maybe I was just each time checking at a moment that they are in sync. With a temperature having 1 decimal you would expect a more continuous change.

Other owners, having the same brand but a different type (mono block vs split), are having a continuous change for this entity. So either it's something strange in the firmware of the heat pump or... something is going wrong when processing the EMS telegrams.

### Question

Is there a way that I can capture the telegrams for this entity to see if indeed they are making these big jumps?

### Screenshots

**Example of my TL2 (split)**

[image]

**Example of the TL2 (same brand but monoblock)** 

[image]

### EMS-ESP system info

[emsesp_system_info (1).json](https://github.com/user-attachments/files/18269362/emsesp_system_info.1.json)

MichaelDvP, 2024-12-30:
TL2 is telegram 0x48F, offset 12 as INT16. You can watch the telegram in logger or use custom entity to see the changes.
I guess the telegram is sent out rarly so the changes are bigger. Maybe in some systems a bosch internet gateway queries the value more often.

HansRemmerswaal, 2024-12-30:
Maybe I do something wrong...

- I connected using Putty
- Entered watch 0x48f
- See a number of logging, all for HpTemperatures but no "offset 12"
- After rebooting the ems-esp all the entities gets populated, also the TL2, so I did the above again right after reboot and before I got the TL2 value in Home Assistant. Same result.

**As shortly after a reboot the entities gets populated, it suggests that the ems-esp is sending a kind of telegram to request the values? Is this assumption correct and can I do the same to read the value of TL2?**

I see we have the read <deviceID> <type ID> [offset] [length] but I can't make it working, also not after using the su. In the manual I can't find an example of how to use it.

ems-esp:$ su
Password:
000+00:12:15.014 N 19: [shell] Admin session opened on console pty0
ems-esp:# read 0x48f
Not enough arguments for command
ems-esp:# read 0x48f offset 12
Invalid deviceID
000+00:12:39.414 E 20: [system] Invalid device ID (143) for read command

000+00:00:06.622 I 0: [shell] User session opened on console pty0
ems-esp:$
ems-esp:$
ems-esp:$
000+00:00:07.339 I 1: [emsesp] Detected EMS device: boiler (0x08)
000+00:00:10.179 I 2: [emsesp] Detected EMS device: thermostat (0x10)
000+00:00:10.556 I 3: [emsesp] Detected EMS device: thermostat (0x38)
000+00:00:10.956 I 4: [emsesp] Detected EMS device: gateway (0x48)
ems-esp:$ watch 0x48f
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x48F
000+00:00:46.991 N 5: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 E7 00 E7 00 E9 00 00 00 00 00 5E 00 00 00 DC 00 5D 00 8F
000+00:00:47.176 N 6: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 95 00 24 00 3C 00 E3 (offset 24)
000+00:01:47.054 N 7: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 E6 00 E7 00 E9 00 00 00 00 00 5E 00 00 00 DC 00 5D 00 8F
000+00:01:47.239 N 8: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 95 00 24 00 3C 00 E3 (offset 24)
000+00:02:46.955 N 9: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 E6 00 E7 00 E8 00 00 00 00 00 5E 00 00 00 DC 00 5D 00 8F
000+00:02:47.139 N 10: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 94 00 24 00 3C 00 E3 (offset 24)

MichaelDvP, 2024-12-30:
The terminal watch command don't like the 0x prefix, we should add some code to allow it.
Seems the telegram is only send every minute, so faster changes are not reported and can lead to steps.
Try if you get some intermediate values if you manually use some `read 8 48F` maybe the temperature is swinging up and down more than one degree withing the minute.

HansRemmerswaal, 2024-12-31:
Trying to understand how to read the telegram... Current value for TL2 = 5.3

ems-esp:# read 8 48F
000+17:51:34.042 N 95: [telegram] Rx: 0B 88 FF 00 19 03 8F F8
000+17:51:34.078 N 96: [telegram] Rx: 08 0B FF 00 03 8F 01 72 01 1F 01 20 01 04 00 00 00 00 00 35 00 00 01 07 00 09 00 0C D4
000+17:51:34.083 N 97: [emsesp] boiler(0x08) W me(0x0B), HpTemperatures(0x048F), data: 01 72 01 1F 01 20 01 04 00 00 00 00 00 35 00 00 01 07 00 09 00 0C

08 = sender = boiler(0x08)
0B = receiver = me(0x0B)
FF = EMS-PLUS indicator
00 = offset
03 8F = EMS-Plus telegram type = HpTemperatures(0x048F)
01 72 01 1F 01 20 01 04 00 00 00 00 00 35 00 00 01 07 00 09 00 0C D4

How do I continue? The last line is the data and CRC according to the [documentation ](https://emswiki.thefischer.net/doku.php?id=wiki:ems:plus-telegramme)I could find.

The current value is 5.3 which is stored somehow in an int16 with min/max value -3199/3199. How is this encoded in this data?

Another question, why is it 0x048F while in the raw data I see 038F?
**03** 8F = EMS-Plus telegram type = HpTemperatures(0x**04**8F)

device name,device type,product id,shortname,fullname,type [options...] \| (min/max),uom,writeable,discovery entityid v3.4,discovery entityid,modbus unit identifier,modbus block,modbus scale factor,modbus offset,modbus count
"Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco/Geo 5xx, WLW196i/WSW196i",boiler,172,hptl2,air inlet temperature (TL2),int16 (>=-3199<=3199),C,false,sensor.boiler_air_inlet_temperature_(TL2),sensor.boiler_hptl2,5,0,1/10,159,1

Sbried, 2024-12-31:
If you look at offset 12/13 (int16 = 2 bytes) of your data you find hex 00 35 which calculates to dec 53. The value is sent in 0,1°C resolution, so after dividing by 10 you get your 5,3°C.

To your other question: to save memory also the EMS-plus data types are stored in 2 bytes. To Identify them 0x0100 is added. See #2084 and #2025

proddy, 2024-12-31:
great answer @Sbried - you beat me to it

HansRemmerswaal, 2024-12-31:
I completely overlooked that 0x35... For some reason I converted another decimal value to hex and was searching for that one...

if it's broadcasted every 60 sec then I do not understand why the temperature is **always** changing with at least 1 degree. E.g. 4.6 - 3.6 - 5.0 - 3.9 - 2.9 - 4.3 ... That suggest a problem with the firmware in my heat pump as it's not likely that it's always changing within 1 minute with 1 degree of more.

Sbried, 2024-12-31:
I looked at my TL2 for the last 5 mins: 0,6 - 0,7 - 0,8 - 0,9 - 0,5 - 0,4 - 0,5 - 0,9 - 0,6 - 0,5 - 0,7 - 0,5 - ...
And it´s broadcasted every 10 -20 seconds. Much more frequent than every minute.

HansRemmerswaal, 2024-12-31:
@Sbried for the NEFIT/BOSCH it seems to be a difference between SPLIT and MONO block. The mono block is giving a continuous temperature. The firmware version for the MONO block are also higher than for the SPLIT, so maybe a defects that has been resolved in the firmware.

Below an example where within a few seconds it jumps from 3F (6.3) to 4A (7.4).
Strange also that the time stamps are no in sequence...

But anyhow, seems to be a problem in the firmware of my NEFIT. Closing the issue...

000+09:21:31.552 N 169: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 F0 00 F0 00 F2 00 00 00 00 00 **3F** 00 00 00 DF 00 3E 00 8C
000+09:21:31.737 N 170: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 8F 00 0C 00 25 00 ED (offset 24)
000+**09:22:31**.639 N 171: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 EF 00 F0 00 F2 00 00 00 00 00 **3F** 00 00 00 DF 00 3E 00 8C
000+**09:22:31**.824 N 172: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 8F 00 0C 00 25 00 ED (offset 24)
000+**09:23:24**.240 N 173: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 EF 00 F0 00 F1 00 00 00 00 00 **4A** 00 00 00 DF 00 3E 00 8C
000+**09:23:24**.425 N 174: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 8F 00 0C 00 25 00 EC (offset 24)
000+**09:23:31**.477 N 175: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 96 00 EF 00 F0 00 F1 00 00 00 00 00 **4A** 00 00 00 DF 00 3E 00 8C
000+**09:23:31**.662 N 176: [emsesp] boiler(0x08) B all(0x00), HpTemperatures(0x048F), data: 00 8F 00 0C 00 25 00 EC (offset 24)

closed 2024-12-31 by HansRemmerswaal