emsesp/EMS-ESP32 #2313: Date is off by 1 year

moritzthecat, 2024-12-30:
### DESCRIPTION

Device / Thermostat shows date by one year off (2025 instead of 2024)
[image]
On Thermostatat terminal the correct date is visible.

Thus MQTT messages and sensor info is off by one year. Logging is not possible as time context is wrong.

NTP service is setup and status reports correct date time.
[image]

Trying to set correct datetime in EMS-ESP (thermostat) and HA both failed
LOG

Log Level
Max Buffer Size
Compact
Auto Scroll
2024-12-30 14:50:40.000 I 1: [emsesp] EMS-ESP version 3.7.2-dev.6 (app0 partition)
2024-12-30 14:50:40.000 I 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2024-12-30 14:50:40.000 I 3: [emsesp] Board profile set to E32V2
2024-12-30 14:50:40.000 I 4: [emsesp] Loaded EMS device library (122 entries)
2024-12-30 14:50:40.000 I 5: [emsesp] Starting Scheduler service
2024-12-30 14:50:40.000 I 6: [emsesp] Starting Custom Entity service
2024-12-30 14:50:40.000 I 7: [emsesp] Ethernet connected (Local IP=192.168.30.117, speed 100 Mbps)
2024-12-30 14:50:40.000 I 8: [emsesp] Starting mDNS Responder service
2024-12-30 14:50:40.000 I 9: [emsesp] Starting NTP service
2024-12-30 14:50:40.000 I 10: [temperaturesensor] Starting Temperature Sensor service
2024-12-30 14:50:40.000 I 11: [analogsensor] Starting Analog Sensor service
2024-12-30 14:50:40.000 I 12: [emsesp] Starting Modules service
2024-12-30 14:50:40.000 I 13: [module] No external library modules found
2024-12-30 14:50:40.000 I 14: [emsesp] Starting Web Server
2024-12-30 14:50:45.529 I 15: [emsesp] IPv6 (Eth) local: fe80:0000:0000:0000:ca2e:18ff:fe92:1837
2024-12-30 14:50:47.529 I 16: [emsesp] IPv6 (Eth) ULA: fd71:5470:59ef:5c4d:ca2e:18ff:fe92:1837
2024-12-30 14:50:55.725 I 17: [emsesp] Detected EMS device: boiler (0x08)
2024-12-30 14:50:56.189 I 18: [emsesp] Detected EMS device: controller (0x09)
2024-12-30 14:50:56.446 I 19: [emsesp] Detected EMS device: thermostat (0x10)
2024-12-30 14:50:56.727 I 20: [emsesp] Detected EMS device: mixer (0x21)
2024-12-30 14:50:56.977 I 21: [emsesp] Detected EMS device: solar (0x30)
2024-12-30 16:05:57.023 I 22: [mqtt] MQTT connected
2024-12-30 16:47:28.334 W 23: [thermostat] Set date: invalid data, wrong length
2024-12-30 16:47:28.336 W 24: [command] Command datetime: 30.12.2024 16:47 failed (Error)
2024-12-30 16:47:28.336 E 25: [mqtt] MQTT command failed with error Command datetime: 30.12.2024 16:47 failed (Error) (Error)
2024-12-30 17:18:51.486 W 26: [thermostat] Set date: invalid data, wrong length
2024-12-30 17:18:51.486 W 27: [command] Command datetime: 30.12.2024 17:18 failed (Error)
2024-12-30 17:19:20.001 W 28: [thermostat] Set date: invalid data, wrong length
2024-12-30 17:19:20.001 W 29: [command] Command datetime: 30.12.2024 17:18 failed (Error)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.2-dev.6","uptime":"000+02:12:25.976","uptimeSec":7945,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":157,"maxAlloc":107,"freeCaps":4170,"usedApp":2399,"freeApp":2273,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4047,"model":""},"network":{"network":"Ethernet","hostname":"ems-esp","TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"192.168.1.1","tzLabel":"Europe/Berlin"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":1658,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"esp32-92182ec8","keepAlive":60,"cleanSession":true,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":15815,"busReads":4435,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"34.11","entities":69,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x1A 0x35 0x26 0x2A 0x2E 0x3B","handlersIgnored":"0x1E 0x23"},{"type":"thermostat","name":"CW400/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"18.06","entities":98,"handlersReceived":"0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x0291 0x02CE 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlersPending":"0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E 0x016E","handlersIgnored":"0xBF"},{"type":"mixer","name":"MM100/MM100","deviceID":"0x21","productID":160,"brand":"","version":"24.05","entities":7,"handlersReceived":"0x02D8","handlersFetched":"0x02CE","handlersIgnored":"0x16 0x33 0x0255 0x0292 0x02E2 0x02EC 0xBF"},{"type":"solar","name":"SM100, MS100","deviceID":"0x30","productID":163,"brand":"","version":"27.05","entities":34,"handlersReceived":"0x0362 0x0363 0x0366 0x0364 0x036A","handlersFetched":"0x0358 0x035A 0x0380 0x038E 0x0391","handlersPending":"0x035D 0x035F 0x035C 0x0361","handlersIgnored":"0x0368 0x03A9 0x03AA 0xBF"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"34.11","entities":0}]}

```

### TO REPRODUCE

Well this is fresh setup and I have no idea what else I can do ?

### EXPECTED BEHAVIOUR

Correct usage of datetime

### SCREENSHOTS

cf. above

proddy, 2024-12-30:
Thanks for reporting. I've confirmed this bug, also happens on my old RC20

MichaelDvP, 2024-12-30:
It's a wrong format in strftime (week based year %G instead of year %Y), so it shows first week 2025.

moritzthecat, 2024-12-30:
have installed EMS-ESP version 3.7.2-dev.7 and can confirm that the issue has been resolved. Year is now 2024.
[image]
thanx for really super fast fix.

closed 2024-12-30 by moritzthecat