emsesp/EMS-ESP32 #2319: Found missing words for translate (version 3.7.2-dev.7)

misa1515, 2025-01-01:
### DESCRIPTION

Found missing translation of words (screenshots attached)

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

### SCREENSHOTS

[image]

[image]

proddy, 2025-01-01:
thanks. need to add those.

mattreim, 2025-01-01:
Hello @misa1515, the words are [there]([interface/src/i18n/sk/index.ts](https://github.com/emsesp/EMS-ESP32/blob/a8775b22002b1a5217ca08ce65d42d5a9bcc5ec5/interface/src/i18n/sk/index.ts)) and just need to be translated.

misa1515, 2025-01-01:
Hello @mattreim , thank you for focus info. Now updated.

closed 2025-01-01 by misa1515