emsesp/EMS-ESP32 #2321: Device rename does not stick

Imwithstupid007, 2025-01-01:
Hi,

I use a BBQKees gateway E32 V2 (EMS-ESP V3.7.1) to monitor my boiler (Nefit Trendline II). In EMS-ESP the boiler shows up as "CBS/GB*72, Trendline, Cerapur, Greenstar Si". 

[image]

I renamed the device several times to "Nefit Trendline II".

[image]

However, after some time (days-weeks) it defaults back to the standard.
The rename "survives" a normal reboot indicating that it is stored properly.

Why is it defaulting back to the standard?

MichaelDvP, 2025-01-02:
The name is stored in the customization file, you can check in settings->download the file and the name. 
Have you restored an older custoization?
Can you specify what have triggert the reset to default name? A crash of ems-esp? Are other data, especially in customization are affected?
We have not seen something like that before, but for mqtt settings there is a similar report that settings are cleared after some time (with unspecified conditions).

Imwithstupid007, 2025-01-02:
I have not restored an older customization.
Only thing i have done after receiving/connecting the device is to select a few favorites and rename the device (and made a mqtt connection).
The first few times this happened I assumed it was a result of "playing around" with the device. But the last time the ESP-EMS was operational for maybe 1-2 weeks and from one day to another the device name was default again. Other changes I made seem to be unaffected (mqtt functions, favorites are accounted for).

The device is restarted yesterday and I will check daily if the name is still correct.

Imwithstupid007, 2025-01-02:
PS: I assume a device restart is effectively the same as a power cycle?

tp1de, 2025-01-02:
I recognized the same behavior while updating between different versions.

MichaelDvP, 2025-01-02:
The rename option is added a few versions ago, using an older version (and changing customizations) will erase the setting. Also the logic for reading  and display the internal name from telegram 0x1 has changed and effects the custom name.

> PS: I assume a device restart is effectively the same as a power cycle?

Mostly, power cycle is a cold start, all registers are in powerup state, restart is a soft reset, u-ram values and some registers are not reseted. But this should not affect the values saved in filesystem like the custom name.

proddy, 2025-01-04:
> The device is restarted yesterday and I will check daily if the name is still correct.

Has the happened again since?

Imwithstupid007, 2025-01-04:
No, name is still correct, even after a few power cycles.

Imwithstupid007, 2025-01-04:
Speak of the devil. I just disconnected and reconnected the power and the device name was back to standard.
Together with the EMS-ESP the boiler was disconnected too (maybe of some use).

proddy, 2025-01-04:
that is weird. When it happens can you
- download the customization file and attach it here (Settings->Download/Upload->Customizations). See if your custom name is in it.
- look at the Status->Hardware->Application and note the partition, as it may be switching. it's app0 or app1

Imwithstupid007, 2025-01-04:
OK, this is the customization file as it is now (with device name changed). The partition is app0.

[emsesp_customizations.json](https://github.com/user-attachments/files/18307466/emsesp_customizations.json)

[image]

I will upload the corresponding items when the device defaults again.

BTW; Shouldn't the changed device name already be in this customization file?

proddy, 2025-01-04:
Yes, it should be and called `custom_name`. Mine looks like:

```json
.....
....
 "masked_entities": [
      {
        "product_id": 123,
        "device_id": 8,
        "custom_name": "Trendline",
        "entity_ids": [
          "08heatingactive",
          "08tapwateractive",
          "08selflowtemp|<90",
          "08curflowtemp",
          "08syspress"
        ]
      }
]
```

Imwithstupid007, 2025-01-04:
Custom name is definitely set.....

[image]

As a test I changed the name again (to "TrendLine II") and now it shows up in the customization file.

>  "masked_entities": [
      {
        "product_id": 123,
        "device_id": 8,
        "custom_name": "TrendLine II",
        "entity_ids": [
          "08heatingactive",
          "00selflowtemp|<90",
          "08curflowtemp",

mattreim, 2025-01-04:
For me there have been a few times with a new firmware where an entity has changed (change of name or translation). The `emsesp_customizations.json` was always reset and I had to enter it manually.

And my BC400 no longer has a manufacturer.
[image]

proddy, 2025-01-05:
@mattreim was the previous a 'brand' shown for your BC400?
And when you say the customization file was reset, what do you mean exactly? All the customizations are gone or just the custom name of the device?

mattreim, 2025-01-05:
> @mattreim was the previous a 'brand' shown for your BC400?

I think so.

>And when you say the customization file was reset, what do you mean exactly? All the customizations are gone or just the custom name of the device?

The custom names look like this again: `"custom_name": ""` and the order.

mattreim, 2025-01-06:
v3.7.2-dev.8 - 1 -> App1
rename boiler to native -> I think because #2322

v3.7.2-dev.8 - 2-> App0
No name changes

v3.7.2-dev.8 - 3-> App1
No name changes -> Thanks, it looks much better

proddy, 2025-01-11:
please re-open if this happens again. I can't reproduce or see why it would loose the custom information,

Imwithstupid007, 2025-01-23:
Today the device name was reset again.

[image]

Custom name in customizations file was also reset:

`{
  "type": "customizations",
  "Customizations": {
    "ts": [],
    "as": [],
    "masked_entities": [
      {
        "product_id": 123,
        "device_id": 8,
        "custom_name": "",
        "entity_ids": [
          "08heatingactive",
          "00selflowtemp|<90",
          "08curflowtemp",
          "08rettemp",
          "08boiltemp",
          "08burnmaxpower|>23<121",
          "08heatingpump",
          "00selburnpow|<121",
          "08curburnpow",
          "08servicecode",
          "08servicecodenumber"
        ]
      }
    ]
  }
}
`

Apperantly for some unknown reason the device reset yesterday.

[image]

Favorites were unaltered.

I have not enough rights to reopen this issue.

proddy, 2025-01-24:
I'll look into the code again to see how this could happen. The frequent reboots is not a good sign and something that should be addressed first. Please update to the latest 3.7.2 and if you still see restarts I think you may need to replace the gateway hardware.

Imwithstupid007, 2025-01-24:
@proddy 
Do you mean switching to the latest development firmware (currently 3.7.2 dev 14 is indicated)?

proddy, 2025-01-24:
> [@proddy](https://github.com/proddy) Do you mean switching to the latest development firmware (currently 3.7.2 dev 14 is indicated)?

yes, then you're on the latest code and we have the same reference point.

Imwithstupid007, 2025-01-24:
> yes, then you're on the latest code and we have the same reference point.

OK, done.

proddy, 2025-01-24:
great, now we wait!

mattreim, 2025-03-03:
When updating to #2439 the `custom_name` was reset, but I can't reproduce it.